lvgl/lvgl #1489: slider knob does not properly track touch point (v6.1)

BillHargen, 2020-05-07:
**Describe the bug**

There is an issue with the slider widget where its value indication does not track where it is being pressed.  This is most noticeable on a wide (horizontal) slider that has a fairly small range.  Touching near the midpoint of the slider precisely sets its value, but as you drag towards either end, the displayed value moved progressively further ahead of the touch point.  The behavior is as if the touchable width of the slider is 80% of its displayed width.

Please note that I am not expecting the knob to exactly track the touch point.  I understand that it will always "snap" to one of the max-min+1 points based on the slider's range.  The issue is that it selects the next value earlier than it should as the touch point moves towards an end.

**To Reproduce**

Create a horizontal slider using most of the screen width.  Set a moderate height (the effect should be more noticeable with a greater height). Set the range to have around 10 values.  Touch near the midpoint and drag towards either end.

**Expected behavior**

The indicated value should always be within one half of the inter-value distance from the touch point, but this is not the case as the touch moves towards an end.  The slider reaches an end much sooner than it should.

**Additional context**

I examined the code in the lv_slider_signal() routine where it computes the new slider value based on the touch coordinate.  There are three bugs with the computation:
1. Converting from pixels to values divides by max-min+1.  It should divide by just max-min.
2. The integer division truncates rather than rounds and there is no compensation for this.
3. The major error is that the computation factors in the width of the slider.  While this is necessary when the "knob in" mode is enabled, it uses the same computation for both modes.  This has the effect of assuming that the slider is shorter and causes the end values to be reached too soon.

I rewrote the computation from scratch and corrected the issue.  I verified it for both the normal and knob-in modes for a horizontal slider.  The diff is below, including a detailed description of the new computation.  I examined the v7.0 code and it is significantly different.  Based on inspection, I doubt that it has this issue, probably because the knob-in option seems to have been removed.

```diff
@@ -530,18 +530,86 @@ static lv_res_t lv_slider_signal(lv_obj_t * slider, lv_signal_t sign, void * par
     } else if(sign == LV_SIGNAL_PRESSING) {
         lv_indev_get_point(param, &p);
         int16_t tmp = 0;
+        /*
+         * The formulas in the code below convert the touch coordinate (in the
+         * range of 0 to screen_width-1) to a slider value (in the slider's min
+         * to max value range). The following describes the basis for these
+         * formulas for the case of a horizontal slider.  It uses the following
+         * abbreviations:
+         *   x: X coordinate of touch (p.x)
+         *   x1: X coordinate of left edge of slider (slider->coords.x1)
+         *   w: slider width, in pixels
+         *   minv: slider's minimum value (ext->bar.min_value)
+         *   maxv: slider's maximum value (ext->bar.max_value)
+         *
+         * The basic formula for a slider at the left edge of the screen (zero-
+         * based X coordinate) and a zero-based slider range is:
+         *   x / w * maxv
+         *
+         * Correcting the order of operation to account for truncating integer
+         * division and biasing the numerator so that the integer division will
+         * round to the closest integer value:
+         *   (x * maxv + w / 2) / w
+         *
+         * Biasing the X input for a slider not at the left edge:
+         *   [(x - x1) * maxv + w / 2] / w
+         *
+         * And biasing the output for a non-zero-based slider range:
+         *   [(x - x1) * (maxv - minv) + w / 2] / w + minv
+         *
+         * This assumed a slider where the knob's center point moves along the
+         * slider's entire width, and the knob's body extends beyond the slider
+         * when at either end.  For the "knob in" case, where the full width of
+         * the knob is contained within the slider's boundary, the effective
+         * width of the slider must be reduced by the width of the knob (by the
+         * left half for when the knob is at the left end plus the right half
+         * for when the knob is at the right end).  Also, it is necessary to
+         * bias the touch coordinate by half the width of the knob.
+         *
+         * The modified formula for the "knob in" case adds the following
+         * abbreviation:
+         *   kw: knob's width; same as the slider's height (h)
+         *
+         * The adjusted formula is:
+         *  [(x - x1 - kw / 2) * (maxv - minv) + (w - kw) / 2] / (w - kw) + minv
+         *
+         * Note that, while the adjusted formula is identical to the prior
+         * formula when the knob width is 0, and the code could use a single
+         * computation and just set "kw" to zero for the non-knob-in case,
+         * separate computations are done for execution efficiency.
+         *
+         * Finally, for the cases of vertical sliders, the X coordinates are
+         * replaced with Y, and the slider width and height (knob width) are
+         * interchanged.
+         */
         if(w > h) {
-            lv_coord_t knob_w = h;
-            p.x -=
-                slider->coords.x1 + h / 2; /*Modify the point to shift with half knob (important on the start and end)*/
-            tmp = (int32_t)((int32_t)p.x * (ext->bar.max_value - ext->bar.min_value + 1)) / (w - knob_w);
-            tmp += ext->bar.min_value;
+            /*Horizontal slider*/
+            if(ext->knob_in == 0) {
+                lv_coord_t x1   = slider->coords.x1;
+                lv_coord_t minv = ext->bar.min_value;
+                lv_coord_t maxv = ext->bar.max_value;
+                tmp = (((int32_t)p.x - x1) * (maxv - minv) + w / 2) / w + minv;
+            } else {
+                lv_coord_t x1   = slider->coords.x1;
+                lv_coord_t minv = ext->bar.min_value;
+                lv_coord_t maxv = ext->bar.max_value;
+                lv_coord_t kw   = h;
+                tmp = (((int32_t)p.x - x1 - kw / 2) * (maxv - minv) + (w - kw) / 2) / (w - kw) + minv;
+            }
         } else {
-            lv_coord_t knob_h = w;
-            p.y -=
-                slider->coords.y1 + w / 2; /*Modify the point to shift with half knob (important on the start and end)*/
-            tmp = (int32_t)((int32_t)p.y * (ext->bar.max_value - ext->bar.min_value + 1)) / (h - knob_h);
-            tmp = ext->bar.max_value - tmp; /*Invert the value: smaller value means higher y*/
+            /*Vertical slider*/
+            if(ext->knob_in == 0) {
+                lv_coord_t y1   = slider->coords.y1;
+                lv_coord_t minv = ext->bar.min_value;
+                lv_coord_t maxv = ext->bar.max_value;
+                tmp = (((int32_t)p.y - y1) * (maxv - minv) + h / 2) / h + minv;
+            } else {
+                lv_coord_t y1   = slider->coords.y1;
+                lv_coord_t minv = ext->bar.min_value;
+                lv_coord_t maxv = ext->bar.max_value;
+                lv_coord_t kh   = w;
+                tmp = (((int32_t)p.y - y1 - kh / 2) * (maxv - minv) + (h - kh) / 2) / (h - kh) + minv;
+            }
         }
 
         if(tmp < ext->bar.min_value)
```

embeddedt, 2020-05-07:
@BillHargen Thank you for the detailed write-up! Can you send a pull request?

embig71, 2020-05-08:
For a vertically oriented slider, the coordinates must be inverted, because in this case, the slider moves mirrored, i.e.

            /*Vertical slider*/
            if(ext->knob_in == 0) {
                lv_coord_t y1   = slider->coords.y1;
                **lv_coord_t minv = ext->bar.max_value;**
                **lv_coord_t maxv = ext->bar.min_value;**
                tmp = (((int32_t)p.y - y1) * (maxv - minv) + h / 2) / h + minv;
            } else {
                lv_coord_t y1   = slider->coords.y1;
                **lv_coord_t minv = ext->bar.max_value;**
                **lv_coord_t maxv = ext->bar.min_value;**
                lv_coord_t kh   = w;
                tmp = (((int32_t)p.y - y1 - kh / 2) * (maxv - minv) + (h - kh) / 2) / (h - kh) + minv;
            }

kisvegabor, 2020-05-08:
@embig71 
Thank you for pointing this out.
Could you send a new pull request?

BillHargen, 2020-05-08:
My apologies for breaking this.  I agree with the proposed fix (though I would add a comment to explain it).
@embig71 Let me know if you would prefer for me to create the pull request.

embeddedt, 2020-05-11:
Closed by #1490 and #1492. Thanks to both of you!

closed 2020-05-11 by embeddedt