lvgl/lvgl #1491: Driver initialization references uninitialized memory

xennex22, 2020-05-07:
Call chain
`lv_disp_drv_register()`
`lv_obj_create(NULL, NULL)`
`lv_theme_apply()`
`theme_apply()`
`lv_obj_refresh_style()`
`lv_obj_invalidate()`
`lv_obj_invalidate_area()`
Which then compares the current screen with the sys and top layers, which have not been created yet.

I'm wondering if `theme_apply()` should not call `lv_obj_refresh_style` for screen themes.
```
if(name != LV_THEME_SCR)
    lv_obj_refresh_style(obj, LV_STYLE_PROP_ALL);
```

Or `lv_disp_drv_register` should NULL out the layers and default screen.

xennex22, 2020-05-07:
Probably `theme_apply()` needs to be call `lv_obj_refresh_style` for screen themes if you are changing themes at runtime.

kisvegabor, 2020-05-08:
Hi,

Good catch. I've fixed it [here](e6f498a2) by clearing the allocated `disp`.

xennex22, 2020-05-08:
The fix looks good.  Thank you.

closed 2020-05-08 by xennex22