lvgl/lvgl #1497: Change lv_calendar signal function (in 6.1 version)

embig71, 2020-05-14:
Hi.
When using the calendar widget, I found a problem of quick and convenient change of the current year in lv_calendar. For example, such a need arises when RTC fails. I propose to divide the calendar header into three active zones. Use the left and right as before to change the current month, and the middle one to select the year at the application level.

**src/lv_objx/lv_calendar.c**
```
        /*If the header is pressed mark an arrow as pressed*/
        if(lv_area_is_point_on(&header_area, &p)) {
            if(p.x < header_area.x1 + lv_area_get_width(&header_area) / 2) {
            lv_coord_t w = lv_area_get_width(&header_area);
            ext->year_pressing = 0;
            if(p.x < header_area.x1 + (w / 3)) {
                if(ext->btn_pressing != -1) lv_obj_invalidate(calendar);
                ext->btn_pressing = -1;
            } else {
            } else if(p.x > header_area.x1 + (2*w / 3)) {
                if(ext->btn_pressing != 1) lv_obj_invalidate(calendar);
                ext->btn_pressing = 1;
            }
            } else ext->year_pressing = 1;

            ext->pressed_date.year  = 0;
            ext->pressed_date.month = 0;
```

**src/lv_objx/lv_calendar.h**
```
    lv_calendar_date_t * highlighted_dates; /*Apply different style on these days (pointer to an
                                               array defined by the user)*/
    int8_t btn_pressing;                    /*-1: prev month pressing, +1 next month pressing on the header*/
    int8_t year_pressing;                   /*1: pressing on year area*/
    uint16_t highlighted_dates_num;          /*Number of elements in `highlighted_days`*/
    lv_calendar_date_t pressed_date;
    const char ** day_names;   /*Pointer to an array with the name of the days (NULL: use default names)*/
```
I used to select the year ddlist from the given list of years:
```
static lv_obj_t *mbox1 = NULL;
static lv_obj_t *cal1 = NULL;
static const int first_year = 2018, last_year = 2100;
//--------------------------------------------------------------------------------------------------
//  ddlist event handler
//--------------------------------------------------------------------------------------------------
static void ddl_setyear_cb (lv_obj_t * obj, lv_event_t event) {
    if (event == LV_EVENT_VALUE_CHANGED) {
        char buf[32];
        lv_ddlist_get_selected_str (obj, buf, sizeof(buf));
        printf ("ddl_setyear option: %s\n", buf);
        fflush (stdout);
        int year = lv_ddlist_get_selected (obj) + first_year;
        lv_calendar_date_t *date = lv_calendar_get_today_date (cal1);
        date->year = year;
        lv_calendar_set_today_date (cal1, date);
        lv_calendar_set_showed_date (cal1, date);
        lv_obj_set_click (cal1, true); // Restore calendar click activity
        lv_obj_del_async (obj);
    }
}

//--------------------------------------------------------------------------------------------------
//  Calendar event handler
//--------------------------------------------------------------------------------------------------
static void calendar_cb (lv_obj_t * obj, lv_event_t event) {
    if (event == LV_EVENT_CLICKED) {
        lv_calendar_ext_t *ext = (lv_calendar_ext_t*)lv_obj_get_ext_attr (obj);
        //printf ("calendar_evt_handler   %d   %d\n", (int)event, ext->year_pressing);
        //fflush (stdout);
        if (ext->year_pressing) {
            lv_obj_t *ddl1 = lv_ddlist_create (cal1, NULL);
            char years[2048], buf1[64]; 
            years[0] = '\0';
            for (int i = first_year; i <= last_year; ++i) {
                sprintf (buf1, "%4d", i);
                strcat (years, buf1);
                if (i < last_year) strcat (years, "\n");
            }
            // Make ddlist for year select
            lv_ddlist_set_options (ddl1, years);
            lv_ddlist_set_anim_time (ddl1, 200);
            lv_ddlist_set_sb_mode (ddl1, LV_SB_MODE_DRAG);
            lv_ddlist_set_fix_width (ddl1, lv_obj_get_width (obj) / 3);
            lv_ddlist_set_draw_arrow (ddl1, true);
            lv_ddlist_set_align (ddl1, LV_LABEL_ALIGN_CENTER);
            if ((last_year - first_year + 1) > 8) lv_ddlist_set_fix_height (ddl1, lv_obj_get_height (obj) - 15);
            lv_obj_align (ddl1, obj, LV_ALIGN_IN_TOP_MID, 0, 3);
            lv_obj_set_event_cb (ddl1, ddl_setyear_cb);
            lv_calendar_date_t *date = lv_calendar_get_today_date (cal1);
            int year_idx = date->year - first_year;
            lv_ddlist_set_selected (ddl1, year_idx < 0 ? 0 : year_idx); // Select current year in list
            lv_ddlist_open (ddl1, LV_ANIM_ON);
            lv_obj_set_click (cal1, false); // Disable calendar click activity
        } else {
            lv_calendar_date_t *date = lv_calendar_get_pressed_date (obj);
            if (date) {
                lv_calendar_set_today_date (obj, date);
            }
        }
    } else if (event == LV_EVENT_LONG_PRESSED) {
    }
}

//--------------------------------------------------------------------------------------------------
// Set date button event handler
//--------------------------------------------------------------------------------------------------
static void setdate_cb (lv_obj_t * obj, lv_event_t event) {
    if (event == LV_EVENT_CLICKED) {
        mbox1 = create_modal_box ((int)(0.95 * LV_HOR_RES), (int)(0.95 * LV_VER_RES), 1);
        cal1 = lv_calendar_create (mbox1, NULL);
        lv_obj_set_event_cb (cal1,  calendar_cb);
        lv_obj_align (cal1, NULL, LV_ALIGN_IN_TOP_LEFT, 5, 5);
        lv_obj_set_size (cal1, lv_obj_get_width (mbox1) - 55, lv_obj_get_height (mbox1) - 15);

        lv_calendar_date_t today;
        time_t rawtime;
        struct tm *ptm;
        time( &rawtime );
        ptm = localtime( &rawtime );
        today.year = 1900 + ptm->tm_year;
        today.month = ptm->tm_mon + 1;
        today.day = ptm->tm_mday;

        lv_calendar_set_today_date (cal1, &today);
        lv_calendar_set_showed_date (cal1, &today);
        //obj_expand_animated (mbox1, 300, &lv_app_idle);
        obj_expand_animated (mbox1);
    }
}

```

embeddedt, 2020-05-14:
I like the idea! Unfortunately, it can only be added to 7.1, but I expect a branch to be made very soon for that.

I especially like the approach of leaving it up to the user to create the dropdown, and only implementing the functionality for telling whether the year is pressed inside the calendar itself.

ali-rostami, 2020-05-15:
Hi
Another idea for changing the year and month is doing the same thing we have in win 10.
I think it's nicer than using a dropdown list and doesn't need more space than the calendar itself.
what do you think?

embig71, 2020-05-15:
Hi
Yes, of course, visually it is much better. I did not want to change the functionality of the widget much and transfer the implementation of the choice of the year to the application level. And ddlist does not require a rectangle more than a calendar rectangle.
[image]

kisvegabor, 2020-05-15:
I also like this improvement. The only thing I miss is an API function for that. 
Could you send a PR based on `dev-7.0`?

embig71, 2020-05-15:
Sorry. I use only branch 6. I can do PR for 6.1.2 if that works. And I did not expect an additional API.

embeddedt, 2020-05-15:
> I can do PR for 6.1.2 if that works. 

We can try it. (It will probably have to be merged manually as there's been some changes to folder names, etc.)

I think @kisvegabor would like an API for retrieving `ext->year_pressing`.

embig71, 2020-05-15:
For access isolation to ext struct? Ok.

kisvegabor, 2020-05-18:
> I think @kisvegabor would like an API for retrieving ext->year_pressing.

Yes, I meant that. :slightly_smiling_face:

embeddedt, 2020-06-08:
I'll close this issue as the PR is approved and waiting in #1501.

closed 2020-06-08 by embeddedt