lvgl/lvgl #1500: [7.0] lv_img_design() debug code left in

xennex22, 2020-05-15:
I'm fairly sure the second line was left in by mistake after debugging.
```
if(mode == LV_DESIGN_COVER_CHK) {
        return LV_DESIGN_RES_NOT_COVER;
        lv_design_res_t cover = LV_DESIGN_RES_NOT_COVER;
        if(ext->src_type == LV_IMG_SRC_UNKNOWN || ext->src_type == LV_IMG_SRC_SYMBOL ||
           ext->angle != 0) return LV_DESIGN_RES_NOT_COVER;

        if((ext->cf == LV_IMG_CF_TRUE_COLOR || ext->cf == LV_IMG_CF_RAW) && ext->angle == 0 && ext->zoom == LV_IMG_ZOOM_NONE) {
            cover = _lv_area_is_in(clip_area, &img->coords, 0) ? LV_DESIGN_RES_COVER : LV_DESIGN_RES_NOT_COVER;
        }

        if(lv_obj_get_style_image_opa(img, LV_IMG_PART_MAIN) < LV_OPA_MAX) return LV_DESIGN_RES_NOT_COVER;

        return cover;
```

C47D, 2020-05-15:
OFFTOPIC: Hi @xennex22, how do you find this kind of issues? Static code analysis? Code coverage?

I should the compiler should mark the code after the `return LV_DESIGN_RES_NOT_COVER;` line as non reached and throw an error or a warning?

xennex22, 2020-05-16:
The compiler complains about a dynamic assignment is dead code.

kisvegabor, 2020-05-18:
It was a leftover from debugging something. I've removed it, thank you.

closed 2020-05-18 by kisvegabor