lvgl/lvgl #1504: lvgl render object display abnormal

lgl88911, 2020-05-17:
I test in RGB565 LCD panel of embed platfrom, all lvgl object display abnormal.The same code is displayed normally on SDL of Emulator,  I think the two differ only in color format。
You can see this phenomenon from the picture below, it seems that the lvgl object is covered with some color bars
[image]

barbiani, 2020-05-17:
Check the invert 16 bit color for spi devices in the config file.

lgl88911, 2020-05-18:
@barbiani  Thanks for your reply
I config below
```
#define LV_COLOR_16_SWAP 1
```
The graphics using lvgl color are not normal，I use the RGB interface LCD ，So there shouldn't be a problem invert 16 bit color for spi devices
[image]

kisvegabor, 2020-05-19:
@lgl88911 It's hard to decide but I'm not sure that the "working LV_COLOR_BLUE" is really the brightest blue. Can you draw a box next to "seems to be good" blue rectangle like this:
```c
for(i = 0; i < 0xffff; i+= 4) {
  your_box_draw(10, 10, 20, 20, i);
}
```

and see is there is a more blue color?

lgl88911, 2020-05-19:
@kisvegabor Thanks for your reply！
I'm not too sure what you mean. LV_COLOR_BULE looks OK.
The problem I encountered was that the object displayed abnormally. Can you tell me what kind of problem the experiment you want is to check?

I assume you want to see all the blue. For RGB565, I have drawn the colors 0x0000 ~ 0x001F. These cover all the blue. You can see this color bar in the middle of the picture below
[image]
the test code is
```
for(int i = 0; i<0x01f; i++){

			for(int j=0; j<BUF_W*BUF_H; j++){
				color[j] = i;
			}

				desc.buf_size = 20 * 2U * 20;
				desc.width = 20;
				desc.pitch = 20;
				desc.height = 40;

				display_write(display_device, 50+i*20, 180, &desc, buf);
		}
```

kisvegabor, 2020-05-19:
I meant it's hard to decide if its 100% blue or 80%. But your gradient image confirms that it's fine. Can you do the same with red and green too?

lgl88911, 2020-05-19:
@kisvegabor Thanks so much! 
Thanks to your extensive experience, it seems that the problem has been found, RED is abnormal.
In order to distinguish red errors more clearly, I added a gap between each color.
I think this may be a problem with the driver or hardware connection. I am checking the related content. If you have further suggestions, I will be very grateful.

[image]

```
			u16_t *colorg = buf;
		for(int i = 0; i<0x3f; i++){

			for(int j=0; j<BUF_W*BUF_H; j++){
				colorg[j] = i<<5;
			}

				desc.buf_size = 10 * 2U * 20;
				desc.width = 10;
				desc.pitch = 10;
				desc.height = 40;

				display_write(display_device, 50+i*10, 230, &desc, buf);
		}

			u16_t *colorr = buf;
		for(int i = 0; i<0x01f; i++){

			for(int j=0; j<BUF_W*BUF_H; j++){
				colorr[j] = i<<11;
			}
				printf("r %04x\n", colorr[0]);
				desc.buf_size = 20 * 2U * 20;
				desc.width = 20;
				desc.pitch = 20;
				desc.height = 40;

				display_write(display_device, 50+i*22, 280, &desc, buf);
		}
```

embeddedt, 2020-05-19:
Perhaps you have some parallel interface wires swapped? I don't know how your display connects to the main board.

lgl88911, 2020-05-19:
@embeddedt 
Indeed, It's connects error, I have fixed it. Now work well, Thanks so much
[image]
[image]

closed 2020-05-19 by lgl88911

kisvegabor, 2020-05-20:
Great! :slightly_smiling_face:

lgl88911, 2020-05-21:
@kisvegabor @embeddedt @barbiani Thanks for your help!