lvgl/lvgl #1506: How to chang width between buttons in button matrix?

Rhaoma, 2020-05-18:
I'm quite new to LittleVGL, and just coding in general. As of known everything seems to work fine, but i cant seem to figure out how to change the width **between** the buttons in a button matrix. Because of the relatively small display the space between the buttons causes them to be a quite bit smaller, than if the space hadn't been so big. I feel like im just oblivious, and have overlooked something plain obvious, but I keep going back to the site documentation and I still can't find anything.

Any one who has an idea of how to change size of the space between the buttons?

embeddedt, 2020-05-18:
I think increasing the padding on `LV_BTNM_STYLE_BG` should also increase the space between the buttons.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-05-18 by embeddedt

Rhaoma, 2020-05-18:
Couldn't quite make it work, but thanks for the suggestion.