lvgl/lvgl #1510: Navigation and editing with few buttons

fhorinek, 2020-05-19:
Hi,

I am building interface with no touch screen and with just 5 buttons (left, right, enter, back, menu).
I need to navigate and edit the widgets with them. But there is no input type which support this kind of input.

`LV_INDEV_TYPE_KEYPAD` does not support editing and using `LV_INDEV_TYPE_ENCODER` will need additional layer in application to simulate stepping when the button is held.

I need some hybrid input method for this.
How would you prefer to implement this?

My idea is to declare new type `LV_INDEV_TYPE_NAVIGATION` and create new `indev_navigation_proc` which will forward the input to keypad or encoder proc

or to have option `can_edit` in `lv_indev_drv_t` and modify `indev_keypad_proc`

thank you for your suggestions

kisvegabor, 2020-05-20:
Hi,

This type of navigation is really missing. 

I think it's closer to the encoder indev because of edit handling. At first look, it seems only [this part](https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_indev.c#L611-L632)  needs to be extended with `UP` and `DOWN` handling.

puzrin, 2020-05-20:
I'd recommend to check web docs about "Accessibility". Very primitive explanation sounds "How to make webpage usable with keyboard only" (for Web-TV and other devices without mouse). That will give you main ideas about principles.

Currently lvgl has no appropriate machinery. If you remember, i had to make hacks to emulate autorepeat for long press in my dispencer project.

fhorinek, 2020-05-21:
I decided to create new indev type for this. This way it won't create bugs in existing keypad or encoder procs.

#1514

vlisivka, 2020-05-28:
I did this for device with Up, Down, Enter, and Esc keys, see there: https://github.com/vlisivka/lvgl_dash/commit/7cf367432dccf2a79000265d62e7b0ea698e77c3

puzrin, 2020-05-28:
@kisvegabor this may be a topic for roadmap (keyboard navigation). In context of web approach:

- LVGL requires manual magick with `group`, adding unconvenient tight-coupling from all app code to indev.
- On webpage everything is organized automacally via `tabindex` atributes inside of parent childs. Widgets receives only focus on/leave + keycodes. Widgets don't know anything about indev.

I mean, web uses "page layout" to describe navigation. That's more generic & flexible than "group".

fhorinek, 2020-05-28:
I forgot to close this.
already implemented and approved for merge 
https://github.com/lvgl/lvgl/pull/1522

closed 2020-05-28 by fhorinek

positron96, 2024-10-20:
Hello guys. I have exactly the same setup (5 nav buttons) and am wondering what input method I should use, and how. I have a KEYPAD input device; should I just change the type to ENCODER and continue to send keys and states?
(there is a link to documentation in the PR, but it's broken)

kisvegabor, 2024-10-23:
Right now I think KEYPAD is the best choice. , However, we need to add long press and repeat support for non-ENTER buttons too as it was already requested many times.  Now if these extra events are needed they requires some custom code in the driver. :slightly_frowning_face:

kisvegabor, 2024-10-24:
Great! 

>  I tried keypad type input device, and could not make navigation/edit change on long press work

Yes, navigation/edit mode is only for encoders.