lvgl/lvgl #1512: website/blog link issue

trollcop, 2020-05-20:
> # Important: issues that don't use this template will be ignored/closed.

Link to 'complete project' from lvgl blog is incorrect.

A clear and concise description of what the bug is.

1. open https://blog.lvgl.io/2018-01-03/linux_fb#download-a-ready-to-use-project
2. click the In "lv_port_linux_frame_buffer" repository link
3. the link leads to https://github.com/lvgl/lv_port_linux_frame_bufferr (notice extra 'R' at end)
4. observe 404 error from github

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally this should work in the PC simulator unless the problem is specific to one platform.

Same as above.

A clear and concise description of what you expected to happen.

It should go to https://github.com/lvgl/lv_port_linux_frame_buffer as intended.

Thanks

embeddedt, 2020-05-20:
Fixed in https://github.com/lvgl/blog/commit/84a99ce4053073993c8931a70e4a4fe0d466056d.

closed 2020-05-20 by embeddedt

multitab2022, 2023-12-12:
I was worried because the contents of the blog were not compiled.
https://blog.lvgl.io/2018-01-03/linux_fb
- main.c

Thank you for the information.

https://github.com/lvgl/lv_port_linux_frame_buffer