lvgl/lvgl #1513: Latest build crashing when calling lv_theme_material_init() function to update theme colours

pete-pjb, 2020-05-21:
**Describe the bug**

With the latest master LVGL build the system is crashing when calling the lv_theme_material_init() function to update colours and dark or light theme settings.

I have used the following function to date with no issues to update the theme when the user alters colours or chooses dark/ light theme:

```c
static void update_gui_theme( void ) {

	lv_theme_t	*th;

	th = lv_theme_material_init(confp->gui_colour, LV_THEME_DEFAULT_COLOR_SECONDARY, confp->style ? LV_THEME_MATERIAL_FLAG_LIGHT : LV_THEME_MATERIAL_FLAG_DARK,
			LV_THEME_DEFAULT_FONT_SMALL, LV_THEME_DEFAULT_FONT_NORMAL, LV_THEME_DEFAULT_FONT_SUBTITLE, LV_THEME_DEFAULT_FONT_TITLE);
	lv_theme_set_act(th);										//Apply the theme
	setup_keyboard( confp->gui_colour );
	update_scrl_bar_style();
	lv_obj_invalidate(lv_scr_act());
}
```

**To Reproduce**

Call lv_theme_material_init() after lvgl is up and running.

**Expected behavior**

Reinitialise theme with out crashing.

**Screenshots**
See Call stack...

[image]

Any thoughts would be appreciated.

Kind Regards,

Pete

kisvegabor, 2020-05-21:
I fixed it, thank you.

Some updates ware accidentally merged to master.

pete-pjb, 2020-05-21:
Thanks @kisvegabor all good now!

closed 2020-05-21 by pete-pjb