lvgl/lvgl #1515: [master] Roller with encoder behavior

Wielebny666, 2020-05-21:
Hi,
roller with encoder not working properly.
When turning left the menu scrolls up. 
To the right doesn't react, sometimes jumps to another position.

```
 model_choise = lv_roller_create(parent, NULL);
    lv_roller_set_options(model_choise, "Audi\nMercedes\nOpel\nVolvo\nVW", LV_ROLLER_MODE_INIFINITE);
    lv_roller_set_visible_row_count(model_choise, 2);
    lv_obj_align(model_choise, NULL, LV_ALIGN_IN_BOTTOM_RIGHT, 0, 0);
    lv_obj_set_event_cb(model_choise, model_choise_event_cb);
    lv_group_add_obj(group, model_choise);
```
in v6 everything was fine.

regards

kisvegabor, 2020-05-22:
Hi,

I fixed it here: ab4d8c91

kisvegabor, 2020-05-28:
As the issue should be fixed I close it now.

closed 2020-05-28 by kisvegabor