lvgl/lvgl #1516: slimulator

zhepoch, 2020-05-22:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

A clear and concise description of what the bug is.

**To Reproduce**

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally this should work in the PC simulator unless the problem is specific to one platform.

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

kisvegabor, 2020-05-22:
Hi,

Probably it happens becasue you run your these code in three without using a mutex around `lv_task_handler`. `tick_thread` can interrupt the internal lvgl tasks at any time and mess up things. Read more here: https://docs.lvgl.io/v7/en/html/porting/os.html

It works well if I move the test code into the main thread:
```c

  lv_obj_t *scr;
  lv_obj_t *btn1;
  lv_obj_t *btn2;
  lv_obj_t *btn3;
  lv_coord_t x_cord1 = 20;
  lv_coord_t y_cord1 = 20;
  lv_coord_t x_cord2 = 20;
  lv_coord_t y_cord2 = 20;
  lv_coord_t x_cord3 = 20;
  lv_coord_t y_cord3 = 20;
  int cnt1 = 800;
  int cnt2 = 400;
  int cnt3 = 0;
  int i= 0;

  scr = lv_disp_get_scr_act(NULL);
  btn1 = lv_btn_create(scr, NULL);
  btn2 = lv_btn_create(scr, NULL);
  btn3 = lv_btn_create(scr, NULL);
  lv_obj_set_size(btn1, 50, 50);
  lv_obj_set_size(btn2, 50, 50);
  lv_obj_set_size(btn3, 50, 50);
  lv_style_t style_btn3;
  lv_style_t style_btn2;

  lv_style_init(&style_btn2);
  lv_style_set_bg_color(&style_btn2, LV_STATE_DEFAULT, LV_COLOR_RED);
  lv_obj_add_style(btn2, LV_BTN_PART_MAIN, &style_btn2);
  lv_style_init(&style_btn3);
  lv_style_set_bg_color(&style_btn3, LV_STATE_DEFAULT, LV_COLOR_GREEN);
  lv_obj_add_style(btn3, LV_BTN_PART_MAIN, &style_btn3);

  while (1) {
      /* Periodically call the lv_task handler.
       * It could be done in a timer interrupt or an OS task too.*/
       lv_task_handler();
      usleep(5 * 1000);

      if (++i >= 1) {
          i = 0;
          if (cnt1 < 400) {
              x_cord1 = 20 + cnt1;
          } else if (cnt1 >= 400 && cnt1 < 600) {
              y_cord1 = 20 + cnt1 - 400;
          } else if (cnt1 >= 600 && cnt1 < 1000) {
              x_cord1 = 1000 - cnt1 + 20;
          } else if (cnt1 >= 1000 && cnt1 < 1200) {
              y_cord1 = 1200 - cnt1 + 20;
          } else {
              cnt1 = 0;
          }
          cnt1 +=1;
          lv_obj_set_pos(btn1, x_cord1, y_cord1);

          if (cnt2 < 400) {
              x_cord2 = 20 + cnt2;
          } else if (cnt2 >= 400 && cnt2 < 600) {
              y_cord2 = 20 + cnt2 - 400;
          } else if (cnt2 >= 600 && cnt2 < 1000) {
              x_cord2 = 1000 - cnt2 + 20;
          } else if (cnt2 >= 1000 && cnt2 < 1200) {
              y_cord2 = 1200 - cnt2 + 20;
          } else {
              cnt2 = 0;
          }
          cnt2 += 1;

          lv_obj_set_pos(btn2, x_cord2, y_cord2);

          if (cnt3 < 400) {
              x_cord3 = 20 + cnt3;
          } else if (cnt3 >= 400 && cnt3 < 600) {
              y_cord3 = 20 + cnt3 - 400;
          } else if (cnt3 >= 600 && cnt3 < 1000) {
              x_cord3 = 1000 - cnt3 + 20;
          } else if (cnt3 >= 1000 && cnt3 < 1200) {
              y_cord3 = 1200 - cnt3 + 20;
          } else {
              cnt3 = 0;
          }
          cnt3 += 1;

          lv_obj_set_pos(btn3, x_cord3, y_cord3);
      }

  }
```

kisvegabor, 2020-05-28:
As the above proposed code should fix the issue I close it.

If it's still not working as expected please comment here and reopen the issue.

closed 2020-05-28 by kisvegabor