lvgl/lvgl #1523: Feature proposal:  run lvgl applications in WebAssembly

xwang98, 2020-05-25:
Feature proposal:  run lvgl applications in WebAssembly

We would like to introduce a new usage of lvgl in WebAssembly Micro Runtime (WAMR,  https://github.com/bytecodealliance/wasm-micro-runtime) project. WAMR is a [bytecode alliance](https://bytecodealliance.org/) project for standalone WebAssembly runtime that supports a wide range environment. 

There are two lvgl related samples in WAMR project:

- [littlevgl](https://github.com/bytecodealliance/wasm-micro-runtime/tree/master/samples/littlevgl) :  the lvgl application and lvgl library are built together into WASM bytecode
- [gui](https://github.com/bytecodealliance/wasm-micro-runtime/tree/master/samples/gui) : The lvgl library is built into the host runtime, and lvgl user codes are built into WASM bytecode. It is more an experiment and just enabled a few widgets.

The samples demostrated lvgl based applications can be dynamically installed to target devices. The sandbox mechanism by WASM even enable running 3rd parity apps on your own devices. 

The `littlevgl` is a quite easy way to implement, but it will cause relative bigger binary size since to whole lvgl library is built into the WASM bytecode. The `gui` sample produce very small wasm binary size (less 2K in the sample), which is very useful in some embedded environment. As the lvgl was designed to build together with user code, it require quite a bit extra effort to implement the `gui` usage model .

This proposal is about some lvgl programming interface change to enable smooth support for the separation of lvgl library (in host runtime) and lvgl user application in wasm bytecode. 

**1) partition current lvgl API**

In the lvgl WASM usage, the host runtime should do setup and configuration  which is just like ususal usage in embedded environment. And this part is supposed to be implemented by the device vendor. The apps in WASM propbably only call the widgets related API. maybe we should separate the currently livgl.h to two header files and one of them is used by wasm application code. 

**2) all the widget object structrue is no longer visible to the wasm application**. All the created widget objects are returned as handles. This is because lvgl user code won't be able to access the widget data in native space due to the wasm sandbox. 

**3) avoid using data structure type in the API auguments.** To pass the structured data between WASM and nature will require additional effort  for serialization.

**4) callback** is supported in different way when the wasm app needs the runtime to call back to its function. But it should be fine here.

**5) consider lvgl user code may be programmed in other languges**, like TypeScript, Rust and Go. It should be no problem in theory, just need some effort to expriment.

There is some information related to this topic here:

https://github.com/bytecodealliance/wasm-micro-runtime/tree/master/core/app-framework/wgl

In a summary, if lvgl can cover this requirement by WASM usage in a future release, we will be very easy to leverage WASM to build mini lvgl apps that can independently developed by 3rd party and dynamically installed to the devices. If lvgl community is interested in this opportunity, we can have more dicussion here.

embeddedt, 2020-05-25:
I like the idea of creating a call gate-styled API. It would also be useful for integrating LittlevGL into an operating system and then having user applications independently create/delete their own objects.

kisvegabor, 2020-05-27:
@xwang98 Thank you for introducing your project. It offers great opportunities for LVGL users.

Let's prioritize the problems:

> 1) partition current lvgl API

It's not blocking, right? Just "nice to have".

> 2) all the widget object structrue is no longer visible to the wasm application

LVGL has set/get functions to modify structure elements, so it should be fine. 

> 3) avoid using data structure type in the API auguments. 

Does this code still works?
```c
lv_anim_t a;
lv_anim_init(&a);
lv_anim_set_time(&a, 1000); //Animate in 1000 ms
lv_anim_set_exec_xcb(&a, lv_obj_set_x); //Call this to animate
...
```

If so can you show a code example that is not working?

> 4) callback 

It's fine if fine here. :slightly_smiling_face: 

> 5) consider lvgl user code may be programmed in other languges

We had a discussion about it regarding the Micropython binding and could eliminate some functions  by not including them into `lvgl.h` or adding an `_` prefix to private functions. (The Micropython binding is generated with a custom script  which can skip prefixed functions). Could you use this trick somehow as well?

kisvegabor, 2020-07-08:
@xwang98 
Could you comment on this?

xwang98, 2020-07-12:
@kisvegabor 

Here is my input for your feedback on each item:
#1: yes
#2: yes. 
#3: I think it may not work but I am not very sure. 
#4: right
#5: I am not sure. 

Overall, to support such use case we will need revisit all the lvgl APIs and classify what part should be exposed to the user code running in WebAssembly. It is a bit like the Windows GUI handles where the widges data is located in the kernal(?), and we need to ensure the object data is only access by set/get on the object handle.

kisvegabor, 2020-07-14:
If I recall correctly, right now there is no "field level" manipulation in LVGL. That is all data is set/get via functions.

closed 2020-08-15 by stale[bot]