lvgl/lvgl #1525: Latest master merge has broken button states...

pete-pjb, 2020-05-26:
**Describe the bug**

The latest commit to the master repository appears to have broken the button states ie the LV_BTN_STATE_DISABLED no longer disables the button and state LV_BTN_STATE_ACTIVE has completely disappeared?  I am guessing that some kind of regression has happened?

**To Reproduce**
Use latest build of LVGL

**Expected behavior**
Button States to function as they did prior to last merge...

Kind Regards,

Pete

pete-pjb, 2020-05-26:
It seems having merged the next newest version the LV_BTN_STATE_DISABLED has begun to work again.  With the removal of the LV_BTN_STATE_ACTIVE I have replaced this state with LV_BTN_STATE_RELEASED in my code and all appears to be well...

I will close this issue as I have a working system again...

closed 2020-05-26 by pete-pjb

kisvegabor, 2020-05-27:
Hi,

Actually the old version was a regression and the last version without `LV_BTN_STATE_ACTIVE` is valid.

pete-pjb, 2020-05-27:
Okay thanks for the confirmation @kisvegabor. 👍 :smiley: