lvgl/lvgl #1526: can't display Turkish characters with UTF8 enabled

kadirsevil, 2020-05-26:
Hi,
I enabled utf8 support in the lvgl_conf.h file.
`#define LV_TXT_ENC LV_TXT_ENC_UTF8`

When i create tab with title that has Turkish characters, only ascii characters is displayed.
`lv_tabview_add_tab(tv, "Türkçe");`

Do I make something wrong?
Thanks by advance,

Kadir

embeddedt, 2020-05-26:
Your font also needs to have Turkish characters included in it. I'm not sure if the default font includes these or not.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-05-26 by embeddedt

kadirsevil, 2020-05-26:
How can i add new font?

embeddedt, 2020-05-26:
It's explained in the documentation. Look under Overview -> Fonts.