lvgl/lvgl #1527: lv_spinbox_set_range range_min not working

luksal, 2020-05-26:
**Describe the bug**

When using the spinbox, `lv_spinbox_decrement` allows to go below the defined `min_range`

**To Reproduce**

Use the example code from https://docs.lvgl.io/v7/en/html/widgets/spinbox.html and change
```c
lv_spinbox_set_range(spinbox, -1000, 90000);
lv_spinbox_set_digit_format(spinbox, 5, 2);
```
to 
```c
lv_spinbox_set_range(spinbox, 0, 255);
lv_spinbox_set_digit_format(spinbox, 3, 0);
```

**Expected behavior**

Stop at 0 and don't go negative

pete-pjb, 2020-05-27:
Hi @luksal,

I also have seen this behaviour, a work around is to set the digit format first then set the range.
I am not sure if the code was intended to work that way... But it worked for me...

Cheers,

Pete

luksal, 2020-05-27:
Thanks @pete-pjb this workaround solves the problem. But I think, that this is still a bug or needs at least some comment in the documentation about this

pete-pjb, 2020-05-27:
No worries @luksal,  I expect it is a bug, perhaps @kisvegabor or @embeddedt can share their thoughts when they have time.

Cheers,

Pete

pete-pjb, 2020-05-27:
Hi @luksal, 

I have been looking into this and it seems using the latest commit from @kisvegabor it appears to be fixed...  It's the commit on the master branch with hash #4f386f1 chart: fix y axis label drawing.

Might be worth a try for you.

Kind Regards,

Pete

kisvegabor, 2020-05-27:
> I have been looking into this and it seems using the latest commit from @kisvegabor it appears to be fixed... It's the commit on the master branch with hash #4f386f1 chart: fix y axis label drawing.

It was related only to charts but probably an earlier commit has fixed it.

I couldn't reproduce the issue with the latest master.

pete-pjb, 2020-06-17:
I assume this is now okay - so I will close it.

closed 2020-06-17 by pete-pjb