lvgl/lvgl #1530: Contributor's License Agreement

kisvegabor, 2020-05-27:
Hi,

Today I've set up a process to ask contributors to sign an [agreement](https://gist.github.com/kisvegabor/c547c6ed1c7637746ceceaa0cb9aa7ac) mainly about you do not reserve any right for your contributions. I think it's quite natural for every contributor but it's better to make it really clear to avoid possible legal issues later. 

A comment will be automatically added to PRs if a committer hasn't signed the agreement yet. You can sign it with 1 clicking on an "I agree" button. (it will be linked to your GitHub account)

amirgon, 2020-05-27:
So.... I'll have to consult with my lawyer now before commiting anything to lvgl?!?  :unamused:
I'm contributing code to MIT licensed project. Why isn't that enough?

kisvegabor, 2020-05-27:
As you know we just established a company for LVGL and has advantages and disadvantages too. 
One of the disadvantages is we need to take more care of the legal parts to avoid any malicious things or misuse of LVGL. Therefore we found reasonable to ask contributors to explicitly tell (sign) what is actually already started in the MIT license. 

So there is nothing new in the agreement and it's the only purpose to minimize risk and legally protect ourselves.

Have font anything specific in it which worries you? 
As always feedback is very welcome.

embeddedt, 2020-05-27:
@amirgon I also had similar concerns about the CLA, but from my understanding, there are no new requirements/restrictions in here; it's just a more explicit way of clarifying that LVGL (the company) is free to do what they want with the code.

puzrin, 2020-05-28:
> So.... I'll have to consult with my lawyer now before commiting anything to lvgl?!? unamused
> I'm contributing code to MIT licensed project. Why isn't that enough?

As far as i remember, that's formal permission to relicense your code under any licence (GPL, custom and so on).

Right now, while project uses MIT, it's not principal. Any commercial company CAN use LVGL in prorietary producs. Situation would be different, if LVGL used GPL and so on. Then CLA could allow to provide commercial services without pain.

@kisvegabor for clarity, it would be nice to tell "Text of this CLA is standard [link] and doesn't has any modifications".

---

But in general, until LVGL uses MIT, nobody cares what CLA says.

BrianPugh, 2020-05-28:
To qualm any fears, this is super standard on any larger MIT project.

kisvegabor, 2020-05-29:
> for clarity, it would be nice to tell "Text of this CLA is standard [link] and doesn't has any modifications".

I've used this [SAP's CLA](https://gist.github.com/CLAassistant/bd1ea8ec8aa0357414e8) without any modifications (except renaming SAP to LVGL).
It is the standard CLA offered by [cla-assistant.io](https://cla-assistant.io/) (this site provides the service which adds a comment to ask for signature)

kisvegabor, 2020-06-27:
There were some negative feedback about using CLA but fortunately, we found a more friendly alternative. It's called DCO (Developers Certificate of Origin). See here: https://en.wikipedia.org/wiki/Developer_Certificate_of_Origin

It's shorter, easier to understand, and widely accepted by huge open-source projects (e.g. Linux).

Usually accepting the DCO happens via `-s` commit flag which adds a sign-off to the commit message (name + email address).
It's quite incontinent but after [some research](https://opensource.stackexchange.com/questions/10013/is-signed-off-by-really-needed-for-a-github-hosted-project), it turned out that it's not really required in our case.  Instead of asking developers for sign-off, a bot will automatically comment on every new Pull request to inform the contributors about the DCO.

It's planned to go live with v7.1 on the 7th of July. 

As always, feedback is welcome! :)

pete-pjb, 2020-06-27:
Sounds much easier :smiley:

turoksama, 2020-07-05:
hello  there，does it also become effective from the past when signed? 
It looks like no conflicts with the MIT license?

kisvegabor, 2020-07-06:
DCO doesn't require a "one time sign". You need to aware if that when you contribute to the project. 

To be sure you are aware of it a bot will comment about it on every new PR.

kisvegabor, 2020-07-07:
I've started the auto-comment bot.

Here is an example PR: https://github.com/lvgl/lvgl/pull/1642

Any feedback on the wording of the comment is very welcome :slightly_smiling_face:

pete-pjb, 2020-07-07:
It looks fine to me... 👍

puzrin, 2020-07-07:
Formatting... line breaks without empty lines are not nice. May be something like below?

> Thank you for raising your pull request. To ensure that all licensing criteria is met all repositories of the LVGL project apply a process called DCO (Developer's Certificate of Origin).
>
>The text of DCO can be read here: [developercertificate.org](https://developercertificate.org)
>
>By contributing to any repositories of the LVGL project you state that your contribution corresponds with the DCO.

Also, it's no quite clear "what should i do" (nothing).

kisvegabor, 2020-07-07:
> Formatting... line breaks without empty lines are not nice. May be something like below?

I agree, it was my inexperience with YAML.

> Also, it's no quite clear "what should i do" (nothing).

Updated

New version:
https://github.com/lvgl/lvgl/pull/1643

embeddedt, 2020-07-07:
The new version looks good to me. :+1:

puzrin, 2020-07-07:
LGTM

kisvegabor, 2020-07-07:
Great!

I close this issue then and start to add the `auto-comment.yml` file to every repo.

Thank you very much for sharing your views on this topic.

closed 2020-07-07 by kisvegabor

kisvegabor, 2020-07-07:
I added the auto-comment config to all repos and updated the docs with a related section:
https://docs.lvgl.io/latest/en/html/contributing/index.html#developer-certification-of-origin-dco

If you have any ideas to make it more complete  (especially the "Accepted licenses and copyright notices" section) feel free to share your ideas.

puzrin, 2020-07-07:
Is auto-comment added only once for every new committer or in each PR? Insant comments would be annoying.

pete-pjb, 2020-07-07:
Hi @kisvegabor,

Last time I updated the docs I was able to edit the .md files in the dev branch, but they are not there for the master branch.
I have copy and pasted the text below with a few minor English corrections, if that's helpful?

Kind Regards,

Pete

Developer Certification of Origin (DCO)
Overview
To ensure that all licensing criteria is met all repositories of the LVGL project apply a process called DCO (Developer's Certificate of Origin).

The text of DCO can be read here: https://developercertificate.org/.

By contributing to any repositories of the LVGL project you state that your contribution corresponds with the DCO.

No further action is required if your contribution fulfills the DCO. If you are not sure about it feel free to ask us in a comment.

Accepted licenses and copyright notices
To make the DCO easier to digest here are some practical guides about specific cases:

Your own work
The most simple case is when the contribution is solely your own work. In this case you can simply send a Pull Request without worrying about any licensing issues.

Use code from online source
If the code you would like to add is based on an article, post or comment on a website (e.g. StackOverflow) the license and/or rules of that site should be followed.

For example in case of StackOverflow a notice like this can be used:

/* The original version of this code-snippet was published on StackOverflow.
 * Post: http://stackoverflow.com/questions/12345
 * Author: http://stackoverflow.com/users/12345/username
 * The following parts of the snippet were changed:
 * - Check this or that
 * - Optimize performance here and there
 */
 ... code snippet here ...
Use MIT licensed code
As LVGL is also MIT licensed other MIT licensed code can be integrated without issues. The MIT license requests to add a copyright notice to the derived work. So you need to copy the original work's license file or it's text to the code you want to add.

Use GPL licensed code
As GPL license is not compatible with MIT license so LVGL can not accept GPL licensed code.

When you get started with LVGL
Even if you're just getting started with LVGL there are plenty of ways to get your feet wet. Most of these options don't even require knowing a single line of code of LVGL.

kisvegabor, 2020-07-07:
@pete-pjb 
The structure of docs has changed a little. Now master only host to built websites and there are branches for the sources (md files).
The branch `latest` is for lvgl `master` and `dev` to lvgl `dev`. To reach the md files the easiest thing to do is selecting a  version in the left, got the page you need, and click the "Edit on GitHub" button on the top of the page.

Does it work for you? If you could you send a PR for `latest`?

 @puzrin 
> Is auto-comment added only once for every new committer or in each PR? Insant comments would be annoying.

It's a simple tool so it comments once in every new PR. 
Probably GitHub actions can do something smarter (filter to users?) but I didn't have time to dig so deep.

pete-pjb, 2020-07-07:
Hi @kisvegabor ,

I have submitted a pull-request to the ```latest``` branch having edited the .md file.  I hope that is what you expected. 😄 

Kind Regards,

Pete

kisvegabor, 2020-07-07:
Perfect, thank you @pete-pjb  :slightly_smiling_face:

puzrin, 2020-07-07:
@kisvegabor Seems [this](https://github.com/actions/starter-workflows/blob/master/automation/greetings.yml#L9) should help to restrict message to first PR only.

I did not tried. Digging docs to learn something useful for my repos.

kisvegabor, 2020-07-08:
Great! Curiously waiting for what you find!

puzrin, 2020-07-15:
https://docs.lvgl.io/latest/en/html/intro/index.html#license

> The only thing you need to do is to add a notice about you are using LVGL in your product. This notice can be placed in the user guide, on your website, on an about screen, or anywhere else where users might see it.

If "need" means "must", this is not compatible with MIT licence. The principle moment is only "if user redistributes lvgl src with their own sources, they must keep info about lvgl licence/authorship"

I'd suggest to drop this paragraph or replace with something else.

kisvegabor, 2020-07-16:
I've checked some sources to figure out what is really expected from the users. Despite of the simplicity of the licenses, there are many interpretations... 

So let's see the resources I used:

http://sourceauditor.com/blog/common-open-source-license-obligations/
> These are followed by MIT style licenses, so named, because these licenses follow a template created by MIT.  In these licenses, the rights are similar to public domain and the only obligation is attribution.  For MIT style licenses, this requirement is often satisfied by leaving the original copyrights that came with the open source software in place and also providing credit to the open source authors in the end user documentation.

https://opensource.stackexchange.com/questions/6081/i-am-using-a-mit-licensed-module-in-a-closed-source-consumer-device-what-are-my

https://softwareengineering.stackexchange.com/questions/178486/what-exactly-does-the-condition-in-the-mit-license-imply

In Firefox: about:license#react-mit

----

I see that one point is really missing from the text in the docs: "in case of open source software a  license and copyright notice of LVGL must be included in the software".

puzrin, 2020-07-16:
See https://opensource.stackexchange.com/a/6144 (1).

I'd suggest to remove your own interpretation, because it's not precise and can be confusing.

But i will not insist.

kisvegabor, 2020-07-16:
If you can show a practical description of how to include an MIT licensed software I happily added it. :slightly_smiling_face: 

I usually found only "add copyright notice". But where? How? How in Open source? How in closed source?

embeddedt, 2020-07-16:
> But where? How? How in Open source? How in closed source?

I think it's usually done by distributing license information either within the software or alongside it. For example, the UI could include a way for people to see the license, or a licensing booklet could be distributed along with the product, or a LICENSE file could be provided along with a binary.

puzrin, 2020-07-16:
IMO project doc should not answer this question, as it should not explain what is C programming language :).

If this it not acceptable - i would add short link where to read deails. To SO thread or something else, but without my own opinion. Because everything possible already been said by more experienced people.

In my own projects i only mention license name in expeceted places (where humans or bots will look for it)

embeddedt, 2020-07-16:
I agree with @puzrin. Let's just mention that the project uses the MIT license. Anyone who's shipping a product with it should be able to go look up the legal ramifications of that.

kisvegabor, 2020-07-20:
Alright, I removed that paragraph: https://docs.lvgl.io/latest/en/html/intro/index.html#license