lvgl/lvgl #1533: Particle system

kisvegabor, 2020-05-28:
Hi,

It would be nice to have a 2D "particle system" widget in LVGL. 
It makes possible to create fancy animation like this:
[image]

In short particle system means emitting images from a source. A lot of parameters can be set such as:
- particle properties
  - min/max lifetime
  - min/max speed at: 0%, x%, y%, 100% of the life time
  - min/max alpha at: 0%, x%, y%, 100% of the life time
  - min/max recolor intense at: 0%, x%, y%, 100% of the life time
  - recolor intense at: 0%, x%, y%, 100% of the life time
  - min/max angle at: 0%, x%, y%, 100% of the life time
  - min/max zoom at: 0%, x%, y%, 100% of the life time
  
- emitter:
  - 2 points creating a line: particles are emitted along the line
  - direction of the particles: maybe via calback to calculate from various properties at any given mooment 
  - min/max born interval (ms): how often will new particles born

Here is a tutorial about how it works in Unity: https://www.youtube.com/watch?v=d7McyujCHc4

Contribution is very welcome in this feature. If you have any ideas or you are interested in contributing into this feature, feel free to comment here!

kisvegabor, 2020-07-08:
@embeddedt 
Why was this topic pinned?

embeddedt, 2020-07-08:
@kisvegabor I thought these "contribution request" topics were intended to remain open, so people would see a list of good first issues to work with. Did I miss something?

kisvegabor, 2020-07-08:
Let's find out how to handle these "contribution requests". At some point, we should close them if there is no interest in them but maybe the default 20 days is too quick.

embeddedt, 2020-07-10:
How about we just close it after the stale bot has sent a message twice?

kisvegabor, 2020-07-10:
Sounds good to me!

yqinic, 2020-08-12:
Hi,

I have some experiences on developing the particle system. I am wondering do you have any specific requirement in mind & how many effects do you want to achieve? e.g. rain, cloud, fireworks...

Best,
Yuan

kisvegabor, 2020-08-13:
@yqinic 
Some simple effects I imagined:
- Bubbles going up
- 2D explosion on click  (particles go into a radial direction, like a simple firework) 
- Small particles "randomly" moving in the background (like dirt in water)
- Particles grow to cover the screen -> Under them the screen changes (not visible) -> Particles shrink -> The new screen appears
- Rain of coins if you win

I think these are quite basic.

A real firework effect with multiple particle systems would be great as well. I think the parameters I listed above it's also possible.

closed 2020-09-11 by stale[bot]