lvgl/lvgl #1534: Add radial gradient and gradient angle

kisvegabor, 2020-05-28:
Hi,

Now LVGL supports horizontal and vertical gradient. 

It'd be great to support radial gradient and angle for linear gradient. 
Here is a tool to generate these type of gradients in CSS gradients: https://angrytools.com/gradient/

Now the [grad_get](https://github.com/lvgl/lvgl/blob/ec5b10142f367ba204bc2bdb8ce867ef9969b34d/src/lv_draw/lv_draw_rect.c#L577) function is used to calculate gradient.

These are tasks to complete:
- [ ] How to describe the more complex gradient types in the styles?
- [ ] Implement linear rotated gradient
- [ ] Implement radial gradient

Contribution is very welcome in this feature. If you have any ideas or you are interested in contributing to this feature, feel free to comment here!

embeddedt, 2020-05-28:
I suggest that a new enum property be added called `lv_gradient_type_t`. It can be set to `LV_GRADIENT_LINEAR` or `LV_GRADIENT_RADIAL`. There'd also need to be a new integer property for the angle, but it can be ignored for radial gradients.

kisvegabor, 2020-05-29:
In CSS there are some properties for radial-gradient too:
- size: farthest-corner (default), closest-side, closest-corner, farthest-side
- position

I'm not sure we need to support all the size types, but position would be nice.
Or we could do it differently by defining:
- position
- radius: somehow make possible to use px and %

What do you think?

embeddedt, 2020-05-29:
> What do you think?

I like the second approach (position and radius).

For the issue of handling pixels vs percentages, I have an idea. Why don't we take advantage of the fact that `lv_coord_t` can be negative, and consider values that are less than 0 to be percentages? (We can `#define LV_COORD_PERCENT(x) (-(x))` to make it more understandable.)

scvetic, 2020-05-29:
Hi,
I thing is best to using two point to define radius or linear vector.

Four month ago for one project I need to implement some radial and linear gauge on embedded platform, for that purpose me need gradient and polygon filling, similar to BeauGauge ActiveX control.
Two month ago I first time see LVGL library and start to test gradient and polygon functions. I'm using idea from cairo library.

[image]
[image]
[image]
[image]

[image]
[image]
[image]

scvetic, 2020-05-29:
kisvegabor,
I can send to you codeon your mail, and maybe you can using some of that.

kisvegabor, 2020-06-01:
That would be awesome!
But could you upload it to GitHub instead of sending it in email?

closed 2020-07-01 by stale[bot]

planevina, 2024-03-21:
I want this....