lvgl/lvgl #1535: lv_win: implement lv_win_add_btn_left()

kisvegabor, 2020-05-28:
Now `lv_win_add_btn` always adds a button to right. However, on smartphones it's quite common to have e.g. back button on the left. 
[image]

The tasks to make it ready:
- [ ] How to store in the button that it should be the left or right?
- [ ] Add lv_win_add_left based on lv_win_add_btn
- [ ] Update [lv_win_realign](https://github.com/lvgl/lvgl/blob/ec5b10142f367ba204bc2bdb8ce867ef9969b34d/src/lv_widgets/lv_win.c#L638) to position the buttons.

Contribution is very welcome in this feature. If you have any ideas or you are interested in contributing to this feature, feel free to comment here!

C47D, 2020-06-09:
Hi,

I've started to work about this in [here](https://github.com/C47D/lvgl/tree/add_win_left_button), there's only the `lv_win_add_btn_left` function declaration for now.

C47D, 2020-06-09:
In `lv_win_realign` we should also track if the button is the first button on the left or in the right side of the header, isn't?

kisvegabor, 2020-06-09:
> I've started to work about this in here, there's only the lv_win_add_btn_left function declaration for now.

Awesome! 

Instead of modifying `lv_btn_ext_t` we can create `lv_winbtn_ext_t` based on `lv_btn_ext_t`. Similarly as `lv_btn_ext_t` is based on `lv_cont_ext_t`.

`lv_winbtn_ext_t` can be local to `lv_win` and contain only an alignment field.

I suggest it because button alignment is not so general for a normal button. 

Regarding the API `lv_win_add_btn(win, img, alignment)` can work IMO.

What do you think?

C47D, 2020-06-09:
It sounds pretty good, I will add another commit with your suggestions.

C47D, 2020-06-09:
Hi @kisvegabor,

I've added `lv_win_btn_ext_t` and defined `lv_win_btn_create` in [here](https://github.com/C47D/lvgl/tree/add_win_left_button), I've used the window example and it's not crashing but I'm not sure if `lv_win_btn_create` is properly done, can you take a look at it?

I've still have to update the `lv_win_realign` and add setter and getter for the `alignment_in_header` data.

kisvegabor, 2020-06-10:
@C47D Let's continue in the PR.

> How many buttons the header can have?

I think we don't need to limit it. If the user adds too many they will overlap, but he needs to take care of it.

C47D, 2020-06-14:
The PR was merged into the dev branch, should we keep this issue open?

embeddedt, 2020-06-14:
It will close itself when 7.1 is released.

kisvegabor, 2020-06-15:
Carlos, could you add few lines about to in the docs?
https://github.com/lvgl/docs/blob/master/v7/en/widgets/win.md

C47D, 2020-06-15:
Sure, will do.

commit a2651650 (2020-06-12): Add lv_win_add_btn_left (#1566)

Closes #1535
closed 2020-06-16 by kisvegabor