lvgl/lvgl #1536: Add variable binding module

kisvegabor, 2020-05-28:
Now if a label display's a variable's value and this value changes the label needs to be updated manually. It's not sure that the label exists at the (it might be created only when its screen is loaded) so it needs to be checked. If the displayed value is composed of more variables things get even more complicated. Another difficulty if multiple objects use the changed value. In this case, you need to update all of them at every place where the variable can change. 

As solution could be developed a variable binding module which could work like this:
- Maintain a linked list of `lv_obj_t *` and variable pointer pairs.  
- When an object is created it can subscribe to a variable. E.g. `lv_bind_listen(label, &motor_rpm)`
- When a variable's value has changed a `lv_bind_update(&motor_rpm)` function needs to be called to notify all subscribed objects.
- Notifying the object means sending an `LV_EVENT_REFRESH`. So this event needs to be handled in the listener object.
- In `lv_obj_del` all instances of the object being deleted should be removed from the linked list.

What do you think about it?

Contribution is very welcome! If you are interested in working on this feature feel free to comment here.

puzrin, 2020-05-30:
> What do you think about it?

I already mentioned before, this is about "reactive programming". You could just find a suitable library.

kisvegabor, 2020-06-01:
> I already mentioned before, this is about "reactive programming". You could just find a suitable library.

Do you know any simple C library for this? I couldn't find any. :frowning_face:

puzrin, 2020-06-01:
I could not quick-find too (only for [cpp](https://github.com/ReactiveX/RxCpp)). But this does not rejects base principles:

- programming patterns are known
- that's a separate abstraction layer, not specific for lvgl (not a part of lvgl).

kisvegabor, 2020-06-02:
> programming patterns are known

I've read the docs of [RxCpp](http://reactivex.io/RxCpp/) and [RxJS](https://medium.com/thecodinghype/reactive-programming-in-javascript-using-rxjs-3b05e186c115). Correct me if I'm wrong but it seems in principle we are on the right path. Working should be updated though to use `Observe` and `Observable`.

These libs are huge and very complex, I think even only the very core features would be enough here to keep it simple and lightweighted.

> that's a separate abstraction layer, not specific for lvgl (not a part of lvgl).

Shouldn't we take advantage of a tightly coupled module which works smoothly with LVGL? E.g. 
- send events to objects directly instead of calling another callback (where the user needs to send an event to the object)
- clear the observer object automatically when it's deleted

Although it would make it possible to observe a variable without an `lv_obj_t`.

C47D, 2020-06-02:
I also think this should be outside LVGL scope and more in the user application layer.

kisvegabor, 2020-06-02:
@puzrin @C47D Could you suggest an API which is LVGL independent still easy to use with LVGL?

C47D, 2020-06-02:
I was thinking on the user having to implement the design pattern, not including it into LVGL, this is just my personal opinion. 

But maybe others have better ideas.

puzrin, 2020-06-02:
> These libs are huge and very complex, I think even only the very core features would be enough here to keep it simple and lightweighted.

> Shouldn't we take advantage of a tightly coupled module which works smoothly with LVGL? E.g.

Problem is, primitive implementaions are very limited and suitable for "hello world" only. I'd prefer lvgl coninues remove existing kludges before start adding new advanced ones :)

---

If JS is not a problem, i'd suggest to inspect different [much beter] libs instead of Rx. https://knockoutjs.com/documentation/introduction.html - that's the most "classic".

Problem with all that kind frameworks - bulk updates of DOM (too much of cascade changes + DOM is slow). So, people invented:

1. Limited refresh rate for DOM.
2. Virtual DOM.

After knockoutjs it worth take a look at "Angular" and "vue.js".

kisvegabor, 2020-06-03:
Of course, I'm okay with doing it correctly but we should the limitation f C language into account. I suppose there is a reason why there in reactive lib for C.

I haven't checked knockoutjs yet, but if you have any ideas on requirements I'd be happy to hear them.

puzrin, 2020-06-03:
I'm not ready to write tech spec for that. It's not trivial.

The only critical point is to keep that separate from lvgl. In this case, even if you do something "strange", that will not affect main project. Or someone could do alternate implementation if not agree with your opinion (and can use CPP if desired).

barbiani, 2020-06-03:
How about having a structure that describes the variable, callback pointers
to change its value and a collection of objects depending on it?

puzrin, 2020-06-03:
@kisvegabor https://en.wikipedia.org/wiki/Model%E2%80%93view%E2%80%93viewmodel - here is a good point to start dig about MVVM, with references to all mentioned frameworks.

kisvegabor, 2020-06-04:
@barbiani 
> How about having a structure that describes the variable, callback pointers
to change its value and a collection of objects depending on it?

People said above we should separate it from LVGL and create an independent module. I"m fully convinced though, because it'd be much easier to use an LVGL specific module.

@puzrin 
Thanks!
Finally, I understood the core ideas MVP, MVC, and MVVM. So binding exists only in MVVM but the whole concept was [criticized](https://en.wikipedia.org/wiki/Model%E2%80%93view%E2%80%93viewmodel#Criticism) by its author. 

The question I see:  How much would it help if we create a feature-rich C binding manager compared to a "one-day-job" binder which can notify the LVGL objects if a value changed? Is it really worth the effort to dig so deep? What and where could we win with a general binder?

puzrin, 2020-06-04:
> The question I see: How much would it help if we create a feature-rich C binding manager compared to a "one-day-job" binder which can notify the LVGL objects if a value changed? Is it really worth the effort to dig so deep? What and where could we win with a general binder?

At current moment i think primitive binder will be about "useless" and full featured will be not rationale spend of time. But if you do any as separate module, that will be safe for main project, at least.

kisvegabor, 2020-06-04:
> At current moment i think primitive binder will be about "useless" 

IMO depends on what tasks can be solved even with a primitive binder. Considering the use case I mentioned in the [initial comment](https://github.com/lvgl/lvgl/issues/1536#issue-626395436), even a primitive binder could be quite useful. Can you mention one ore use case(s) that can not be solved with an outlined binder but could be solved with a more advanced one?

> But if you do any as separate module, that will be safe for main project, at least.

I see only one issue with the separate module: it introduces a dependency, for example, demos, tutorials, etc. E.g. "To run this example you also need to clone the binding module and add it to your project".

puzrin, 2020-06-04:
> Can you mention one ore use case(s) that can not be solved with an outlined binder but could be solved with a more advanced one?

I'd say a bit different. Primitive implemenation already can be done via ETL and so on. Thing you propose is very small part of MVVM in context of "convenient use".

> I see only one issue with the separate module: it introduces a dependency, for example, demos, tutorials, etc. E.g. "To run this example you also need to clone the binding module and add it to your project".

Yes, lack of deps manager is an endless pain. Consider if you could use PlatformIO as primary development IDE (it exists for eclipse too). Or may be https://conan.io/ (i did not used it).

embeddedt, 2020-06-04:
> Consider if you could use PlatformIO as primary development IDE (it exists for eclipse too). Or may be https://conan.io/ (i did not used it).

One of the perks of LVGL is the ability to quickly and easily add it to any project/IDE. For that reason, I think we should avoid tailoring core components to specific IDEs.

puzrin, 2020-06-04:
> One of the perks of LVGL is the ability to quickly and easily add it to any project/IDE. For that reason, I think we should avoid tailoring core components to specific IDEs.

I do not propose to break that. I propose to use advantages "when available".

kisvegabor, 2020-06-04:
> I'd say a bit different. Primitive implemenation already can be done via ETL and so on. Thing you propose is very small part of MVVM in context of "convenient use".

Probably it's only my ignorance but still doesn't know a real-life example where the proposed solution wouldn't work but a more advanced do.

> I do not propose to break that. I propose to use advantages "when available". Users still can continue use git submodules. But developpers could use more simple approach for coding & examples.

I idea is to keep the simplicity on as many platforms as possible. And moving a core feature outside of LVGL would be difficult to manage on many platforms.

puzrin, 2020-06-04:
> Probably it's only my ignorance but still doesn't know a real-life example where the proposed solution wouldn't work but a more advanced do.

It's for single elements only. Will not help for mapping arrays to list views, auotupdate from filtered and calculated arrays.

> I idea is to keep the simplicity on as many platforms as possible. And moving a core feature outside of LVGL would be difficult to manage on many platforms.

We have different opinion about what is "core feature". JPEG/PNG still stays standalone, and nobody died :). IMO feature can became core when one of condition satisfied:

- project does not work without it.
- it's very stable and popular.

I could understand if you do it separate, and join after reach success. To keep existing api stable.

kisvegabor, 2020-06-05:
> It's for single elements only. Will not help for mapping arrays to list views, auotupdate from filtered and calculated arrays.

It really won't be supported, but I don't think there is a big demand for it.

> I could understand if you do it separate, and join after reach success.

I see your point because LVGL itself can wok without it, e.g. it doesn't required for the widgets or other modules. So it's not that tightly coupled as the animations for example. 

However, I see this binding as a new (simpler) way to use LVGL which should be demonstrated via examples and tutorials and shouldn't be more difficult to add than enabling it in `lv_conf.h`.

puzrin, 2020-06-05:
> It really won't be supported, but I don't think there is a big demand for it.

I have opposite real world experience with JS. And this example is just the most obvious. Modern frameworks have more features to simplify programming.

kisvegabor, 2020-06-08:
Alright. Then let's collect what would be still required.
I've checked Knockout.JS and found binding to styles and classes pretty useful.
https://knockoutjs.com/documentation/style-binding.html

Do have something concrete in mind to start writing a list from "nice to have" features.

kisvegabor, 2020-06-08:
I've further examined ko.js and got a better view of what is called a state-of-the-art binding. Thank you @puzrin.

Obviously, in C we have less room to make the binding module so easy to use but still can do great things.

Just thinking aloud...

So the first thing is to create observables (i.e. things whose value change should trigger changes in the UI). We can use only a fixed set of observer types. For first these 3 types seems obvious:
- int (a general `typedef int32_t lv_bind_int_t` should work)
- string (simple `const char *`)
- boolean

There should be functions to manipulate observable values:
```c
lv_bind_observable_t * obs_int;  //global
...
obs_int = lv_bind_create_observable_int(&my_int);
lv_bind_set_observable_value_int(obs_int, 80);  //Notify every object/style who is interested in this observable
```
It'd be nice to create observables from object attributes, e.g. object state, text area text etc. E.g. 
```c
lv_bind_create_observable_attr(btn1, LV_BIND_OBSERVABLE_TYPE_ATTR_STATE);

//or
lv_bind_create_observable_attr(textarea1, LV_BIND_OBSERVABLE_TYPE_ATTR_TEXT);
```
When the state or text changes the object can call a `lv_bind_refresh()` function.

In ko.js the observables are bound to [HTML elements](https://knockoutjs.com/documentation/visible-binding.html), in our case these are `lv_obj`s. If the observable changes it affacts the UI element somehow. So to create a binding:
```c
lv_bind_<type>_to_obj(obs, obj, <some other parameters>); //bind to an object
```
Binding can affect HTML `style` and `class` usage too so these might be also reasonable:
```c
lv_bind_<type>_to_style(obs, &style, <...>);
lv_bind_<type>_to_style_list(obs, obj, part, <...>);
```

My initial idea was to simply trigger an `LV_EVENT_REFRESH` and the user will read the relevant variables there and apply required changes. However, there are typical actions to do when typical conditions are met. Due to the flexibility of JS in ko.js you can apply any condition but in C we are limited. So the gist is if a condition is true, perform an action on the bound object/style/etc. 
Here are some examples
```c
//change the background to red in a style if a value is >= 25
obs_int = lv_bind_create_observable_int(&my_int);
lv_bind_t * bind =  lv_bind_int_to_style(obs_int, &style, lv_bind_if_eq_gt, 25);
lv_bind_add_action_style(bind, LV_STATE_DEAFULT, LV_STYLE_BG_COLOR, LV_COLOR_RED); //perform this on `style`

//use a dynamic value instead of 25 and custom condition tester (reuse `obs_int`)
lv_bind_t * bind =  lv_bind_int_to_style_cb(obs_int, &style, my_condition_test, get_int);
lv_bind_add_action_style(bind, LV_STATE_DEAFULT, LV_STYLE_BG_COLOR, LV_COLOR_RED);

//or change the background to red in style if a string become "hello":
obs_str = lv_bind_create_observable_int(my_buf);
lv_bind_t * bind =  lv_bind_str_to_style(obs_str, &style, lv_bind_if_str_eq, "hello");
lv_bind_add_action_style(bind, LV_STATE_DEAFULT, LV_STYLE_BG_COLOR, LV_COLOR_RED);

//bind a bool variable to an object
obs_bool = lv_bind_create_observable_bool(&my_bool);
lv_bind_t * bind =  lv_bind_bool_to_obj(obs_bool, obj, lv_bind_if_true);  
lv_bind_add_action_attr_bool(bind, LV_BIND_OBJ_SET_HIDDEN, false); //unhide `obj`
lv_bind_add_action_img_src(bind, &my_img);  //multiple actions
lv_bind_add_action_local_style(bind, LV_STLYE_BG_COLOR, LV_IMG_PART_MAIN, LV_STATE_DEAFULT, LV_COLOR_RED);

//bind object attribute to an other object 
obs_textarea_text = lv_bind_add_observable_attr(textarea1, LV_BIND_OBSERVABLE_TYPE_ATTR_TEXT); //Watch the text of the text area
lv_bind_t * bind =  lv_bind_str_to_obj(obs_textarea_text, next_btn, lv_bind_if_str_eq, "1234");
lv_bind_add_action_clear_state(bind, LV_STATE_DISABLED);  //Enable the next button if the PIN is ok

// Add  a warnings class 
obs_int = lv_bind_create_observable_int(&my_int);
lv_bind_t * bind1 =  lv_bind_int_to_style_list(obs_int, title_label, LV_LABEL_PART_MAIN, lv_bind_if_gt, 1000);
lv_bind_add_action_style_add(bind, &style_warning); //Add `style_warings` to `title_label`

lv_bind_t * bind2 =  lv_bind_int_to_style_list(obs_int, title_label, LV_LABEL_PART_MAIN, lv_bind_if_lt_eq, 1000);
lv_bind_add_action_style_remove(bind, &style_warning); //Remove `style_warings` to `title_label`
```

So what can be natively modified by the binding if a condition is met?
- obj
  - hidden
  - drag
  - state
  - text (if applicable)
  - image src (if applicabe)
  - refresh event to remain generic
  - local styles
- style
  - any properties with a given state
- style lists
  - add/remove styles

I'm very curious to hear your opinion!

puzrin, 2020-06-08:
Please, understand me right, i usually participate in 3 cases:

- When i really need feature (good fonts, for example).
- When it costs me nothing.
- When i wish to learn something.

In this case, it cost me nothing to warn you about huge gap with modern frameworks. But i'm not ready to participate in design of such non trivial thing. Because i see no reasons to restrict myself with plain C, and because that's not critical for my current projects (or something similar can be done via ETL).

kisvegabor, 2020-06-09:
Okay

embeddedt, 2020-06-09:
I'd suggest making actions more generic. Things like `lv_bind_add_action_img_src` and `lv_bind_add_action_attr_bool` seem too specific to me. I'd propose a set of more generic APIs like:

```c
typedef void (*lv_bind_obj_attr_ptr_cb_t)(lv_obj_t * obj, void *x);
typedef void (*lv_bind_obj_attr_bool_cb_t)(lv_obj_t * obj, bool x);
typedef void (*lv_bind_obj_attr_int32_cb_t)(lv_obj_t * obj, int32_t x);
typedef void (*lv_bind_obj_attr_int_cb_t)(lv_obj_t * obj, int x);

void lv_bind_add_action_obj_attr_bool(bind, obj, lv_obj_set_hidden, false);
void lv_bind_add_action_obj_attr_int32(gauge, obj, lv_gauge_set_value, 32);
```

The idea is that we design the binding API to fit existing LVGL widget functions instead of having to duplicate them elsewhere. It also means that the binding automatically works with everything in the library. We can probably even autogenerate the code for some of these using macros.

kisvegabor, 2020-06-09:
> I'd suggest making actions more generic. Things like lv_bind_add_action_img_src and lv_bind_add_action_attr_bool seem too specific to me. I'd propose a set of more generic APIs like:

Great idea! Let's continue this way.

So reacting on observable's change seems okay. 
However settings the value of observables is still too specific. E.g. this:
```c
obs_textarea_text = lv_bind_add_observable_attr(textarea1, LV_BIND_OBSERVABLE_TYPE_ATTR_TEXT); 
```
Following your idea, we could add one (or a list) os observables to the objects and make them automatically set the observable.

```c
obs_textarea_text = lv_bind_create_observable_obj_str(lv_textarea_get_text);
lv_obj_add_observable(textarea1, obs_textarea_text);
```

`obs_textarea_text` says alive if `textarea1` is deleted leave the subscribed actions valid.

C47D, 2020-06-14:
Is this what you are trying to acchieve? https://youtu.be/BJ6oXZlA7X8?t=338

kisvegabor, 2020-06-15:
@C47D Yes, but even more:
- if you simply write a variable it will update the bound UI element
- conditions can be added. E.g. add a warning style if `x > 500`
- use unrelated variables together. E.g. if `x > 500` set a labels text to `str1` else `str2`

It would be a missing layer in LVGL's UI development process:
- drivers: just set/get variables here, e.g read the temperature and store it in `temp1`, set PWM value from `out_pwm1`, etc
- binding: automatically notify the UI if a variable changes
- gui: display data directly from variables or based on related conditions and set variables on press/release etc

I'm still trying to clean up the API but will come back soon with my proposal.

kisvegabor, 2020-06-23:
I cleaned up the API and added some examples to see we can really do some stuff with it. 
Please let me know if you find this approach useful and usable. 

### Create observable variables

There are 3 basic types: bool, int, string.

Simply use global variables and call `lv_bind_sync` if they were changed manually.

```c
bool running;
char user_name[64];
lv_bind_int_t cnt;

cnt++;

lv_bind_sync(&cnt);
```

### Directly bind a variable to a property
Keep a property (e.g. label text) synchronized with a variable. If one changes the other will change too.

I was thinking a lot about using the already existing `set/get` functions (as @embeddedt  suggested) to tell which property to bind but found some issues with this approach:
- Both the related set and get function needs to be known to read or write the given property.
- There are more API functions to change one property. E.g. `lv_textarea`: `set_text`, `add_text`, `add_char`, `del_char`.

So defining some basic properties seems more reasonable to me. 
 
```c
lv_bind_t * bind = lv_bind_create(obj);
/*Random examples to show the API*/
lv_bind_prop_bool(bind, LV_BTN_PROP_CHECKED, &running);
lv_bind_prop_str(bind, LV_TEXTAREA_PROP_TEXT, str, sizeof(str));
lv_bind_prop_str_fmt_int(bind, LV_TEXTAREA_PROP_TEXT, "Counter: %d", &cnt);
lv_bind_prop_str_fmt_str(bind, LV_TEXTAREA_PROP_TEXT, "Hi, : %s", user_name);
lv_bind_prop_str_fmt_bool(bind, LV_TEXTAREA_PROP_TEXT, "State is %s", &running, "On", "Off");
```

If a change in a value is announced with `lv_bind_sync` all bindings are checked if they are related to this variable. If so the related property is changed (e.g. via new signal)

If a property is changed via LVGL's API (e.g. `lv_textarea_set_text`) LVGL checks if there is a variable who is interested in this object's this property. If it founds any, it updates the variable and calls `lv_bind_sync`.

### Add conditions

Add conditions to a binding. 
```c
lv_bind_add_condition_int(bind, lv_bind_if_int_eq, &cnt, 32);
lv_bind_add_condition_str(bind, lv_bind_if_str_eq, user_name, "Jane");
lv_bind_add_condition_str(bind, lv_bind_if_str_eq, user_name, other_string);
lv_bind_add_condition_bool(bind, lv_bind_if_bool_eq, &running, true);
lv_bind_add_condition_custom(bind, custom_func, &p1, &p2);
```
On `lv_bind_sync` all conditions are also checked whether they are interested in this variable. 

Add conditional bindings to the true or false (else) branch of the condition.
```c
lv_bind_prop_bool_if_true(bind, LV_BTN_PROP_CHECKED, true);
lv_bind_prop_bool_if_false(bind, LV_BTN_PROP_CHECKED, false);
lv_bind_prop_bool_if(bind, LV_BTN_PROP_CHECKED, true, false); //In one function

lv_bind_prop_int_if_true(bind, LV_GAUGE_PROP_RANGE_MAX, 200);
lv_bind_prop_int_if_false(bind, LV_GAUGE_PROP_RANGE_MAX, 100);
lv_bind_prop_int_if(bind, LV_GAUGE_PROP_RANGE_MAX, 200, 100);

lv_bind_prop_str_if_true(bind, LV_LABEL_PROP_TEXT,  "Warning");
lv_bind_prop_str_if_false(bind, LV_LABEL_PROP_TEXT,  "All good");
lv_bind_prop_str_if(bind, LV_LABEL_PROP_TEXT,  "Warning", "All good");

lv_bind_style_add_if_true(bind, &style_warn);
lv_bind_style_remove_if_false(bind, &style_warn);
lv_bind_style_if(bind, &style_warn);  //Add if true, remove is false
```

Set OR or AND logic between the conditions
```c
lv_bind_set_condition_logic(bind, LV_BIND_LOGIC_OR/AND);
```

### Attach a custom listener to variable

It could be required to be notified when a variable's value changes. E.g. if the variable is bound to a slider a PWM value needs to be changed accordingly.

```c
lv_bind_notification_t * listener = lv_bind_create_notification(&var1, func);
``` 

### Example

#### Mirror an object with an other
```c
char str[64];
lv_bind_t * bind_ta = lv_bind_create(textarea);
lv_bind_prop_str(bind_ta, LV_TEXTAREA_PROP_TEXT,  str, 64);

lv_bind_t * bind_label = lv_bind_create(label);
lv_bind_prop_str(bind_label, LV_LABEL_PROP_TEXT, str, 64);
```

#### If PIN is correct activate a next button
```c
char pin_expected[4] = {"1234"};
char pin_now[4];

lv_bind_t * bind_ta = lv_bind_create(textarea);
lv_bind_prop_str(bind_ta, LV_TEXTAREA_PROP_TEXT,  pin_act, 4);

lv_bind_t * bind_btn = lv_bind_create(btn);
lv_bind_add_condition_str(bind_btn, lv_bind_if_str_eq, pin_act, pin_expected);
lv_bind_prop_bool_if(bind, LV_BTN_PROP_DISABLED, false, true);
```

#### Set motor RPM with a slider and warning for high RPM

```c
lv_bind_int_t rpm1;

lv_bind_t * bind_slider = lv_bind_create(slider);
lv_bind_prop_int(bind_slider, LV_SLIDER_PROP_VALUE,  &rpm1); // bind anyway
lv_bind_add_condition_int(bind_slider, lv_bind_if_int_gt, &rpm, 500);
lv_bind_style_if(bind_slider, &style_warn);

lv_bind_t * bind_label = lv_bind_create(bind_label);
lv_bind_add_condition_int(bind_label, lv_bind_if_int_gt, &rpm1, 500);
lv_bind_add_condition_int(bind_label, lv_bind_if_int_gt, &rpm2, 800);
lv_bind_add_condition_int(bind_label, lv_bind_if_int_gt, &rpm3, 900);
lv_bind_set_condition_logic(bind_label, LV_BIND_LOGIC_OR);
lv_bind_prop_bool_if(bind_label LV_OBJ_PROP_HIDDEN, false, true);
```

embeddedt, 2020-06-23:
This would certainly be an improvement over the current solution of manually updating objects, but still feels very cumbersome and intuitive compared to React. There's probably nothing we can do about it because we're limited to what C syntax and a relatively weak embedded processor are capable of, but I really wish it was possible to do something like this:

```jsx
<Button disabled={strcmp(currentPin, "1234")}>Next</Button>
```

I find that syntax far more readable.

The condition system is also somewhat awkward compared to being able to write out the conditions the way one normally would in C. At a high level, I feel like using callbacks and letting the user implement the conditions directly would be a better approach. It's a few more lines of code but seems much more flexible, and also more readable than `lv_bind_add_condition_int(bind_slider, lv_bind_if_int_gt, &rpm, 500);`.

```c
static void lv_slider_val_update_cb(lv_bind_int_t * rpm1)
{
    lv_obj_t * slider = lv_bind_get_obj(rpm1);
    if(lv_bind_int_get_value(rpm1) > 500)
        lv_obj_add_style(slider, LV_OBJ_PART_MAIN, &slider_warn); /* only added if it doesn't exist */
    else
        lv_obj_remove_style(slider, LV_OBJ_PART_MAIN, &slider_warn); /* only removed if it does exist */
}
lv_bind_int_t rpm1;

lv_bind_t * bind_slider = lv_bind_create(slider);
lv_bind_prop_int(bind_slider, LV_SLIDER_PROP_VALUE,  &rpm1); // bind anyway
lv_bind_int_add_update_cb(&rpm1, lv_slider_val_update_cb);
```

kisvegabor, 2020-06-23:
It'd be really great to write functions directly in conditions and "actions" but I have no idea how to do it in C. :slightly_frowning_face: 

`lv_bind_int_add_update_cb(&rpm1, lv_slider_val_update_cb);`
should be 
`lv_bind_int_add_update_cb(bind_slider, lv_slider_val_update_cb);`
isn't it?

I'm thinking about how your suggestion is different from the original "send LV_EVENT_REFRESH if a variable change" approach? The only difference I see is the event function is replaced with another function.

puzrin, 2020-06-24:
- Seems float binding missed
- Inventing language for logic is waste of time for this case. Logic should stay in user-provided function.
- Formatters of number => strings are very limited. Create chained observables instead: `obseravable_str = transform(observable_int, transformer_fn)` and let user decide how to format. You can provide conversion helpers, if you wish.

> It'd be really great to write functions directly in conditions and "actions" but I have no idea how to do it in C.

Consider progressive approach:

- Optional CPP extension with nice syntax, which should work for many people (because they have modern compilers).
- Cut features for legacy case.

IMO you fall into writing code "for compilers" instead of writing it "for users".

embeddedt, 2020-06-24:
I like the idea of a zero-cost C++ extension that adds the variable binding module. It would make the code syntax a lot cleaner. We'd of course leave the library itself as plain C.

If we did opt to take this route, I suggest that we avoid using any features from the C++ STL, and just make use of the basic constructs like classes and lambdas. That way, C++ won't add much to the code size.

kisvegabor, 2020-06-24:
@puzrin 

> Seems float binding missed

`float` is not used in LVGL for performance and memory reasons. If you use `float` in one place the coed will be significantly larger. (don't know exact measures now)

> Inventing language for logic is waste of time for this case. Logic should stay in user-provided function.

Okay, it aligns with the approach of @embeddedt 
 
> Formatters of number => strings are very limited. Create chained observables instead: obseravable_str = transform(observable_int, transformer_fn) and let user decide how to format. You can provide conversion helpers, if you wish.

Sounds good.

> Consider "progressive" approach, as in web:

Sounds reasonable but can't see which feature can be added to CPP right now.

> IMO you fall into writing code "for compilers" instead of writing it "for users".

I feel that too :laughing: 

----

### Reiterating

The most important goal is to separate Model (hardware, back-end) from View (LVGL widgets).

I see 3 ways of information flow:
1. **The UI changes and variable needs to be updated**
   - E.g. export the slider's value to a variable
   - The user needs to be notified on value change. (E.g. to update PWM)
   - Something like this works with `LV_EVENT_VALUE_CHANGE`
2. **The UI changes and another UI element needs to updated.**
   - E.g.When the correct PIN is entered, activate a button
   - Possible via events but needs to know the pointer of the related object (which could be even deleted)
3. **A variable is updated and the UI needs to be changed**  
   - E.g. increment a labels value on button (on I/O) press, light a LED widget on I/O change, etc.
   - There is no mechanism for it now.

As I see we are getting closer again to original concept to use the events. Here is my (next) idea:

```c

lv_bind_int_t cnt;
const char * name[128] = {"Counting"}; 

//Create bindings. They are deleted automatically when the object is deleted
lv_bind_t * bind1 = lv_bind_create(label); 
lv_bind_add_var(bind1, &cnt);  //tell that `label` is interested in `cnt`
lv_bind_add_var(bind1, name);
lv_bind_add_action_str_cb(bind1, lv_label_set_text, formatter);  //Called when any bound varaible changes

lv_bind_t * bind2 = lv_bind_create(box); 
lv_bind_add_var(bind2, &cnt); 

//Custom notification for any varaible
lv_bind_watch_t * notify1 = lv_bind_create_watch(&var);
lv_bind_watch_add_listener(notify1, listener_cb);
lv_bind_watch_set_user_data(notify1, 55);

//Update the variables and notify the subscribed objects
cant  = 5;
lv_bind_sync(&cnt);  //Sends LV_EVENT_REFRESHthe the box and update the label automatiaclly

strcpy(name, "John");
lv_bind_sync(&var); //Update the label's text automatically
...

//Called when `name` or `cnt` changes to create a text for `lv_label_set_text`
const char * formatter(lv_binding_t * bind)
{
   lv_bind_int_t * x = lv_bind_get_var(bind, 0);  //Get the 1st variable of the binding (cnt)
   const char * s = lv_bind_get_var(bind, 1);  //Get the 2st variable of the binding (name)
    static buf[256];
    lv_snprintf(buf, 256, "%s: %d", s, x);
}

// Called when `cnt` changes
void box_event(lv_obj_t * box, lv_event_t e) 
{
    if(e == LV_EVENT_REFERESH) {
          lv_bnd_t * bind = lv_bind_act();  //The triggering binding
          lv_bind_int_t * x = lv_bind_get_var(bind, 0);  //Get the 1st variable of the binding (cnt)
          if(*x > 100) lv_obj_add_style(box, &style_warn);
          else  lv_obj_remove_style(box, &style_warn);
    }
}

// Called when the user click a button
void btn_event(lv_obj_t * btn, lv_event_t e) 
{
    if(e == LV_EVENT_CLICKED) {
          cnt++;
          lv_bind_sync(&cnt);  //Announc the change of `cnt`
    }
}

//Custom listener for a variable. Called when `cnt` changes
void listener_cb (lv_bind_watch_t * n)
{
  if(n->user_data == 55) {
    lv_bind_int_t * v = n->var;
    if(*v > 100) gpio_set(POARTA, 12);
    else gpio_clear(POARTA, 12);
}
```

As @puzrin suggested we can have predefined for matters like below. 
```c
//lv_bind_add_action_str_cb
lv_bind_format_str_first();                 //Simply get the first varaible of the binding
lv_bind_format_str_last();                 //Simply get the last varaible of the binding
lv_bind_format_str_first/last_dec/hex(); 
lv_bind_format_str_concat(); // Concatenate to a string in the binding's user data

//lv_bind_add_action_int_cb
lv_bind_format_int_first();                 //Simply get the first varaible of the binding
lv_bind_format_int_last();                 //Simply get the last varaible of the binding
lv_bind_format_int_add/mult/sub/div();

//lv_bind_add_action_bool_cb
lv_bind_format_bool_first();                 //Simply get the first varaible of the binding
lv_bind_format_bool_last();                 //Simply get the last varaible of the binding
lv_bind_format_bool_first/last_invert(); //Invert he last/first varaible.
```

More bindings can be created fo an object to make the use of predefined formatter easier.
```c
bool pin_ok;
lv_bind_int_t remaining_tries;

lv_bind_t * bind1 = lv_bind_create(next_btn);
lv_bind_add_var(bind1, &pin_ok);
lv_bind_add_action_bool_cb(bind1, lv_btn_set_disabled, lv_bind_format_bool_first_invert); 

lv_bind_t * bind2 = lv_bind_create(next_btn);
lv_bind_add_var(bind2, &remaining_tries);

void box_event(lv_obj_t * box, lv_event_t e) 
{
    if(e == LV_EVENT_REFERESH) {
          if(remaining_tries < 2) lv_obj_add_style(box, &style_warn);
          else  lv_obj_remove_style(box, &style_warn);
    }
}

```

puzrin, 2020-06-24:
> > Consider "progressive" approach, as in web:
> 
> Sounds reasonable but can't see which feature can be added to CPP right now.

I would add 2 things:

- Some observables (+ calculated ones as dependencies), to sync interface updates.
- Facades for widgets. To use `obj.style_add(A).style_remove(B)` notation. That should not cost too much.

> `float` is not used in LVGL for performance and memory reasons. If you use `float` in one place the coed will be significantly larger. (don't know exact measures now)

As i said before, it's not good idea to think about compilers and hardware, when design things for users. Design "ideal" api first, then cut features second. Floats are convenient to think in business logic domain. If you use M4 (with FPU), it has enough flash memory to not worry about.

Probably, you could get issues, caused by `double` promotion. That's solved by [appropriate warning flag](https://github.com/puzrin/dispenser/blob/master/platformio.ini#L51).

> If we did opt to take this route, I suggest that we avoid using any features from the C++ STL, and just make use of the basic constructs like classes and lambdas. That way, C++ won't add much to the code size.

See https://github.com/ETLCPP/etl, it brings convenience of STL at low cost. Without it working with strings for gui is nightmare :). It also provides correct `float` formatters.

embeddedt, 2020-06-24:
None of the built-in widgets use `float` at the moment, so including it in the binding wouldn't be particularly useful.

puzrin, 2020-06-24:
> None of the built-in widgets use `float` at the moment, so including it in the binding wouldn't be particularly useful.

lvgl core itself does not need floats, because it's not responsible for formatting. But observables api is for different abstraction layer (convenient data binding/presentation), and floats may be useful there.

In other words, nobody forces you use floats, but "bindings api" without those would be too restricted.

Anyway, if you decide to give cpp a chance, i think concrete types will be hidden behind cpp templates.

kisvegabor, 2020-06-25:
I suggest not thinking too much about floats now. We can add them later at any time. 

But what about the proposed API/working mechanism? :slightly_smiling_face:  
Sorry for sending the umpteen very log post... it must be a pain to read them :pray:

kisvegabor, 2020-06-25:
Okay, no problem. It could be still useful for people coming  with C background (the majority of LVGL users)

puzrin, 2020-06-25:
> Okay, no problem. It could be still useful for people coming with C background (the majority of LVGL users)

That's a correct statement, but not describing all situaion. I guess, the same [most of] users will have no problems with CPP [much more convenient] api. From my point of view, you have much better opportunities to improve lvgl convenience.

Current choice looks like this:

1. Good api for 90% (?) of users.
2. Bad api for 100% of users.

In first case, 10% may be disappointed, in second case, 90% may be disappointed :).

embeddedt, 2020-06-25:
I would agree with @puzrin here (not for the entire library, just for the binding system itself). C syntax quickly becomes unmanageable when there are this many callbacks and state variables to keep track of.

Here's an alternative paradigm I was thinking of last night that avoids needing to change/add much code. When an object is created (or redrawn for the first time) it sends `LV_EVENT_REFRESH`. The user should call `lv_[objtype]_set_[value]` in order to retrieve the value from a variable.

`LV_EVENT_REFRESH` should become a propagating event, i.e. it is sent to children of the object as well (I'm not sure if it should also be sent to parents?). This means that when someone wants to update their UI, they can simply call `LV_EVENT_REFRESH`, and all objects will automatically pick up new values. (To make sure that CPU cycles aren't wasted, users should check that the value has in fact changed.)

This means that all of the logic for an object to set/remove options and retrieve a value is put in `LV_EVENT_REFRESH`, meaning that there is one source of information about the object's current state.

Here's an example of what this might look like in practice:

```c
uint32_t current_temperature = 0;
lv_obj_t * temperature_gauge;

static void temperature_interrupt(void)
{
    lv_obj_refresh(temperature_gauge); /* The gauge can now update whatever state it needs to from the global variables */
}

static void gauge_event_cb(lv_obj_t * gauge, lv_event_t event)
{
    if(event == LV_EVENT_REFRESH)
    {
        lv_gauge_set_value(gauge, current_temperature);
    } else if(event == LV_EVENT_VALUE_CHANGED) { /* Not used by the gauge, just here as an example */
        /* Do something with the value */
    }
}
int main(void)
{
    config_sensor();
    lv_init();
    lv_obj_t * gauge = lv_gauge_create(lv_scr_act(), NULL);
    /* Any styles/flags that don't need to change on refresh could be added here */
    lv_obj_set_event_cb(gauge, gauge_event_cb);
}
```

What do you think about that? One could then build a more convenient C++ binding on top of this to prevent namespace pollution and/or automatically sends `LV_EVENT_REFRESH` when the variable changes.

From what I can tell it solves all three of the problems:

* UI changes, variable needs to update - `LV_EVENT_VALUE_CHANGED`
* UI changes, UI needs to update - can be solved at the moment with pointers to other objects (or by sending `LV_EVENT_REFRESH` to the parent and letting the parent update another UI element).
* variable changes, UI needs to update - `LV_EVENT_REFRESH` (with an optional, more intelligent C++ binding)

kisvegabor, 2020-06-25:
@embeddedt 

I think there are 2 minor typos in the example code:
- `current_temperature` should be set in `temperature_interrupt`
- `lv_obj_t * gauge = lv_gauge_create(lv_scr_act(), NULL);` should be `temperature_gauge = lv_gauge_create(lv_scr_act(), NULL);`

Are these true? (Just to be sure I don't misunderstand anything)

I see some issues with your suggestion
- the user still needs to know whether the object exists or not
- the user needs to know which objects are interested in a given variable (and list them every time the variable changes)
- the user needs to know the pointer of all related objects. 
- the user can't react if a variable is changed by the UI. E.g. if the `act_speed` variable is changed by a slider a PWM needs to be updated.

If we bind objects to variables we can simply "invalidate" the variable and all interesting objects will be notified if exist. On calling `lv_bind_create(obj);` a `VALUE_CHANGE` event can be sent to allow to set the correct initial state.

If you find the `formatter` stuff too complicated we can drop it. IMO the core idea is to 
- connect vars to object for easy notification. 
- notify the user if a variable changes.

```c
//Called when `obj` is created
lv_bind_t * bind = lv_bind_create(obj);
lv_bind_add_var(bind, &var);

//Called in lower level indepentetly from UI
lv_bind_watch_t * watch = lv_bind_watch_craete(&var);
lv_watch_set_cb(watch, custom_func); //Called when `var` changes

var = 10;
lv_bind_sync(&var); //Notfiy `obj` automatically and call `custom_func`
```

embeddedt, 2020-06-25:
> Are these true? (Just to be sure I don't misunderstand anything)

Yes; sorry about that. I typed the code in a rush. :slightly_smiling_face: 

> the user needs to know which objects are interested in a given variable (and list them every time the variable changes)

They can just refresh a parent container and all child objects can check whether the variables have changed. This is similar to how React works on the web. Comparing the current and previous values shouldn't be too expensive.

> the user needs to know the pointer of all related objects.

That could be skipped if they refreshed the parent container, as every object would get an `LV_EVENT_REFRESH` automatically (even if there isn't a pointer referencing it).

> the user can't react if a variable is changed by the UI. 

I don't quite understand this; isn't that what `LV_EVENT_VALUE_CHANGED` is for?

kisvegabor, 2020-06-25:
> Yes; sorry about that. I typed the code in a rush. slightly_smiling_face

No problem. :slightly_smiling_face: 

> They can just refresh a parent container and all child objects can check whether the variables have changed. This is similar to how React works on the web. Comparing the current and previous values shouldn't be too expensive.

Practically it means refreshing the whole display (including the active screen, top, and sys layers) to be sure you don't miss anything. 
Or with multiple displays all the displays. In fact, following this approach a single `lv_refr_all_objs()` would be enough. 
It could be used on language change as well. 

But isn't it overkill? Let's say there are 50 variables rapidly changing (e.g. with 20 Hz). It means 1000 refresh/seconds for all displays and objects.

To know if an object needs to be updated it needs to 
1. save the previous value (extra effort and memory)
2. get its current value, e.g. with `lv_roller_get_selected_id(roller)` (extra effort and performance)

In my opinion, doing any of the above is a little bit problematic.

> I don't quite understand this; isn't that what LV_EVENT_VALUE_CHANGED is for?

Ups, that's true. :slightly_smiling_face: 
An exception could be if the variable can be changed by many objects then you need to repeat the same "action" every place. (I don't an example now)
Besides, it seems like a better isolation of layers if the UI code is free from hardware settings.

What do you think?

puzrin, 2020-06-25:
> But isn't it overkill? Let's say there are 50 variables rapidly changing (e.g. with 20 Hz). It means 1000 refresh/seconds for all displays and objects.

https://knockoutjs.com/documentation/rateLimit-observable.html

Angular and some other frameworks have global rate limiter for dom update. They work with virtual dom and sync changes with real one in bulk with 20-50FPS frequency.

embeddedt, 2020-06-25:
> Practically it means refreshing the whole display (including the active screen, top, and sys layers) to be sure you don't miss anything.

I think you wouldn't need to refresh that many objects in practice. Only the objects that are actually being shown to the user need to be refreshed.

> To know if an object needs to be updated it needs to:
> 1. save the previous value (extra effort and memory)

Objects are already doing this, though.

> 2. get its current value, e.g. with lv_roller_get_selected_id(roller) (extra effort and performance)

How expensive is it to get an object's current value? In most cases it's as simple as retrieving an integer from memory.

kisvegabor, 2020-06-25:
@puzrin 
> Angular and some other frameworks have global rate limiter for dom update. They work with virtual dom and sync changes with real one in bulk with 20-50FPS frequency.

That's reasonable. We can limit it by setting up a task to start refresh if it was asked since the last call. 
Managing a "virtual DOM" is way out of scope now, so user can decide to be out of sync for a short time or refresh immediately. 

@embeddedt 
Okay, I'm convinced :slightly_smiling_face: 

Thank you for both of you :slightly_smiling_face: 

So the final conclusion is to have a single `lv_obj_refresh(NULL for all or pointer)` function and the user needs to handle all the things in the event callbacks?

embeddedt, 2020-06-25:
> So the final conclusion is to have a single `lv_obj_refresh(NULL for all or pointer)` function and the user needs to handle all the things in the event callbacks?

Yes.

kisvegabor, 2020-06-25:
Okay, that's pretty simple. I can add it tomorrow.

puzrin, 2020-06-27:
Could you clarify, how refresh happens? If i need to update 100 different objects, can i stack changes? I mean:

- start transaction
- mark objects dirty (+ change required data)
- end transaction
- redraw

Question2. What about if i need to delete 100 objects in bulk?

Question3. Is it possible to just select desired frame rate (see `AnimationFrame` on web) to recalculate layout & redraw?

kisvegabor, 2020-06-29:
>  If i need to update 100 different objects, can i stack changes? I mean:

There can be more functions for more precise control:
- `lv_obj_refresh()` send event to all objects now. Sent 100 times if called 100 times
- `lv_obj_refresh_scheduled()`  send only one event in every `AnimationFrame`

> What about if i need to delete 100 objects in bulk?

Delete happens immediately but the screen is redrawn only in every `LV_DISP_REFR_PERIOD` milliseconds.

> Is it possible to just select desired frame rate (see AnimationFrame on web) to recalculate layout & redraw?

For redraw yes (see above) but layouts are recalculated automatically. It wasn't so slow so far to make it necessary.

kisvegabor, 2020-07-08:
I've implemented it as we discussed. See https://github.com/lvgl/lvgl/commit/15b7ea6614fa4ba383ed298e8b0886e4ed585eb0

The docs are also updated:https://docs.lvgl.io/dev/en/html/overview/event.html#refresh-event

puzrin, 2020-07-08:
Principle of debounced refresh request looks reasonable.

But creating separate task for each unique event looks suspicious about used memory https://github.com/lvgl/lvgl/commit/15b7ea6614fa4ba383ed298e8b0886e4ed585eb0#diff-4aea7b208af4a53e4a82def44eeaa246R1815 (and reallocaions). May be create 'dirty' bit flag in each object?

kisvegabor, 2020-07-08:
Every task needs 24 bytes. My approach was: if you update 20-30 objects (500-700 bytes) you will rather refresh all objects with `NULL`.

Having a "dirty" bit would require checking every object in **every** ~30ms which raises some performance-related question.

puzrin, 2020-07-08:
Well, if api is stable, inernals can be reorganized later.

I see another problem - race condition. If refresh is posponed, object can be deleted before. See https://lodash.com/docs/#debounce for possible debouncer modes. If you don't track trailing event, you can get nasty unsync in some cases. If you track trailing - deleted object will cause crash.

puzrin, 2020-07-08:
> Having a "dirty" bit would require checking every object in **every** ~30ms which raises some performance-related question.

Full scan per animation frame may be normal. I'd suggest to verify optimizations/design criterias. I suppose, lvgl is for limited resources, with limited complexity. That means, it sould not rely on advanced cache. So:

1. It MUST be able to make full (re)draw of "animation frame" with desired FPS.
2. Optimizations can drop objects out of screen, vackground and so on. But those can not reuse data from previous state (previous animation frame).

I do not mean this approach is best. I mean it's consistent and simple enough. If you try to use more advanced approach, things become 10x more complicated to keep consistency.

kisvegabor, 2020-07-09:
> I see another problem - race condition. If refresh is posponed, object can be deleted before

Ah, good catch! I fixed it here: 09250d0a
A lot of other things need to be handled anyway during deleting objects so adding one more is not a problem.

Honestly, I find the current solution more simple and elegant. I vote for leaving it as it is and - as you mentioned - if we find a problem later we can change the internals without breaking the API.

puzrin, 2020-07-09:
Unfortunately, some problems can not have elegant solution. You are on the way to reinvent smart poiners and object destructors :)

1. The root of problem is, object has property "dirty" and you wish denormalaze data store. I would not do it until impossible to avoid, because it causes ~ the same range of problems as atomic access in multicore systems. Right data design is more imporant than unproven premature optimizations.
2. If you work with denormalized data, formal requirement is method to normalize it back. You selected one-way forced synk. IMO, method "is_object_valid" would be less coupled.
3. If you need fast scan/access to sparce data ("dirty" property), formal answer is "you need sparse index"

[11 more replies]

closed 2020-07-14 by kisvegabor