lvgl/lvgl #1537: Add lv_obj_align_x/y()

kisvegabor, 2020-05-28:
[lv_obj_align](https://github.com/lvgl/lvgl/blob/ec5b10142f367ba204bc2bdb8ce867ef9969b34d/src/lv_core/lv_obj.c#L875) is a powerful function to easily align object. 

However, in some case, different alignment should be applied vertically then horizontally. For example, align a button horizontally to the middle of the screen but vertically below a chart object.

To make this kind of alignments easier it'd be great to allow separate x and y alignment.  

At first look, the implementation could be very similar to `lv_obj_align` but instead of `lv_obj_set_pos` `lv_obj_set_x` or `lv_obj_set_y` should be called.

The task I see now:
- [ ] Implement the new functions and test them
- [ ] Update the docs
- [ ] Create some examples

Contribution is very welcome in this feature. If you have any ideas or you are interested in implementing this feature, feel free to comment here!

embeddedt, 2020-06-22:
What if we just added a flag to `lv_obj_align` that lets you choose which axes the alignment is applied on? It's not the cleanest API, but it means that no extra code should be needed.

kisvegabor, 2020-06-23:
As it's a common function and already has 5 parameters I think it'd be better to add new functions. However, to avoid code duplication an internal `static` function with "axis selector" sounds good to me.

kisvegabor, 2020-06-26:
I've implemented these functions and `lv_obj_align_origo_x/y` and updated the docs.

Finally, I didn't add examples for this trivial feature.

closed 2020-06-26 by kisvegabor