lvgl/lvgl #1538: Support multi core rendering

kisvegabor, 2020-05-28:
MCUs are evolving very fast and already a lot of companies have multi core MCUs (e.g. ESP32). I'm thinking about how could we take advantage of the extra core.

My first idea is if the display is refreshed in multiple independent chunks we can delegate the rendering of chunks to multiple cores. When a chunk is ready it calls `disp_flush_cb` to send it to the display. 

LVGL is not ready to works like this right now because there are some common global resources that should be moved to "render instances".

The first unclear part to me is how should a platform-independent "rendering task elevator" look like. That is how can lvgl ask the MCU to start a task with some parameters on a core and after that start the same task with different parameters on another core.  

It'd be great to hear your ideas and opinion on this topic!

excitedbox, 2020-06-02:
I think it might be best to have the option to dedicate a core to the display and one to the main tasks. for things like a cnc or 3d printer where keeping position is important it would decrease the chance of missing steps. I chose the esp32 just for this reason So I could run critical tasks on their own core while the rest such as the input, display and connectivity is on another.

kisvegabor, 2020-06-02:
@excitedbox 
It's an important point. 
So it allocating a core/thread by the user is useful even if only one 1 core/thread is used for rendering. Although in this scenario LVGL itself would run on the same core as everything else, only rendering would be sent to the other core.

How does it look like on ESP? I mean how do you tell ESP to run a task on the other core? It'd be important to start that task very very quickly when required. E.g. low OS tick (<= 1ms) and very high priority.

I'm not sure nut I think on POSIX systems creating a `pthread`s would do the job. Can you confirm that dynamically creating `phread`s for new rendering tasks to delegate them to different cores could work?

barbiani, 2020-06-02:
A high priority task pending on a lightweight signal takes about 7us to
resume on the esp32.

kisvegabor, 2020-06-02:
@barbiani 

Both are great! Thank you!

closed 2020-07-01 by stale[bot]

WebDust21, 2024-04-23:
I know this is an old (and closed!) issue, but my 2 cents:

Processors these days are ever faster.  But more often than not, they're severely retarded--sometimes almost effectively strangled--by data I/O rates in and out of the processor.  We've all bought a cheap Chinese tablet, "it's got a quad-core 1.4GHz processor"--only to be utterly disappointed by the horrible performance.  The processor isn't at fault...it's the bottleneck of the FLASH memory access to blame.

Take the ESP32 for instance.  ALL program code is stored in an external FLASH chip, which is accessed via either DIO or QIO (i.e. 2-bit or 4-bit bus).  In essence, with an SPI clock speed of 80MHz, this works out to 20MB/s for DIO, and 40MB/s for QIO.
Meanwhile, the CPU has a 240MHz clock.  The Tensilica Xtensa instruction set is a varied-length method (i.e. based on the instruction, it can be anywhere from 1-3 bytes as far as I've seen).  So _best case_, assuming every instruction is one byte (which isn't the case!), in order for the CPU to be completely unhindered, it needs _240MB/s_ minimum for ONE core.  
But the ESP32 implementation has TWO cores that share a 40MB/s bus.

Yes, the ESP framework reads in blocks of data from the SPI Flash, and executes it out of the 520KB internal RAM (which as far as I know does not have any waitstates).  BUT every single time a function call is made, the ESP has to go back to the FLASH chip and pull in the new subroutine...then drop back to the previous one after the return.
As a result, while the ESP32 is quite a powerful CPU, it's effectively forced to spend most of its time waiting for the next instruction to run from SPI FLASH--wasting most of the available processing power.

Fortunately, there is a (somewhat platform specific!) workaround that will result in huge speed gains/improvements on the ESP32 platform.  (Any other platform that runs off an SPI FLASH chip should have something similar.)  It's possible to declare functions "void IRAM_ATTR functionName", which forces the ESP framework to **always** keep said functions in the (no-waitstate) IRAM.  If judicially used in the highest priority framebuffer rendering functions (i.e. pixel draw, alpha-blending, etc.), the severe bottleneck of the SPI FLASH can effectively avoided--significantly increasing functional throughput.
Obviously, this can't be done with every critical function--there's a limited amount of IRAM, shared with the entire remainder of the user application.  But for a few of the most critical rendering functions, it would make a significant improvement to the entire library.  (I did the "IRAM_ATTR" trick on a monochrome LCD library, putting it on the "Draw Pixel" routine.  Without it, you could visibly see the ESP32 "draw down" the LCD.  Afterwards, redraw was effectively instantaneous.)

kisvegabor, 2024-04-25:
Thanks for sharing your thoughts. 

Actually the throughput is even worse due to setting the address and other overheads. 

I think rotating an image is the worst case scenario as we need to read both the image and the code from the flash and the large image will fill the cache, so the code needs to read over and over too.

In LVGL we have [`LV_ATTRIBUE_FAST_MEM`](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L388) exactly for this. Probably we can optimize a little which functions to place into RAM, but it already adds a measurable speed up.

WebDust21, 2024-04-25:
> Actually the throughput is even worse due to setting the address and other overheads.

Right, of course...I was just trying to really generalize things ;-)

Someone recently asked me, "which is faster, a Pentium III processor, or the ESP32?"  A single-core Pentium III running at ~500MHz (for fairness!) and a dual-core ESP32 running at 240MHz...
Well...the Pentium III has a 133MHz 32-bit FSB, which gives a base I/O data rate of ~532MB/sec...compared to the ESP32 at <40MB/sec...!  The ESP32 is just about strangled.

> In LVGL we have [`LV_ATTRIBUTE_FAST_MEM`](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L388) exactly for this. Probably we can optimize a little which functions to place into RAM, but it already adds a measurable speed up.

Interesting.  I guess I was expecting a more-noticeable throughput improvement, but maybe I should disable this just to see what difference it makes:
https://github.com/lvgl/lvgl/blob/79b64c8bd8f2ee51df6954fe0e0b5fbf58ff85db/env_support/cmake/esp.cmake#L51-L53
It does appear that a fairly large number of functions use that attribute.

Did some testing, and I'm not noticing any difference whatsoever with it enabled in lv_conf.h or commented out.  No change either if I add "-DLV_ATTRIBUTE_FAST_MEM=IRAM_ATTR" to the project buildflags (and comment it out in lv_conf.h").  Missing something?

kisvegabor, 2024-04-25:
I'm not a power ESP user, but @zjanosy knows more about these kind of config issues. :slightly_smiling_face:

zjanosy, 2024-04-26:
You can check which functions have been allocated in the internal SRAM by looking at the map file, or in Eclipse right click on the project name, and select "ESP-IDF: Application Size Analysis". In the "Details" view you can see the IRAM usage of individual source files. For more details you need to look into the .map file.

Putting frequently used routines in SRAM does help a little, but only if the function does not call other functions, or accesses variables or constants that are not in SRAM. It is best suited for tight loops, which only read or write SRAM.

Another thing is that the internal cache actually does a good job speeding up flash memory access. Therefore, if a routine is small enough to fit into the cache, and does some lengthy calculation in a loop, then it would be just as fast as running from IRAM. But most LVGL functions are not like that, therefore the overall performance will not be improved just by putting some functions in IRAM.

Actually, if you have enough SRAM, you can try to put all LVGL code into the IRAM. You need to add a "linker fragment" to your code:

https://docs.espressif.com/projects/esp-idf/en/v5.2.1/esp32s3/api-guides/memory-types.html?highlight=linker%20fragments#how-to-place-code-in-iram

My experience is however, that you will only get a modest speed improvement, so do not expect too much.