lvgl/lvgl #1539: simulator crashes when LV_LAYOUT_ROW_MID is selected.

nusret3, 2020-05-28:
Line 123 in lv_examples/src/lv_demo_widgets/lv_demo_widgets.c
```
lv_cont_set_layout(h, LV_LAYOUT_PRETTY_MID); // works fine
// lv_cont_set_layout(h, LV_LAYOUT_ROW_MID); // TFT simulator crashes without any message
```

embeddedt, 2020-05-28:
I've transferred this issue to the main LVGL repository since it's not specific to the simulator.

nusret3, 2020-05-28:
> I was thinking about making layout and fit to general `lv_obj` properties or style properties.
> This way e.g. a label's width could be set the parent's width.
> 
> I'd make `lv_cont` useless.
> 
> What do you think?

If I were you I'd keep it for a while for legacy support. Just an advice.

kisvegabor, 2020-05-29:
> I like that idea! HTML/CSS have something similar where there is no "special" container object.

Great! Then the question is whether it should be stored in styles or in lv_obj properties. I see these pros and cons 
**Styles**
\+ CSS does the same
\+ Uses RAM only when used
\- Slower to get

**lv_obj**
\+ Fast 
\- Consumes RAM for every obj type.

The speed is important because these functions might be called a lot of times but at least not during rendering but during object creation. 

To save RAM in lv_obj I've already considered putting the rarely used field into a `lv_obj_special_t` struct and store an `lv_obj_special_t *` in `lv_obj_t` and allocate it only when one of its parameters is used. E.g. these parameters can be there: 
- the new layout and fit settings
- event_cb
- ext_click_pad
- ext_draw_pad?
- some bits from the bitfield
- group_p
- protect
- realign
- user_data 

Depending on the enabled features it's ~20 bytes. But +4 byte to store a pointer to `lv_obj_special_t *` and the allocation also has 4 byte overhead. So it doesn't seem a big gain but at least a good direction IMO to allow more features in lv_obj.

> We could keep the lv_cont API names around for backwards compatibility and then drop them entirely in 8.0.

> If I were you I'd keep it for a while for legacy support. Just an advice.

I agree.

nusret3, 2020-05-29:
May be you should keep event_cb in where it was. It is regularly used in most apps. Otherwise just adding an event callback will allocate many bytes in RAM.

nusret3, 2020-05-29:
@embeddedt I am not familiar with CSS. I am an old fashioned guy but many existing apps based on LVGL are highly depending on inheritance of styles from parent as I see. It will be very painful for those who wants to migrate from 6.x to newer releases.

embeddedt, 2020-05-29:
This would be a separate type of style just for the layout properties. The other properties will continue to use inheritance.

kisvegabor, 2020-05-31:
Actually not all properties are inherited. See [this enum](https://github.com/lvgl/lvgl/blob/4575fec915929a4a733ff26e628ea54668ab343d/src/lv_core/lv_style.h#L96) and look for `LV_STYLE_ATTR_INHERIT`. Mainly only the text  (as in CSS) and image properties are inherited. 

E.g. Background color is not inherited from the parent.

To decide if something should be style or obj property I ask myself: "Is it used by the designer or the programmer?" I think the layout and fit are somewhere in between.

nusret3, 2020-05-31:
> To decide if something should be style or obj property I ask myself: "Is it used by the designer or the programmer?" I think the layout and fit are somewhere in between.

Both may be ?

kisvegabor, 2020-06-01:
> Both may be ?

Do you mean it's both developer and designer world?

nusret3, 2020-06-01:
There is no always sharp lines between developer and designer in many small teams. That's what I meant.

kisvegabor, 2020-06-02:
Got it, and I agree.

Let's place it into the styles first and see how it affects performance.

kisvegabor, 2020-06-23:
Not stale

embeddedt, 2020-06-23:
Here are my ideas for layouts:

* Remove all the existing container layouts and replace them with the CSS flexbox layout specification. React Native uses it exclusively for their layout system, and it's also extremely widespread on regular websites these days.
* Use style properties for specifying the flexbox configuration. It shouldn't affect performance that much because the properties are not inherited.
* Replace the current [`LV_PROTECT_POS`](https://github.com/lvgl/lvgl/blob/7af64df008c5abe4ef88ec766241052b9fe66b1d/src/lv_core/lv_obj.h#L172) flag with an implementation of the CSS [`position` attribute](https://developer.mozilla.org/en-US/docs/Web/CSS/position). For our purposes, I think the  `absolute` and `static` values should be enough. We can always implement more CSS values later if we really wanted to.
* Drop the `lv_cont` object entirely and move the layout logic into `lv_obj`. Essentially, we'd be converting `lv_obj` into an HTML `div`.

Essentially, we'd be moving LVGL away from a custom layout/positioning implementation and towards HTML/CSS standards. It also means that we can take advantage of the wealth of documentation already available for flexbox.

kisvegabor, 2020-06-23:
I like all four ideas. Some questions:
- Should we remove `LV_FIT_...` too and join into layouts?
- Apply a default layout on `lv_obj`? In HTML objects are not placed on top of each other by default.

embeddedt, 2020-06-23:
> Should we remove `LV_FIT_` too and join into layouts?

Yes. It wouldn't be needed anymore as I'm pretty sure flexbox can do the same thing.

> Apply a default layout on `lv_obj`? In HTML objects are not placed on top of each other by default.

I agree that it would be more useful, I'm not quite sure what the default properties would be. We should probably experiment with that. 

However, it would make migration of existing applications easier if layouts were *not* applied on `lv_obj` by default (since I don't know how many people actually use the current layout system for their whole application).

kisvegabor, 2020-06-23:
> I agree that it would be more useful, I'm not quite sure what the default properties would be. We should probably experiment with that.

I'm afraid people would ask "why can't I position the objects". So probably we are better to leave the layouts disabled by default. 

I've checked the [poster](https://css-tricks.com/snippets/css/a-guide-to-flexbox/) you linked in Forum in more detail and all seems doable. It will be quite a big work but I'm sure it will be worth it in the end.

embeddedt, 2020-06-23:
I'd suggest that we start by implementing `position` and flexbox in `lv_obj`, but leave `lv_cont` in the codebase to allow users to continue using legacy layouts. That way they can gradually adopt flexbox without needing to rewrite their entire project (especially since we already dropped an entirely new style system on them in 7.0).

embeddedt, 2020-06-24:
I'm currently working on the rewrite of the image converter, but once that's finished, I can help with this if you'd like. I think the only missing component from the style system is the ability to trigger layout on an object without a full invalidation. We essentially need a hook to find out when certain properties change.

kisvegabor, 2020-06-24:
> I'm currently working on the rewrite of the image converter, but once that's finished, I can help with this if you'd like.

Good to know, thank you! :clap: :tada: 

> I think the only missing component from the style system is the ability to trigger layout on an object without a full invalidation. We essentially need a hook to find out when certain properties change.

Something like this already works in [lv_obj_refresh_style](https://github.com/lvgl/lvgl/blob/63aba1b3de0fa233b09e80d465665c219d82d0ae/src/lv_core/lv_obj.c#L1351). It tries to figure out what kind of actions are required to refresh a property.

---

I'm thinking about the current  `fit` property. It seems flexbox doesn't take into account the children's content size, therefore there is no way to set e.g. a button's width according to the text on it. 

In CSS/HTML we can use `width`, `min/max-width` and `display: inline` to achive this but it's still not clear how to adopt it into LVGL.

Another related difference in LVGL-HTML is the handling of scrollable content. In LVGL we have `lv_page` for this, but HTML can handle it for every element (or at least as far as I know). 
It'd be nice to adopt this approach somehow. The missing concept seems to be an "offset" which is added to every children's coordinates when the object s being scrolled.

embeddedt, 2020-06-24:
> The missing concept seems to be an "offset" which is added to every children's coordinates when the object s being scrolled.

I think the scroll offset should be added to `lv_obj`, and taken into account when objects are refreshed. Essentially, we'd want to deprecate the idea of having separate objects for basic functionality like this.

Another advantage of moving LVGL towards HTML like this is that we may be able to implement a GUI designer using pure HTML, and just convert it to LVGL syntax at the end.

kisvegabor, 2020-06-24:
> I think the scroll offset should be added to lv_obj, and taken into account when objects are refreshed. Essentially, we'd want to deprecate the idea of having separate objects for basic functionality like this.

I agree. Need to think about what is the cleanest way to do it. 
Maybe it's enough to simply change the coordinates of all children. When all children are visible on a given side of a parent just stop change the coordinates further in that direction. Although this way the positions set by the user will change. `offset` could come here into the picture to correct this change.

kisvegabor, 2020-06-27:
I suggest breaking it into more steps an open an issue for each. 

1. Flexbox: There are some questions about implementation, but it seems doable in general. 
2. New scrolling: There are many questions and features to discuss.

So I  close this issue now. Let's continue in #1614 and #1615.

closed 2020-06-27 by kisvegabor

embeddedt, 2020-12-31:
Partial fix in #1986; the real fix is to use v8 layouts, however the v8 API in general is still unstable and subject to change.