lvgl/lvgl #1540: lv_tabview works  width btn incorrectly.

iDalink, 2020-05-29:
As title description, when I add btn to lv_tabview by lv_tabview_add_tab function, I got fine display. But when I track btn to left, tabview did not scroll follow my fingner util my touch reach edge of btn.

**To Reproduce**

    lv_obj_t *tabview;
    tabview = lv_tabview_create(main_obj, NULL);
    lv_tabview_set_btns_hidden(tabview, true);
	lv_tabview_set_anim_time(tabview, TABVIEW_ANIM_TIME);

    //list
	int line_num = sizeof(data_src) / sizeof(data_src[0]);

	for (long i = 0; i < line_num; i++)
	{
	    lv_obj_t *tab = lv_tabview_add_tab(tabview, "T");

		lv_obj_t *item_obj = tab;

		lv_obj_t *button = lv_imgbtn_create(item_obj, NULL);
		lv_imgbtn_set_src(button, LV_BTN_STYLE_REL, data->dft_img);
		lv_imgbtn_set_src(button, LV_BTN_STYLE_PR, data->dft_img);
		lv_obj_align(button, item_obj, LV_ALIGN_IN_TOP_MID, 0, 37-9);
		button->user_data = (void *)i;
		lv_obj_set_event_cb(button, event_handler);
		lv_page_glue_obj(button, true);
}

**Expected behavior**

Tabview scroll in horizontal direction when I track btn childrens.

**Screenshots**
None.

**Additional context**

embeddedt, 2020-05-29:
Are you using 6.x or 7.0? Scroll propagation got an overhaul in 7.0 and many issues have been fixed there.

iDalink, 2020-05-30:
I am using version 6.1.2.

iDalink, 2020-05-30:
I will try out version 7.0 later. Here close the issues temporarily.

closed 2020-05-30 by iDalink