lvgl/lvgl #1544: lv_label_set_text_fmt floats broken in v7

curly-brace, 2020-06-01:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

after upgrading to version 7 labels that were showing floats now show just 'f' letter.
that is also true for 'g', 'e' and even 'a'.
but decimal numbers still work fine.

**To Reproduce**

create a label and set text using
`lv_label_set_text_fmt(label, "%f", 100/3.0);`

**Expected behavior**

should show the float instead of 'f' letter.

embeddedt, 2020-06-01:
Does flipping [this value](https://github.com/lvgl/lvgl/blob/ef1242f23452821766c3e075236b840bb344f398/src/lv_misc/lv_printf.c#L41) to `0` fix the issue? If so we can just expose this in `lv_conf.h`.

curly-brace, 2020-06-01:
nope, flipping to 0 does not work, but after i've removed that line, it worked.
there weren't such a setting in previous version, right? don't remember anything like this.

embeddedt, 2020-06-01:
It was disabled by default in [`16ed5c6`](https://github.com/lvgl/lvgl/commit/16ed5c605ab63015620033a4c5b059a1adb5f207#diff-bd9784087846b8320e54d1d0414836e1).

I've [made the option available](https://github.com/lvgl/lvgl/blob/c5c5b2a3932f8acef3b0374a5843569825807162/lv_conf_template.h#L482) in `master`.

closed 2020-06-01 by embeddedt

kisvegabor, 2020-06-01:
Thank you @embeddedt 

I've disabled it to avoid using `float` by default because it might highly increases memory usage.