lvgl/lvgl #1547: Add JPG decoder

kisvegabor, 2020-06-01:
Similarly to the [PNG decoder library based on lodepng](https://github.com/lvgl/lv_lib_lodepng) it'd be great to have a JPG decoder too for LVGL. 

If you are already familiar with a JPG library, it should be straightforward to add it based on the above mentioned PNG library.

A detailed description of LVGL's image decoder interface can be found [here](https://docs.lvgl.io/v7/en/html/overview/image.html#image-decoder).

[Tiny JPED Dec](http://elm-chan.org/fsw/tjpgd/00index.html) looks like a good candidate at first. 

Contribution is very welcome, feel free to comment here if yore interested in creating a JPG decoder interface for LVGL.

xennex22, 2020-06-02:
Please structure this with STM32 hardware jpeg support in mind.

kisvegabor, 2020-06-03:
> Please structure this with STM32 hardware jpeg support in mind.
Sure! If you already know it could you point to a minimal example code to see how it works?

xennex22, 2020-06-03:
Sorry, don't have any working jpeg code yet.  I was thinking more to have a driver interface so that it would be easy to swap between a software driver and hardware driver.  Probably the answer is to make a software image decoder and also a hardware image decoder rather than another driver layer.

kisvegabor, 2020-06-04:
> Probably the answer is to make a software image decoder and also a hardware image decoder rather than another driver layer.

I agree.

kisvegabor, 2020-06-11:
@C47D 
Awesome! :slightly_smiling_face: 

Regarding the API: the highest level API should be the image decoder interface of LVGL. It looks like this in [PNG decoder](https://github.com/lvgl/lv_lib_lodepng/blob/master/lv_lodepng.c#L45-L51).
For the lower-level API which makes the decompression [TinyJPEG](http://elm-chan.org/fsw/tjpgd/00index.html)'s API seems quite flexible. It has a [data input function](http://elm-chan.org/fsw/tjpgd/en/input.html) which makes it easy hide how the data is read.
What do you think? Can we adopt it?

C47D, 2020-06-12:
Hi,

I started working on this, I'm using the codeblocks simulator. Worked on it for about an hour and I'm able to get image information data from a file on the filesystem.

Do you want me to report my work on progress or only when it's done?

Regards

kisvegabor, 2020-06-13:
Hi Carlos,

> LVGL image decoder expects me to give it an array of color data, this array should be of uint8_t's,
does LVGL expects the data in RGB order?

LVGL will treat `dsc->img_data` as `lv_color_t`. 

> I tried to fit the bitmap data from (0,0) to (0,7) into an array later assigned to dsc->img_data before returning from the open callback, but how to tell LVGL how many bytes, we got back from TJPGD?

As you found it in the docs there are 2 options: open the whole image in the `open` function, or line by line in `read_line` function. 

In case of the first option, the size of the buffer is known from `img_w x img_h x sizeof(lv_color_t)`.
In the `open` function you can allocate a large array with this size, decompress all pixels here, and set the address of this array in `dsc->img_data`. This is what the [PNG decoder](https://github.com/lvgl/lv_lib_lodepng/blob/master/lv_lodepng.c#L153) does as well. 

In the case of `read line` LVGL will give the buffer with the required size and you just need to add `width` number of pixels. As you can't tell from where to start decoding the file (JPG doesn't support decompressing arbitrary areas) you can always start from the beginning and buffer 8 lines. LVGL asks the lines in order, so you are good for the 8 first calls. If LVGL asks the 9th line, you need to decompress 8-15 lines, and so on.
You can store the current state in `dsc->user_data`.  Note that, it's not sure that first lines are required. It can happen that only the middle or bottom of the image needs to be redrawn. In this case, the first lines can be discarded. 
 It's a little bit trickier but would be great for smaller systems.
 > /* User defined device identifier */
typedef struct {
    FILE *fp;          /* File pointer for input function */
    uint8_t *fbuf;     /* Pointer to the frame buffer for output function */
    uint16_t wfbuf;    /* Width of the frame buffer [pix] */
} IODEV;

You can `alloc` an `IODEV` and store it in `dsc->user_data`. It allows dealing with multiple files at same time. 
Actually it's required to handle files simultaneously because LVGL always keeps the images open as long as possible. If you can handle one file at a time but have 2 images, thing will mess up.

C47D, 2020-06-13:
Thanks for the explanation @kisvegabor, got something working on the simulator.
For now it only handles opening files and decoding everything in one `jp_decode` call.

[image]

Will upload the code after some cleanup.

C47D, 2020-06-14:
@kisvegabor Some additional questions:

- Should we assume the user have access to functions such as `fread`, `fseek`, `fclose` when reading the image file from the file system?
- Can the library be configured to be used using only open or using readline callabacks?

embeddedt, 2020-06-14:
> Should we assume the user have access to functions such as `fread`, `fseek`, `fclose` when reading the image file from the file system?

I don't think so. They could be using ChaN's FatFS, for instance.

C47D, 2020-06-14:
Or register a drive into lvgl file system?
I ask because the lv_lib_lodepng repo use the aforementioned functions.

C47D, 2020-06-14:
Also, how feasible would it be to add another callback to the image decoder that receives an array of decoded data and the area it corresponds to?

Now the decoder have a callback so the user can feed it with decoded data from a given line of the image.

C47D, 2020-06-15:
@kisvegabor 

I've created a repo where to upload the work in progress (lv_lib_tjpgd)[https://github.com/C47D/lv_lib_tjpgd], I'm working on the dev branch of it with the partial decoding of the image.

When calling `jp_decomp` it always starts to decode the image from the top, if you return 0 from the `outfunc` function it stops the decoding, and I assumed that it would remember where it was so if we call `jp_decomp` again it started from the last point it decoded, but no, it starts from the top again. Maybe I'm missing something, will check for more compatible jpg decoding libraries.

This is the decompression function, from what I understand it always start the horizontal decompression at 0 until `jd->height`.
```c
JRESULT jd_decomp (
	JDEC* jd,								/* Initialized decompression object */
	uint16_t (*outfunc)(JDEC*, void*, JRECT*),	/* RGB output function */
	uint8_t scale							/* Output de-scaling factor (0 to 3) */
)
{
	uint16_t x, y, mx, my;
	uint16_t rst, rsc;
	JRESULT rc;

	if (scale > (JD_USE_SCALE ? 3 : 0)) return JDR_PAR;
	jd->scale = scale;

	mx = jd->msx * 8; my = jd->msy * 8;			/* Size of the MCU (pixel) */

	jd->dcv[2] = jd->dcv[1] = jd->dcv[0] = 0;	/* Initialize DC values */
	rst = rsc = 0;

	rc = JDR_OK;
	for (y = 0; y < jd->height; y += my) {		/* Vertical loop of MCUs */
		for (x = 0; x < jd->width; x += mx) {	/* Horizontal loop of MCUs */
			if (jd->nrst && rst++ == jd->nrst) {	/* Process restart interval if enabled */
				rc = restart(jd, rsc++);
				if (rc != JDR_OK) return rc;
				rst = 1;
			}
			rc = mcu_load(jd);					/* Load an MCU (decompress huffman coded stream and apply IDCT) */
			if (rc != JDR_OK) return rc;
			rc = mcu_output(jd, outfunc, x, y);	/* Output the MCU (color space conversion, scaling and output) */
			if (rc != JDR_OK) return rc;
		}
	}

	return rc;
}
```

kisvegabor, 2020-06-15:
I agree to use lvgl's files system if possible. POSIX IO handling was deeply routed in lodepng and in lack of time I haven't updated it yet. However, if we write these parts we should use the more generic option.  

> This is the decompression function, from what I understand it always start the horizontal decompression at 0 until jd->height.

I see, so it always wants to decompress the whole image. What I can imagine is to modify the lib. Fortunately, [jd_decomp](https://github.com/C47D/lv_lib_tjpgd/blob/master/tjpgd.c#L922) is quite simple and it's easy to see where a line is decompressed. 
At first look, adding a  `jd_decomp_line` function seems reasonable. What do you think?

C47D, 2020-06-15:
> I agree to use lvgl's files system if possible. POSIX IO handling was deeply routed in lodepng and in lack of time I haven't updated it yet. However, if we write these parts we should use the more generic option.

Will try to send a PR, I'm getting familiar with lvgl fs with a toy project.

> I see, so it always wants to decompress the whole image. What I can imagine is to modify the lib. Fortunately, jd_decomp is quite simple and it's easy to see where a line is decompressed.
At first look, adding a jd_decomp_line function seems reasonable. What do you think?

I've sent a mail to Chan asking if is possible to do partial decoding, I'm 0% familiar on how JPGs are decoded. We can give it a try tho.

kisvegabor, 2020-06-16:
> I've sent a mail to Chan asking if is possible to do partial decoding, I'm 0% familiar on how JPGs are decoded. We can give it a try tho.

Oh, great! I'm curious how it will go. If he can't help I can try to add the line decoding feature.

tuupola, 2020-06-16:
In theory you should be able to track and save how many bytes was read when decompressing an `mcu` and then  later call `lseek()` with this information. Not that JPEGs decompression  is not a line by line operation. It is done in 8x8 blocks (minimum coded unit ie `mcu`) instead.

I had similar (but not the same) problem recently when implementing an MJPEG decoder with the same library.  In my case I had to search for the previous  `EOI` marker and `lseek()` back to that marker.

kisvegabor, 2020-06-16:
@tuupola You are right. It really steps `jd->msx * 8` and `jd->msy * 8`

tuupola, 2020-06-16:
Also good to note that compressing is done using a [zig zag pattern](https://en.wikipedia.org/wiki/JPEG#/media/File:JPEG_ZigZag.svg). I think this is because the algorithm is based on saving the difference to previous block and adjacent blocks are often are quite similar.

C47D, 2020-06-18:
@kisvegabor An update on the jpg decoder, let me know if you see something wrong or if we are on the right track.

---

Thanks for the help @tuupola, I should had get familiar with how JPEG works before starting this, now with your suggestion and some online reading about JPEG I am a bit more familiar on how it works.

I'm currently "cheating" tjpgd telling it the test image is `width` * 8 pixels (we got the `width` value with `jp_prepare`), I've chosen 8 because of the MCU size as you noted on the comment above.

This is the output, I think we are on the right track:
[image]

This is the test image:
[image]

The first time LVGL calls `decoder_read` it asks for 1 line of data, so we call `jp_decomp` when it returns we have the decompressed data stored in `user_ctx.frame_buffer` and we send LVGL the first line of the image (`len * BYTE_ON_PIXEL` bytes). After that we send the buffered data we have stored in chunks of `len * BYTE_ON_PIXEL` bytes, until we no longer have decompressed data.

```c
static lv_res_t decoder_read(lv_img_decoder_t * decoder, lv_img_decoder_dsc_t * dsc, lv_coord_t x, lv_coord_t y, lv_coord_t len, uint8_t *buf)
{
    lv_res_t retval = LV_RES_INV;

    static bool buffered_data = false;
    static uint8_t lines_sent = 0;

    if (buffered_data) {
        // Send LVGL data starting from the last byte we sent
        memcpy((void *) buf, (void *) user_ctx.lvgl_feeder_ptr, len * BYTES_ON_PIXEL);
        user_ctx.lvgl_feeder_ptr += (len * BYTES_ON_PIXEL);
        retval = LV_RES_OK;
        lines_sent++;

        // We already sent all the data we had
        if (7 < lines_sent) {
            buffered_data = false;
            lines_sent = 0;
            free(user_ctx.frame_buffer);
            user_ctx.frame_buffer = NULL;
        }

    } else {
        JRESULT error = JDR_OK;

        /* Allocate memory to store the decompressed data */
        uint32_t decoded_image_buffer_size = user_ctx.size.width * 8 * BYTES_ON_PIXEL;
        user_ctx.frame_buffer = (uint8_t *) malloc(decoded_image_buffer_size);

        size_t before_decompression = ftell(user_ctx.fp);

        jdec.height = 8; // <- Make TJPGD think the image is width * 8 px
        error = jd_decomp(&jdec, on_decoder_line_output_cb, LV_TJPGD_SCALING_FACTOR);

        size_t bytes_to_send = len * BYTES_ON_PIXEL; // bytes of decompressed data in 1 line
        memcpy((void *) buf, (void *) user_ctx.frame_buffer, bytes_to_send); // LVGL need us to put the decompressed data into buf
        lines_sent++; // Keep track of the number of lines sent to LVGL, user_ctx.frame_buffer stores 8 lines, we already sent 1
        user_ctx.lvgl_feeder_ptr = user_ctx.frame_buffer + bytes_to_send; // Keep track of the unsent decompressed data
        retval = LV_RES_OK;

        buffered_data = true; // We now have buffered data
    }

    return retval;
}
```

This is how we keep track of the "consumed" bytes:
```c
static uint16_t on_feed_decoder_cb(JDEC* jd, uint8_t* buff, uint16_t nbyte)
{
    uint16_t retval = 0;

    decoding_ctx_t *ctx = (decoding_ctx_t *) jd->device;
    ctx->fp_consumed_bytes += nbyte;

    if (buff) {
        retval = fread(buff, 1, nbyte, ctx->fp);
    } else {
        retval = fseek(ctx->fp, nbyte, SEEK_CUR) ? 0 : nbyte;
    }

    return retval;
}
```

If we check the value of `user_ctx.fp_consumed_bytes` when we are done sending data to LVGL (on the `decoder_read` function) we got the following, you can see marked in yellow a RST marker, one byte after the data pointer of TJPGD.
[image]

So my question is, should we `fseek` X bytes until we are on the RST marker? And then do another `jp_decompress`?
AFAIK the `EOI` marker is located at the end of the file, so we couldn't read more data from it, maybe I understand something wrong from your explanation above.

Regards and thanks again for the help @kisvegabor and @tuupola 😄

kisvegabor, 2020-06-18:
@C47D 
Thanks for the update!

There are 2 things I don't understand (maybe there are related:
-  How do you tell to TJPGD to skip the first 8 lines when you go to the second 8 lines.
- How can `buff == NULL` in `on_feed_decoder_cb` to run the else branch?

C47D, 2020-06-18:
> @C47D 
> Thanks for the update!
> 
> There are 2 things I don't understand (maybe there are related):
> -  How do you tell to TJPGD to skip the first 8 lines when you go to the second 8 lines.

This is where I'm stuck at the moment, I wanted to test decompressing the first 8 lines of the image, store them and give them to lvgl one line at the time. 

> - How can `buff == NULL` in `on_feed_decoder_cb` to run the else branch?

Buff is null when tjpgd needs to rewind on the stream buffer, that's part of tjpgd.

tuupola, 2020-06-18:
> AFAIK the EOI marker is located at the end of the file, so we couldn't read more data from it, maybe I understand something wrong from your explanation above.

My problem was similar but not the same. I was implementing an [MJPEG player](https://appelsiini.net/2020/esp32-mjpeg-video-player/) with the tjpgd library. MJPEG is basically a bunch of JPG files concatenated together. My problem was that tjpgd was reading past the `EOI` marker which always caused the next frame to be broken. The solution was to rewind filepointer back to the `EOI` marker.

kisvegabor, 2020-06-18:
@tuupola 
> I was implementing an MJPEG player with the tjpgd library.

A video player would be another great feature for LVGL. :sunglasses: 
Would you be interested in contributing with it?

@C47D 
I think the easiest would be to create `jd_decomp_line_init` and `jd_decomp_line_next` functions based on the original `jd_decomp`. 

I can try to add it tomorrow if you wish.

C47D, 2020-06-18:
@tuupola Thanks for the link to your blog post, it's very neat to display video on such small system. Indeed the problem is not the same, as you said you have a bunch of JPEG images one after the other, that's why you can locate the `EOI` marker and call `jp_prepare` again on your while loop.

I think the problem we are facing here is that we can not allocate memory for the whole image, so we were thinking in decoding 8 lines at the time, but I'm stuck on how to continue the JPEG decoding after the 8 lines are being decoded.

Inside `jp_decom` we can wee the `RST` markers from 0 to 25 being identified, I'm stuck now, will need to clear my head for a bit lol.

@kisvegabor To be honest I'm stuck, I don't know how to continue the decompressing after tjpgd is done with the first call to `jp_decom`.

tuupola, 2020-06-19:
IIRC tjpgd itself only needs around 3100 bytes for the work buffer. The output function for tjpgd needs 8 * 8 * 2 bytes if outputting one mcu at a time, less if outputting with putpixel. Does lvgl internals require to output more than one mcu at a time? You mention outputting eight lines.

Do not get fixated too much with the mentioning of the `EOI` marker. My problem was how to continue MJPEG decoding after a frame. This problem is how to continue JPG decoding after a mcu or bunch of mcus. Both have the the same possible solution ie. `fseek()` the filepointer to correct position before continuing. Difference is what is the correct position.

That said, tjpgd was designed  so that `jd_decomp()` is called only once. It might be easier to to modify the output function so that it collects the mcu's you want to display and then flushes them to screen at once. Or if internals require that `jd_decomp()` is called multiple times then do not `fclose()` the filepointer between calls. If the filepointer is not closed there is no need to call `fseek()` either.

tuupola, 2020-06-19:
> A video player would be another great feature for LVGL. 😎 Would you be interested in contributing with it?

@kisvegabor I don't think my C is good enough yet :) I recently started to learn C by also doing an [embedded graphics library](https://github.com/tuupola/hagl). It is more low level thing which I use for my hobby projects. But I must say C is the only language I actually enjoy writing at the moment.

kisvegabor, 2020-06-19:
@C47D I had a busy day so I can take look at it on Monday.

@tuupola 
No problem. Good luck with your library! :slightly_smiling_face:

C47D, 2020-06-19:
@kisvegabor No problem, yesterday I realized that when `jp_decomp` returns the decompressor is still decompressing the bitstream, narrowing down the new function we need to add.

I've also noticed that LVGL does not guaranty asking for lines starting from the line 0, it can ask for lines in the middle of the image so we need to take into consideration that.

kisvegabor, 2020-06-22:
@C47D I opened a PR with line decoding. It's nothing else then separating the decompress function into an init and a next function. 

> Sorry if this is taking more time than expected, it was a new topic for me sweat_smile

There is no expected time.  Just work on it when it feels comfortable.  :slightly_smiling_face:

kisvegabor, 2020-07-13:
I think adding support to full decode only (skip `read_line_cb` for now) is still useful and much easier to complete.

What do you think?

C47D, 2020-07-14:
Sorry, I've been away from my computer for a couple of weeks. Is this feature expected to be included on the next release? I think partial decoding is crucial for embedded targets.
Also we would need to pass the repo from my profile to the LVGL profile, isn't?

kisvegabor, 2020-07-14:
No problem :slightly_smiling_face: 

It can be added independently from the releases.

I think it's better we have something working which can be simply extended later. Partial decoding is critical only on MCUs but some people use LVGL on Raspberry or similar platforms where there are plenty of resources.

If you agree yes. We can move it as `lv_lib_tjpgd` here.

C47D, 2020-07-18:
The current implementation (master branch) uses malloc, would it make sense to use the LVGL malloc wrapper instead?

kisvegabor, 2020-07-20:
Yes, it'd be great. 

Using LVGL's File system would be also a good direction. (lv_lib_lodepng still uses POSIX IO functions)

C47D, 2020-08-12:
@kisvegabor Just to let you know, there's a new JPEG decoder being released soon, not mine but I saw it on Twitter being tested.

kisvegabor, 2020-08-12:
Wow, can you link the tweet?

C47D, 2020-08-12:
This is it https://twitter.com/fast_code_r_us/status/1293304567308976132?s=19

kisvegabor, 2020-08-13:
Hm, it's not for LVGL, right?

C47D, 2020-08-14:
No, it isn't, it is just another JPEG decoder, the creator told me he is documenting it (tjpgd lacks of comments) and it is targeting low end microcontrollers. They also did a GIF decoder library recently, does the LVGL image decoder can support this?

kisvegabor, 2020-08-14:
> The creator told me he is documenting it (tjpgd lacks of comments) and it is targeting low end microcontrollers. They also did a GIF decoder library recently, does the LVGL image decoder can support this?

I've added a GIF decoder a few days ago: https://github.com/lvgl/lv_lib_gif

vinodstanur, 2020-09-04:
Hi Kisvegabor, integrating jpeg to lvgl seems to be very interesting stuff. I got a chance to play around with the tjpeg library after seeing your comment over here. I am a big fan of elm-chan since long time. 

   Yesterday I modified tjpeg library to render the jpeg from any line (for now width * block_width, for me on my test images block width is 16) without the need of full image decomp. So basically I can jd_decomp  even a small part of jpeg from the middle of it as well. (Added few additional parameters to jd structure and some changes in few functions etc). Also to make it work, we have to run jd_decomp in training mode one time after opening the image, and later on we can call jd_decomp for a particular line and it will uncompress only that line_block, this reduces the jpeg's computational complexity height/16 times in a 16 block image. 
     Also now adding a jpeg_decoded_cache in expense of additional width * 16 * byte_per_pixel, but this is expensive and can keep as optional for fast line fetching. If enabled, average decompression complexity for a single width line will be reduced by height times. This might be the best especially in systems like esp32 etc where we cannot allocate too much memory.
[image]
[image]

C47D, 2020-09-04:
This is great @vinodstanur, I didn't have time to continue this, I only had the chance to decode the whole image in one step and not in chunks.

Will you post the source of your project? Mine is two months old and it's very ugly.

kisvegabor, 2020-09-04:
@vinodstanur 
Looks great! Are you interested in contributing with your code?  If so I can open a repo for the jpeg decoder. 

I see no problem with the width*16 (or 8) pixels cache. For a 320px wide image on 16 bit color depth, it's only 10k (5k for 8px blocks) A 320x240 image would be 150k.

vinodstanur, 2020-09-04:
@kisvegabor  Yes, I am interested :)

kisvegabor, 2020-09-04:
Awesome!
Here it is: https://github.com/lvgl/lv_lib_jpg

kisvegabor, 2020-09-28:
With @vinodstanur contribution, we have a great lib to use compressed images in a fast way. 

We still need a normal JPG decoder that can be used e.g. on Raspberry without any additional scripts. @C47D, what do you think, how far is your version from it? The goal is just to decompress the whole image in a large dynamically allocated array.

C47D, 2020-09-28:
I think I only need to clean up my code a little bit.

kisvegabor, 2020-09-28:
> I think I only need to clean up my code a little bit.

That's great! Do you have time and interest to polish it?

vinodstanur, 2020-09-28:
> With @vinodstanur's contribution, we have a great lib to use compressed images in a fast way.
> 
> We still need a normal JPG decoder that can be used e.g. on Raspberry without any additional scripts. @C47D, what do you think, how far is your version from it? The goal is just to decompress the whole image in a large dynamically allocated array.

Hi @kisvegabor ,
 Thanks.
Also since sjpg is of multiple jpgs , If we consider the split factor as 1 (meaning no split), then with a very minimal changes to the lv_sjpg I can make it work with normal jpg as well but the disadvantage is that we have to allocate a complete decompressed image cache unlike the sjpg's  xres * 16 * 2 cache size.

C47D, 2020-09-28:
> > I think I only need to clean up my code a little bit.
> 
> That's great! Do you have time and interest to polish it?

Yes, but I can do it tomorrow night.

kisvegabor, 2020-09-30:
> Also since sjpg is collection of multiple jpgs , If we consider the split factor as 1 (meaning no split), then with a very minimal changes to the lv_sjpg I can make it work with normal jpg as well but the disadvantage is that we have to allocate a complete decompressed image cache unlike the sjpg's xres * 16 * 2 cache size.

Sounds good! Having 1 repo is clearly better. So if normal JPG (both as file or C array) work without any configuration with `spjg` I suggest going this way. 

@C47D don't you mind if we merge the normal JPG support to @vinodstanur's repo?

C47D, 2020-09-30:
No, I don't, this is the repo I was working on https://github.com/C47D/lv_lib_tjpgd

jesse-schein, 2020-10-06:
I'm interested in working on this where I can. I tried to use your library @C47D, but I didnt have luck using it with an ESP32. I switched some things to using PS_MALLOC to use the available 4MB of ram.

I am working on a project to get thumbnails from homeassistant via websocket, the image is base64 encoded and I'm trying to put it on the screen. I am successfully decoding it but cant seem to get LVGL to put it on the screen.

jesse-schein, 2020-10-06:
Oh I see @C47D your code does not work with variables. Maybe I will try to work on it.

C47D, 2020-10-06:
@jesse-schein hi, I was using my pc to work on the library, so my allocation wasn't properly done for embedded. I think @vinodstanur library takes that into account, I'm not sure tho.

closed 2020-11-09 by stale[bot]

kisvegabor, 2020-11-09:
The JPG library is added here:
https://github.com/lvgl/lv_lib_split_jpg