lvgl/lvgl #1549: Couldn't allocate memory

heyhachi0, 2020-06-02:
**Describe the bug**
when I use the latest master branch, it can't exit lv_init().
I found it prompt "Couldn't allocate memory" in lv_mem_alloc() while I debugging.

**To Reproduce**
NULL

**Expected behavior**
NULL

**Screenshots**
NULL

**Additional context**
NULL

**Other**
In function "lv_mem_alloc", statement "size = (size + 3) & (~0x7)" is rigth?
if size = 17, new size will be 16, not 20

pete-pjb, 2020-06-02:
Hi @heyhachi0,

It looks like this was updated only a few hours ago, I think the intention may have been this: 

size = (size + 3) & (~0x3)

It looks like the up to the minute Github source has already fixed the issue.  It may be worth grabbing it again.

Kind Regards,

Pete

heyhachi0, 2020-06-02:
> Hi @heyhachi0,
> 
> It looks like this was updated only a few hours ago, I think the intention may have been this:
> 
> size = (size + 3) & (~0x3)
> 
> It looks like the up to the minute Github source has already fixed the issue. It may be worth grabbing it again.
> 
> Kind Regards,
> 
> Pete

Thank you. I forget to update lv_conf.h.

closed 2020-06-02 by heyhachi0

pete-pjb, 2020-06-02:
No worries, glad you got it sorted... 😃