lvgl/lvgl #1551: Method for automatically applying a style to all objects of a certain type

embeddedt, 2020-06-02:
Currently, there is no API for creating a global style that automatically applies to objects. In CSS, this is how such a feature would work:

```css
button {
    background-color: blue;
}
```

All buttons would now have a blue background color, regardless of the browser's default styling.

To make theme customizations easier, we should also have a feature like this.

kisvegabor, 2020-06-02:
I agree, it'd be a great feature.

The default styles come from the themes. When an object is create `lv_theme_apply(obj, LV_THEME_BTN/LABEL/...)` is called. In this, some styles are added to the object. E.g. for a button it can be adding `style_gray`, `style_pad_large`, `style_radius_small` (just examples).

The user can add styles on top of these theme styles. To modify the default styles we can to insert a new style between the theme styles and the user styles.
The style cascade could look like this:
1. default values from LVGL
2. theme styles
3. dummy style added to every object to allow customization
4. user styles

To change the style of all buttons:
```c
lv_style_t * style= lv_theme_get_style(LV_THEME_BTN);
lv_style_set_...
lv_obj_report_style_mod(style);
```
I see a drawback: every object consumes +4 byte RAM/part (e.g. slider has 4 parts) due to the extra pointer. 

What do you think?

embeddedt, 2020-06-02:
I think a simpler API would be easier to understand and use. We already have `lv_obj_add_style` for adding an `lv_style_t` to an object; can we not also have `lv_style_add_global` to add a global `lv_style_t` for a particular object type?

kisvegabor, 2020-06-02:
Sounds good. 

But is should rather a `theme_set` function becasue the above propesed method allows only one style for this purpose. E.g. `lv_theme_set_style(LV_THEME_BTN, &my_style)`

embeddedt, 2020-06-02:
I think we should allow multiple styles to override the theme. That's how CSS works.

puzrin, 2020-06-02:
> Currently, there is no API for creating a global style that automatically applies to objects. In CSS, this is how such a feature would work:

In CSS that's a bad practice to use selectors without class.

> To make theme customizations easier, we should also have a feature like this.

IMO more common solution is to assign appropriate classes when you create widget. Then you can update class anytime as you wish.

embeddedt, 2020-06-02:
> In CSS that's a bad practice to use selectors without class.

I'm not sure that's the case if you're trying to set up a default baseline. Bootstrap's [Reboot](https://getbootstrap.com/docs/4.0/content/reboot) library is a good example of this.

As @amirgon noted in another thread, having to manually apply a class to every button is quite tedious if you want the change to apply everywhere.

puzrin, 2020-06-02:
> I'm not sure that's the case if you're trying to set up a default baseline. Bootstrap's [Reboot](https://getbootstrap.com/docs/4.0/content/reboot) library is a good example of this.

IMO that's example of another thing - how to make look the same in different browsers. Not actual for lvgl. Also note, "Reboot" is NOT intended to be changed, it's normalizer to add before everything else.

> As @amirgon noted in another thread, having to manually apply a class to every button is quite tedious if you want the change to apply everywhere.

Split problem into parts:

1) Nobody with CSS experience can suddenly wish to use selectors without classes for interface design.
2) If it's difficult to apply classes now - try to invent how to apply those with ease, instead of fall into "bad practice". Via macros and so on.

amirgon, 2020-06-02:
> As @amirgon noted in another thread, having to manually apply a class to every button is quite tedious if you want the change to apply everywhere.

CSS analogy can be Descendant Selectors.
Something like:
```css
.btn_class >  btn
 {
    margin: 10px;
    ...
 }
```

```html
<div id="parent" class="btn_class">
  <div id="child>
    ...  
      <btn ... />
    ...
  </div>
</div>
```
In such case, the `btn_class` need to be specified only on the parent, and the style would apply to all descendants in the hierarchy according to the selection rule (`btn` elements in this case)

Since css is declarative and lvgl is functional, this would need to be converted to some function call, called for each of the parent objects. Something like:
```c
lv_apply_descendant_selector(parent, <some rule to select btn>, &my_style)
```

puzrin, 2020-06-02:
I'd suggest to read BEM https://en.bem.info/methodology/css/. That's one of possible "good" approach how to assign and name classes.

In short - every kind of "unique look" object has it's own unique class + optional modifiers.

kisvegabor, 2020-06-04:
Themes are meant to be the analogy of system-level appearance. E.g. How a HTML `<button>` or `<input>` should look like by default (depends on browser and OS).  So you are not supposed to modify them but you can create own theme and customize it as you wich. 

There is no direct analogy e.g. for Bootstrap because it provides classes to assign them to objects. E.g. `.btn` needs to be assigned every button you create. 

Also there no analogy for `button{}` or `input{}` classes and selectors. 

We don't need to strictly follow how CSS/HTML works but we can and we have to tailor it to our needs. 

So let's decide whether we need selectors. Personally I don't think so. You can create a class and add it. 

Do we really need `button{}` classes? It's doable and much easier than creating a new theme. 

E.g. bootstrap provides classes to apply them as you need. Shall our themes do the same (besides setting the default appearance)? E.g. `lv_theme_get_style(LV_THEME_STYLE_BTN_RED)`.

embeddedt, 2020-06-25:
It seems that the discussion is happening in https://github.com/lvgl/lv_binding_micropython/issues/75, so I'll close this issue. This feature will probably get added to 7.2 (at the latest).

closed 2020-06-25 by embeddedt