lvgl/lvgl #1552: No tick labels on chart

ibutra, 2020-06-02:
**Describe the bug**

I am adding a chart with a series and add tick text to it. The tick marks are shown but not the labels itself. If I increase the y tick length to a great amount (200 or so) the y labels are shown but overlayed on the graph and somewhat opaque. This does not work for x tick labels. If I set their lenght to maximum still no label is shown. 
The chart is inside a container. But even if I just add it to the screen no labels are shown.
Only custom style I applied is size set to 0 on LV_CHART_PART_SERIES. Even without this no labels are shown.
Tried adjusting padding and margin to no avail.
All my fonts are set to `&lv_font_montserrat_30`

**To Reproduce**
```

lv_obj_t* chartscreen_create(lv_obj_t* parent) {
    if (parent == NULL) {
        parent = lv_scr_act();
    }
    lv_obj_t* object = lv_cont_create(parent, NULL);
    lv_cont_set_layout(object, LV_LAYOUT_COLUMN_LEFT);
    lv_cont_set_fit(object, LV_FIT_MAX);

    int height = lv_obj_get_height_grid(object, 3, 1);
    int width = lv_obj_get_width_fit(object);

    chart0 = lv_chart_create(object, NULL);
    lv_chart_set_point_count(chart0, POINT_COUNT);
    lv_chart_set_update_mode(chart0, LV_CHART_UPDATE_MODE_CIRCULAR);
    lv_chart_set_range(chart0, -10, 30);
    lv_chart_set_div_line_count(chart0, 0, 0);
    lv_chart_set_y_tick_texts(chart0, "-10\n0\n10\n20\n30", 1, LV_CHART_AXIS_DRAW_LAST_TICK);
    lv_chart_set_x_tick_texts(chart0, "a\nb\nc\nd", 10, LV_CHART_AXIS_DRAW_LAST_TICK);
    lv_obj_set_size(chart0, width, height);
    lv_obj_add_style(chart0, LV_CHART_PART_SERIES, &style_graph);
    series0 = lv_chart_add_series(chart0, LV_COLOR_BLUE);
    lv_chart_init_points(chart0, series0, 0);

    chart1 = lv_chart_create(object, chart0);
    lv_chart_set_range(chart1, 0, 10);
    lv_chart_set_y_tick_texts(chart1, "0\n1\n2\n3\n4\n5", 1, LV_CHART_AXIS_DRAW_LAST_TICK);
    series1 = lv_chart_add_series(chart1, LV_COLOR_BLUE);
    lv_chart_init_points(chart1, series1, 0);

    chart2 = lv_chart_create(object, chart0);
    lv_chart_set_range(chart2, 0, 250);
    series2 = lv_chart_add_series(chart2, LV_COLOR_BLUE);
    lv_chart_init_points(chart2, series2, 0);

    return object;
```
The created container is  added to the main screen and resized.
But even If I just add one of the charts to the screen directly the problem persists.

**Expected behavior**

I expect to see tick labels.

**Screenshots**
[image]

**Additional context**
(none right now)

pete-pjb, 2020-06-02:
Hi @ibutra,

I had a similar experience when updating my project to version 7.0 to set the colour of the text I used:

```c
lv_obj_set_style_local_text_color(chart0, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_BLACK);
```

To show labels I added padding as follows:

```c
	lv_obj_set_style_local_pad_top(chart0, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 10);
	lv_obj_set_style_local_pad_bottom(char0, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 60);
	lv_obj_set_style_local_pad_left(chart0, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 80);
	lv_obj_set_style_local_pad_right(chart0, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 35);
```

You may need to change to different values for your configuration as your font is much bigger than mine.

Just in case you want to change the font I used this:
```c
	lv_obj_set_style_local_text_font (chart0, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, &lv_font_roboto_8_p);
```

I hope this helps you out.

Kind Regards,

Pete

ibutra, 2020-06-02:
Hi,

Thank you. That works. Will look into wether this can’t be automated.

closed 2020-06-02 by ibutra