lvgl/lvgl #1555: vs compile error.

Samchy, 2020-06-04:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

git clone the newest branch & compile. vsproject throw out an error which abount missing the lv_demo_printer_icon_btn_bg.c

solution: 1. remove the btn_bg.c 
               2. add btn_bg_1.c btn_bg_2.c btn_bg_3.c btn_bg_4.c 

**To Reproduce**

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally this should work in the PC simulator unless the problem is specific to one platform.

**Expected behavior**

**Screenshots**
[image]

[image]

**Additional context**

Add any other context about the problem here.

kisvegabor, 2020-06-04:
Could you send a pull request with updated project?

closed 2020-06-04 by Samchy

Samchy, 2020-06-04:
It's my own pull steps cause this issue.
My step as below:
A) git clone https://github.com/littlevgl/lv_sim_visual_studio_sdl.git
B) cd lv_sim_visual_studio_sdl/visual_studio_2017_sdl
C) git clone https://github.com/littlevgl/lv_drivers.git
D) git clone https://github.com/littlevgl/lv_examples.git
E) git clone https://github.com/littlevgl/lvgl.git

Actually just need one step is fine and not cause this issue. 
git clone --recursive https://github.com/littlevgl/lv_sim_visual_studio_sdl.git

So sorry to borther you.

kisvegabor, 2020-06-04:
Thank for the feedback!