lvgl/lvgl #1557: Init issue with lv_disp_ll

mpikkemaat, 2020-06-04:
When calling lv_init, I experienced unexpected behaviour. After some debugging I found the following:

In lv_init, the default style is set through the LV_THEME_DEFAULT_INIT macro, which calls lv_theme_material_init. This function ends with calling lv_report_obj_style_mod to update all existing objects. To do this, it looks at the lv_disp_ll list for displays, and stops on a null pointer. The _lv_ll_init of this display list however is done after setting up the theme in lv_init. I'm guessing on my system the memory is not set to zero on boot, and therefore created a problem

The solution for me was very simple: move line 152 in lv_core.c:

    _lv_ll_init(&LV_GC_ROOT(_lv_disp_ll), sizeof(lv_disp_t));

to line 143, before setting the style

embeddedt, 2020-06-04:
I've applied your change to `master`.

It might be worth looking into why the memory is not being zeroed on your platform.

closed 2020-06-04 by embeddedt

mpikkemaat, 2020-06-04:
That was quick! I encountered a similar issue in lv_init, where on cache init, the code checks if the cache memory address is Null. If not, it tries to free that block before malloc-ing a new one. I was able to solve it by adding a null init on line 158:

    _lv_img_cache_array=NULL;
    _lv_img_decoder_init();
    lv_img_cache_set_size(LV_IMG_CACHE_DEF_SIZE);

although it might be nicer to do so where the variable was declared.

On the zero-init subject: i think the heap in not is the bss linker section. Isn't the heap usuallu uninitialized?

kisvegabor, 2020-06-04:
Although it's a little bit tricky but `_lv_img_cache_array` is a simple global variable. LVGL is prepared to support garbage collection if used in a binding, therefore some variables are marked as garbage collector roots (`GC_ROOTS`).

Here is how `_lv_img_cache_array` is finally declared as a global variable:
- https://github.com/lvgl/lvgl/blob/4f3dac6477f8387435fdcb49e40c2c448d194664/src/lv_misc/lv_gc.h#L38
- https://github.com/lvgl/lvgl/blob/4f3dac6477f8387435fdcb49e40c2c448d194664/src/lv_misc/lv_gc.h#L48
- https://github.com/lvgl/lvgl/blob/4f3dac6477f8387435fdcb49e40c2c448d194664/src/lv_misc/lv_gc.c#L34

commit 4f3dac64 (2020-06-04): Fix #1557: use of uninitialized linked list