lvgl/lvgl #1558: Looking for new maintainer

kisvegabor, 2020-06-04:
Hi,

First of all, I'd like to thank you for the current maintainers of LVGL. Namely @embeddedt, @amirgon, @C47D, @pablo2048, @puzrin and @MCF. 

You do an amazing job of maintaining your repositories and help a lot of people! :clap: :clap: :clap: 

Right now we have 2 maintainers for the core lvgl repo: @embeddedt  and me. 
I'd like to ask @pete-pjb if he feels like joining us to maintain the lvgl repository. It'd mean getting write permission to this repository.  :slightly_smiling_face: 

---

On the Forum as an experiment, we tried the  "Avatar Flair" feature of the engine. It can add a logo to the avatar if the user belong to a group.  You can see it for @embeddedt and me.  
[image]

Is it okay for you if you (the maintainers) were added to this group to get the logo on your avatar?

C47D, 2020-06-04:
I would like to thank all the contributors on the ESP32 port, specially to @davidjade, @an-erd and @barbiani.

If any of you are interested in helping maintaining the ESP32 let us know, the work done to support LVGL v7 ended up in a lot more than I anticipated but I think we did a great job.

kisvegabor, 2020-06-04:
@pete-pjb 
Glad to hear that you are interested! :slightly_smiling_face: 

There are no strict expectations like you need to review X PRs and send Y patches every week. It means - if you have time and liking for it - you have the opportunity to merge PRs, send simple fixes directly to lvgl. With @embeddedt we usually ask each other to review the nontrivial PRs and still open issues if we find complex bugs. It seemed working so far. 

Although we try to make decisions in a democratic way (i.e. choose what makes more people happy) when we need to make decisions your word has more impact. That's because we believe that you are great both professionally and humanly and taking your though into account would help a lot to move LVGL forward. That's why we asked you to become a maintainer. :slightly_smiling_face:

kisvegabor, 2020-06-05:
@pete-pjb 

Thanks, I've sent the invitation.

pete-pjb, 2020-06-05:
Thanks @kisvegabor, I have accepted the invitation, also received your contact info okay.

Kind Regards,

Pete

kisvegabor, 2020-06-26:
As the original goal was reach I close this issue :slightly_smiling_face:

closed 2020-06-26 by kisvegabor