lvgl/lvgl #1564: Want to implement an effect like this with LVGL, Do you have any suggestion?

incity, 2020-06-09:
[6c875aa3fa6dc5211d59bce308f80acd.zip](https://github.com/lvgl/lvgl/files/4751729/6c875aa3fa6dc5211d59bce308f80acd.zip)

two measuring scale, 1 vertical and 1  horizontal，vertical one scroll with page's vertical scroll and vice versa, horizontal one scroll with page's horizontal scroll and vice versa.
Do you have any suggestion for me @kisvegabor 

thank you very much!!!

xennex22, 2020-06-09:
I have a scale widget that would work I think when placed on a page.

https://forum.lvgl.io/t/widget-to-draw-a-linear-scale/1970

https://github.com/xennex22/lvgl/blob/master/src/lv_widgets/lv_linearscale.c

incity, 2020-06-10:
@xennex22 great thanks, by the way do u have any suggestion for my requirement??

incity, 2020-06-10:
@xennex22 Hi, advice you to use 4 spaces instead of TABs.

incity, 2020-06-10:
[image]
also the coding style. cloud change to follow the main branch? thanks

kisvegabor, 2020-06-10:
I have no good idea now to solve it with the public API of LVGL. 
However, it's definitely doable if you dig a little bit deeper.

I'd create a new signal function for a `lv_page`'s scrollable part and check `LV_SIGNAL_COORD_CHG`. When the coordinates are changed you can set the position of the rulers accordingly. 

See [here](https://blog.lvgl.io/2018-12-13/extend-lvgl-objects) how the signal functions work.

incity, 2020-06-11:
oh, sounds good, i'll wait you. thanks.

embeddedt, 2020-06-11:
> oh, sounds good, i'll wait you. thanks.

I think @kisvegabor is suggesting that you can implement it yourself by overriding the signal function of the page. :slightly_smiling_face:

kisvegabor, 2020-06-11:
Yes :slightly_smiling_face:

incity, 2020-06-12:
@kisvegabor oh im confused, you said you would create new signal function...emmmmm..Did I get it wrong

embeddedt, 2020-06-12:
@incity He was pointing out what he would do if he was needing to implement this effect for himself.

kisvegabor, 2020-06-14:
@incity 
Sorry for the confusion. By "I'd create a new signal function ..." I meant "If I were you I'd create a new signal function ..."

drifting1024, 2020-06-17:
English skill++

closed 2020-06-18 by incity

kisvegabor, 2020-06-18:
> English skill++

@drifting1024 I'm not sure whether you were referring to @incity or me but insulting others for oversights or typos or anything else is not welcome here. 

In this community, we help each other, discuss things in a civil way,  and kindly point out if someone makes a mistake. 

Please read our [code of conduct](https://github.com/lvgl/lvgl/blob/master/docs/CODE_OF_CONDUCT.md).

drifting1024, 2020-06-18:
you got me wrong, you're not funny