lvgl/lvgl #1567: lv_obj_align_origo leading to crash

weixiongmei, 2020-06-10:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Using lv_obj_align_origo to set alignment to an existing lv_obj_t can leading to crash. LVGL revision V7.0

**To Reproduce**

1. Convert the Font Awsome font fine fa-regular-400.ttf for Name:lv_font_symbol Size:64, 2bbp and Enable font compression

2. Using the code below:
```C
  lv_style_t style_symbol;
  lv_style_init(&style_symbol);
  lv_style_set_text_font(&style_symbol, LV_STATE_DEFAULT, &lv_font_symbol);

  lv_obj_t *label_large = lv_label_create(lv_scr_act(), NULL); 
  lv_label_set_text(label_large, "\uf1eb");
  lv_obj_add_style(label_large, LV_LABEL_PART_MAIN, &style_symbol);
  lv_obj_align_origo(label_large, NULL, LV_ALIGN_CENTER, 0, 0);

  lv_obj_t *label_version = lv_label_create(lv_scr_act(), NULL); 
  lv_label_set_long_mode(label_version, LV_LABEL_LONG_SROLL_CIRC);
  lv_obj_set_width(label_version, 150);
  lv_label_set_text(label_version, "Using lv_obj_align_origo will leading to a crash.");
  lv_obj_align_origo(label_version, label_large, LV_ALIGN_OUT_BOTTOM_MID, 0, 10); //Comment this line will runs with no issue
  lv_label_set_anim_speed(label_version, 64);
```

**Expected behavior**

The above code will crash at the line 1070 of the lv_style.c 
**"while(style->map[i] != _LV_STYLE_CLOSEING_PROP) {"**
if the font of the style that set for the label_large is smaller, such as size: 16, or the second arg is set to NULL like 
**lv_obj_align_origo(label_version, NULL, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);** 
the program will not going to crash.

kisvegabor, 2020-06-10:
Hi,

Please attach your font to reproduce the issue.

weixiongmei, 2020-06-11:
[lv_font_default_normal.zip](https://github.com/lvgl/lvgl/files/4762098/lv_font_default_normal.zip)
Here's the normal font i used in the above example, and just noticed that the crash only happens when the objects overlapping. So, if moving one of the object farther, the crashing issue will be solved.

kisvegabor, 2020-06-12:
Oh, I've found something. `lv_style_t style_symbol;` should be static. 
See: https://docs.lvgl.io/v7/en/html/overview/style.html#initialize-styles-and-set-get-properties

weixiongmei, 2020-06-12:
> Oh, I've found something. `lv_style_t style_symbol;` should be static.
> See: https://docs.lvgl.io/v7/en/html/overview/style.html#initialize-styles-and-set-get-properties

Thank you so much, the problem is solved now, sorry for my own mistake~~~

kisvegabor, 2020-06-12:
No problem :slightly_smiling_face:

closed 2020-06-12 by kisvegabor