lvgl/lvgl #1568: Rotary encoder widget

kisvegabor, 2020-06-10:
Hi,

I'm opening this from [Add a rotary knob object](https://forum.lvgl.io/t/add-an-rotary-knob-object/1866/3) forum topic to discuss and implement this widget.

Feel free to send images and describe what would you expect from this widget. 
It's even better if you are interested in participating in the implementation! :slightly_smiling_face:

curly-brace, 2020-06-12:
my suggestions:
* two types of operation modes:
  * rotary selector: widget works like a rotary switch which will have labels around with captions, describing selectable options. like in washing machines of cooking stoves for example.
  * incremental encoder: in which it will have a label in the center of its widget, showing current value. and user will be able to set the increment value for one tick. and this means that encoder will have discrete rotation, not smooth, which should help with precision.
* ability to move the functionality of center label to external label which may be even located on another screen. the feature is that this label will update automatically when encoder rotates
* optional increment/decrement buttons shown
  * above and below center label
  * above and below encoder circle
  * on the sides of the widget

kisvegabor, 2020-06-14:
> rotary selector: widget works like a rotary switch which will have labels around with captions, describing selectable options. like in washing machines of cooking stoves for example.

So there is a circle with radial labels and e.g. the top label is marked as selected? If so I can imagine these additional features:
- allow using images instead of labels too
- allow applying effects on labels/images depending on how far are they from the selected: scale down, blur, opacity, darken, etc.
- allow setting the maximum visible option count
- set the degree between options

Finally, it'd allow creating something like [this](https://www.youtube.com/watch?v=kXfMrvpdp9M).

> incremental encoder: in which it will have a label in the center of its widget, showing current value. and user will be able to set the increment value for one tick. and this means that encoder will have discrete rotation, not smooth, which should help with precision.

Can you show a good looking example for this?

> ability to move the functionality of center label to external label which may be even located on another screen. the feature is that this label will update automatically when encoder rotates

We can send a `LV_EVENT_VALUE_CHG` event where you can set any label.

> optional increment/decrement buttons shown. needed for precise value set. for example main increment is 1 and buttons can increment in 0.5

I think it's hard to make it flexible. I mean a graphic designer probably will design something different than we support out of the box. So in my opinion adding API for external adjustment is enough and we can leave it to the user to implement the extra buttons as he needs. 

What do you think?

acumartini, 2020-06-16:
I am needing this feature for a project I am working on and have started an implementation. I am planning to use `led`'s surrounding a knob that is a child of `btn` to indicate the current value within a range. The reason for being a descendent of `btn` it to support the BUTTON_STATE of the encoder input, which is supported by most digital rotary encoders. However, `btn` functionality will be optional via type setting.

embeddedt, 2020-06-16:
@acumartini That's great! I think we should also try to make it useful for a round slider, as suggested by @peejii73 in [this forum post](https://forum.lvgl.io/t/new-type-of-slider-any-ideas/1892).

acumartini, 2020-06-16:
I agree with your comment on extending the gauge object. Having slider functionality would be awesome for touch screen input. I was struggling with how to incorporate that into my design. 

It's possible we might be talking about two different implementations here... perhaps they can live under one roof.

embeddedt, 2020-06-16:
Are you planning to create several LED objects or just draw several LEDs? I'd advise against creating multiple real LED objects as that will significantly increase the memory requirements per knob.

acumartini, 2020-06-16:
Great point! I assumed drawing them would be the way to go, thnx for the confirmation.

embeddedt, 2020-06-16:
I think you're probably right about separate implementations for the slider vs the encoder. Looking at the features available in 7.x, it would probably be more practical to either add a circular mode to `lv_slider` itself or create a new object that inherits from the `lv_arc` object and adds a knob.

> I assumed drawing them would be the way to go, thnx for the confirmation.

No problem. Extending objects (like what you're doing with `lv_btn`) is not an issue, as it doesn't require any more memory beyond the additional `ext_attr` information. Combining objects is more costly, because you have the overhead of more `lv_obj_t` structures, plus the extra rendering and clipping overhead because they are "separate" objects.

barbiani, 2020-06-16:
Disabling its touch and making the knob small/colorful turns it onto a nice
indicator.

kisvegabor, 2020-06-16:
Let's summarize what was mentioned so far:
1. Encoder like widget: as you turn it, the position change (relative) is reported
2. Circular selector: text/images around a circle and one of them is selected
3. Touch feature to gauge: I think it should be added to `lv_linemeter` instead as the gauge has needles so it looks like an output device.
4. Add a knob to `lv_arc` or create a new object.  (`lv_arcslider`?)

Have I missed something?

acumartini, 2020-06-17:
@kisvegabor Good summary. After digging into this more I believe the most efficient path to the type of feature I need is #4, extending `lv_arc` and adding a knob/btn. Made some pretty progress on this path yesterday.

Adding the slider as a `type` to the `lv_rotary` I am working on would be a natural fit for folks who want to emulate a digital encoder in the UI without the hardware counterpart for control.

Side note: I think it's important for the rotary object to have an idea of range similar to `lv_slider` so that one can control the "sensitivity" of the knob. So range will already be within the worldview of the object. Because you can only extend one parent (pls correct me if I am wrong), we get the most bang-for-the-buck extending `lv_arc`, then some copy pasta for range manipulation <= this is where I am currently at with the impl.

Likely I will need some help with adding a knob to the actual arc (assuming my code meets your standards), but cross that bridge when we get there.

kisvegabor, 2020-06-18:
@acumartini 
Feel free to send a PR with the current stage of your implementation. It's easier to discuss the details if we see your code. 

I agree, to extend `lv_arc` and manually add the knob and range handling.

acumartini, 2020-06-19:
@kisvegabor good call. PR open with my current progress. Will link a simple example program asap. But here is an image of what I have so far running on my display:
[image]

kisvegabor, 2020-06-19:
Thank you! I'm just checking the PR.

curly-brace, 2020-06-20:
Here is the first my suggestion of rotary switch mode
[image]

And this is a incremental encoder with discrete steps (not smooth rotation). Imagine that each step is 0.5.
[image]

kisvegabor, 2020-06-22:
@curly-brace 
Thank you.

Let's see, how your and @acumartini's widget align with the [summary above](https://github.com/lvgl/lvgl/issues/1568#issuecomment-644676920).

@acumartini. As you mentioned it's # 4. I wonder if it'd be reasonable to add it directly into `lv_arc`. With negative padding `LV_ARC_PART_BG` can be used to add the central circle. 

@curly-brace So the gist is to arrange text or images around a circle and make one highlighted. It needs 2 modes of operation: range limited and infinite (restart from the first option on "overflow".   Do we really need the yellow circle? Can't we just change the color of the selected text/image?

curly-brace, 2020-06-24:
yellow circle will give user a feedback that he is actually rotating that knob. just thinking about UX here. it may be not as contrast tho.

kisvegabor, 2020-06-24:
> yellow circle will give user a feedback that he is actually rotating that knob. just thinking about UX here. it may be not as contrast tho.

I see. It makes sense.

kisvegabor, 2020-07-21:
No stale.

Related PR: #1592 

Besides I'm interested in working on "Circular selector: text/images around a circle and one of them is selected".

acumartini, 2020-07-21:
@kisvegabor 
This would be a great addition. Likely a `type` enum for arc styling.

kisvegabor, 2020-07-22:
Actually I meant it for a new widget. In this, the images/text are rotated around a  circle and always the top (or left or right etc) is selected. Effects can be applied based on the position of the image. E.g. zoom which are closer to the selected area.

kisvegabor, 2020-08-17:
#1592 is merged and I right now I have no time to implement the mentioned widget. 

So to keep the issue list clean I close this now.

Thank you very much for your ideas and contribution! 

If you have any ideas or find bugs please open a new issue for them.

closed 2020-08-17 by kisvegabor