lvgl/lvgl #1569: Can't set the color of btn in pressed state and checked_pressed state

Xiaolin-He, 2020-06-10:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**
I set a different color, such as LV_COLOR_RED, for a button in the pressed state. But when I press the button, it's color isn't the color I set. 

**To Reproduce**
    static lv_style_t style1;
    static lv_style_t style2;
    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_pos(btn1, 10, 10);
    lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_pos(btn2, 10, 110);

	lv_style_init(&style1);
        lv_btn_set_checkable(btn1, true);
	lv_style_set_bg_color(&style1, LV_BTN_STATE_CHECKED_PRESSED, LV_COLOR_RED);
	lv_style_set_bg_color(&style1, LV_BTN_STATE_CHECKED_RELEASED, LV_COLOR_YELLOW);

        lv_style_init(&style2);
	lv_style_set_bg_color(&style2, LV_BTN_STATE_PRESSED, LV_COLOR_RED);
	lv_style_set_bg_color(&style2, LV_BTN_STATE_RELEASED, LV_COLOR_YELLOW);

	lv_obj_add_style(btn1, LV_BTN_PART_MAIN, &style1);
        lv_obj_add_style(btn2, LV_BTN_PART_MAIN, &style2);

**Expected behavior**

The button shows the right color in pressed state that was set in code.

**Screenshots**
1. Default state
[image]
2. Pressed state
[image]
3. Released state
[image]

**Additional context**
no

embeddedt, 2020-06-10:
The generic style functions should be used with the [generic states](https://docs.lvgl.io/v7/en/html/overview/style.html#states), not the button-specific states (which are meant to be used with button-specific functions).

Xiaolin-He, 2020-06-12:
Thanks. After changing the state macros to generic states, it works well.

embeddedt, 2020-06-12:
Glad to hear it!

closed 2020-06-12 by embeddedt