lvgl/lvgl #1571: How to keep showing the scrollbar of list, please？

lukashiaa, 2020-06-12:
I have my best... But I didn't see it. I have to configure it as follows, please help me./(ㄒoㄒ)/~~
    lv_area_t ext_area;
    lv_page_ext_t * ext = lv_obj_allocate_ext_attr(new_page, sizeof(lv_page_ext_t));
   ext_area.x1 = 315;
    ext_area.x2 = 319;
    ext_area.y1 = 40;
    ext_area.y2 = 239;
    ext->sb.ver_area = ext_area;
    ext->sb.hor_draw = 1;
    ext->sb.ver_draw = 1;

    ext->sb.style    = &lv_style_pretty;
    ext->sb.mode     = LV_SB_MODE_ON;

curly-brace, 2020-06-12:
why don't you use a page widget directly? like
`lv_obj_t* page = lv_page_create(lv_scr_act(), NULL);`

lukashiaa, 2020-06-12:
> 
> 
> why don't you use a page widget directly? like
> `lv_obj_t* page = lv_page_create(lv_scr_act(), NULL);`

Emm.. I want some features for the list, and I don't want some features for the page.

pete-pjb, 2020-06-12:
### All enhancement requests or questions should be directed to the Forum.

We use GitHub issues for development related discussions.

Please ask your question in the [Forum](https://forum.littlevgl.com/) to make it visible for a larger community. You can simply log in with your GitHub account.

Thank you.

closed 2020-06-12 by pete-pjb

embeddedt, 2020-06-12:
There is no need to manually allocate `ext` structures. I think [`lv_list_set_scrollbar_mode`](https://docs.lvgl.io/v7/en/html/widgets/list.html#_CPPv426lv_list_set_scrollbar_modeP8lv_obj_t19lv_scrollbar_mode_t) will do what you want.

lukashiaa, 2020-06-12:
> ### All enhancement requests or questions should be directed to the Forum.
>
> We use GitHub issues for development related discussions.
>
> Please ask your question in the [...]

Oh, i git it. I'm really sorry.

embeddedt, 2020-06-12:
> I'm really sorry.

No problem! It's nothing personal, just our project policy. :slightly_smiling_face: