lvgl/lvgl #1572: lv_btnmatrix : hidden button fires event

JefkeB, 2020-06-12:
**The problem**
I'm using the lv_btnmatrix as a keypad to enter a value. 
Some values need a decimal separator and some don't. 
So the dot is hidden or not depending on the mode the keypad is in. 
When the dot is hidden and you 'press' at the location where the button normally is the first time the previous pressed button will be sent in the event.

**Test**
Use the lv_ex_btnmatrix_1 from the examples with the simulator. 
Add following line 
`   lv_btnmatrix_set_btn_ctrl(btnm1, 1, LV_BTNMATRIX_CTRL_HIDDEN);`
before 
    `lv_btnmatrix_set_btn_ctrl(btnm1, 11, LV_BTNMATRIX_CTRL_CHECK_STATE);`

**Expected behavior**
Would expect 'not' to receive the LV_EVENT_VALUE_CHANGED event for a hidden button

**Screenshots**
pressing the '3' and after it 'press' at the location the 2 button is expected to be.

[image]

**my solution**
for now solved by adding
`ext->btn_id_act = LV_BTNMATRIX_BTN_NONE;`

[image]

I suppose there is better way to this but for now it solves the problem.

btw a big THANKS for this great library

Jef

kisvegabor, 2020-06-14:
Hi Jef,

Thanks for the report. 
Iv fixed it here: 9260f9e0

Is it working well for you too?

JefkeB, 2020-06-15:
Hi,

Yes it is solved.
Thank you !

closed 2020-06-15 by JefkeB

kisvegabor, 2020-06-15:
Great, thanks for the feedback!