lvgl/lvgl #1573: Can't position obect

kadirsevil, 2020-06-12:
I use 480x272 lcd screen.

In the code below when there is no bottom object everything works fine, front object is in the middle of the x axis, but when i add bottom object, right end of the front object shifts to the left by app. 10px maybe. I could not figured it out why. Can someone help me?

`tab_view = lv_tabview_create(lv_scr_act(), NULL);
  lv_tabview_set_btns_pos(tab_view, LV_TABVIEW_TAB_POS_NONE);

  tab_list[0] = lv_tabview_add_tab(tab_view, "tab0");
  lv_style_init(&tab0_style);
  lv_style_set_bg_opa(&tab0_style, LV_STATE_DEFAULT, 255);
  lv_style_set_bg_color(&tab0_style, LV_STATE_DEFAULT, lv_color_hex(0x2f3243));
  lv_obj_add_style(tab_list[0], LV_LABEL_PART_MAIN, &tab0_style);
  lv_page_set_scrlbar_mode(tab_list[0], LV_SCRLBAR_MODE_OFF);

  static lv_style_t bottom_style;
  lv_style_init(&bottom_style);
  lv_style_set_radius(&bottom_style, LV_STATE_DEFAULT, 0);
  lv_style_set_border_width(&bottom_style, LV_STATE_DEFAULT, 0);
  lv_style_set_bg_opa(&bottom_style, LV_STATE_DEFAULT, 255);
  lv_style_set_bg_color(&bottom_style, LV_STATE_DEFAULT, lv_color_hex(0xE9EEF4));

  lv_obj_t * bottom = lv_obj_create(tab_list[0], NULL);
  lv_obj_add_style(bottom, LV_OBJ_PART_MAIN, &bottom_style);
  lv_obj_set_x(bottom, 5);
  lv_obj_set_y(bottom, 90);
  lv_obj_set_width(bottom, 470);
  lv_obj_set_height(bottom, 177);

  static lv_style_t front_style;
  lv_style_init(&front_style);
  lv_style_set_radius(&front_style, LV_STATE_DEFAULT, 10);
  lv_style_set_border_width(&front_style, LV_STATE_DEFAULT, 0);
  lv_style_set_bg_opa(&front_style, LV_STATE_DEFAULT, 255);
  lv_style_set_bg_color(&front_style, LV_STATE_DEFAULT, lv_color_hex(0xffffff));

  lv_obj_t * front = lv_obj_create(tab_list[0], NULL);
  lv_obj_add_style(front, LV_OBJ_PART_MAIN, &front_style);
  lv_obj_set_x(front, 20);
  lv_obj_set_y(front, 31);
  lv_obj_set_width(front, 440);
  lv_obj_set_height(front, 210);`

embeddedt, 2020-06-12:
It's because you're creating the object on a tab page, which by default tries to position child objects itself.

You can try setting `lv_page_set_scrl_layout(tab_list[0], LV_LAYOUT_OFF)` to turn off the automatic layout.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-06-12 by embeddedt

kadirsevil, 2020-06-12:
Hi embeddedt,

`lv_page_set_scrl_layout(tab_list[0], LV_LAYOUT_OFF)` did not work

kisvegabor, 2020-06-15:
As @embeddedt mentioned please post this question in the forum. Don't forget to mention which version of LVGL you use. (see lvgl.h)

kisvegabor, 2020-06-16:
Meanwhile, I examined this issue and it turned out you see the problem because you are trying to position the objects too close to the edge of tab and the tab becomes scrollable. Try to comment out the  `lv_page_set_scrlbar_mode(tab_list[0], LV_SCRLBAR_MODE_OFF);` line to see it.

The limit when the scrollbars appear is the padding of the scrollable part.

Adding `lv_page_set_scrollable_fit(tab_list[0], LV_FIT_PARENT);` disable automatic addition of scrollbars.

I've update lvgl to allow "auto-expanding" only to the right and bottom. So with the latest `master` your code should work as it is.