lvgl/lvgl #1575: Memory access issue during LVGL initialization `lv_init()`

rafaelcaricio, 2020-06-13:
**Describe the bug**

I imagine there is some issue to dereferencing uninitialized memory in the library bootstrap code `lv_init`.

**To Reproduce**

Code same:
```c
lv_init();
```

The error starts in this line: https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_obj.c#L160

**Expected behavior**

Library should initialize without errors.

**Screenshots**
The debugging session.

[image]

Where the error happens:
[image]

**Additional context**

I'm using latest master branch version `1ca1934d`. The target is a [nRF52832 device](https://wiki.pine64.org/index.php/PineTime#Specifications). 
I'm calling from Rust code, but that shouldn't matter for this issue.
Checking the registers in GDB:
```
print/x *(uint8_t*)0xE000ED29
$1 = 0x82
```

rafaelcaricio, 2020-06-13:
Previously, I was hitting this issue #1557 which was causing same exception code. That's why I updated to use latest `master`.

embeddedt, 2020-06-13:
Can you confirm that `_lv_img_cache_array` is zero before `lv_init` is called? Maybe you have [the same problem](https://github.com/lvgl/lvgl/issues/1557#issuecomment-638886762) (although I'd expect Rust to handle this properly).

rafaelcaricio, 2020-06-13:
For some reason, I cannot reproduce it anymore. :thinking: The only difference is that I've updated to the latest Rust nightly release (`1.46.0-nightly (feb3536eb 2020-06-09)`).

rafaelcaricio, 2020-06-13:
@embeddedt it is happening again now. :cry: 

The `_lv_img_cache_array` is not zeroed.

rafaelcaricio, 2020-06-13:
Someone else was having the same issue. https://github.com/lvgl/lvgl/issues/1302 :thinking:

embeddedt, 2020-06-13:
> The `_lv_img_cache_array` is not zeroed.

I think that would mean that your `.bss` area is not being zeroed. There's probably memory corruption happening somewhere.

rafaelcaricio, 2020-06-13:
Ok, so it is some issue with my platform. Thanks for the pointer!

closed 2020-06-13 by rafaelcaricio

rafaelcaricio, 2020-06-13:
If someone from internet gets here from searches, [this section of the embedonomicon](https://docs.rust-embedded.org/embedonomicon/main.html#life-before-main) will be helpful.

rafaelcaricio, 2020-06-14:
I found the solution, it is a bug in `cortex-m-rt`, described here: https://github.com/rust-embedded/cortex-m-rt/issues/267#issue-609383169

kisvegabor, 2020-06-15:
Thank you, for the link!