lvgl/lvgl #1576: All cursors of textarea are blinking at the same time

weixiongmei, 2020-06-14:
```C
lv_obj_t * textarea1 = lv_textarea_create(lv_scr_act(), NULL);
lv_textarea_set_one_line(textarea1, true);

lv_obj_t * textarea2 = lv_textarea_create(lv_scr_act(), NULL);
lv_textarea_set_one_line(textarea2, true);
lv_obj_set_pos(textarea2, 0, 60);
```
I added two textareas to the screen, but the cursors of the two textarea are both blinking at the same time, wondering is this a bug?

embeddedt, 2020-06-14:
No; this is intended behavior. You should [hide the cursor](https://docs.lvgl.io/v7/en/html/widgets/textarea.html#hide-the-cursor) on both of them and [attach a keyboard](https://docs.lvgl.io/v7/en/html/widgets/keyboard.html#assign-text-area) to one when it is clicked.

weixiongmei, 2020-06-14:
@embeddedt , got it ,thank you~~~

kisvegabor, 2020-06-15:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2020-06-15 by kisvegabor