lvgl/lvgl #1577: Roller Error

lilengqiang, 2020-06-15:
When I use roller and airal.ttf fonts, it shows errors and the last string does not align.

lilengqiang, 2020-06-15:
[image]

lilengqiang, 2020-06-15:
[roller_error.zip](https://github.com/lvgl/lvgl/files/4777940/roller_error.zip)

lilengqiang, 2020-06-15:
[image]

kisvegabor, 2020-06-16:
Hi,

I can't reproduce the issue. Please, attach the generated font and short example code.

lilengqiang, 2020-06-16:
[image]
[image]

[arial.zip](https://github.com/lvgl/lvgl/files/4785522/arial.zip)

kisvegabor, 2020-06-16:
Ah, so you are using FreeType. Do you see the issue if you convert the font into a C array with onlne converter:
https://lvgl.io/tools/fontconverter

lilengqiang, 2020-06-16:
I had no problems using online converters, but I needed to convert a lot of fonts if I wanted to dynamically resize the text

lilengqiang, 2020-06-17:
Label also has the same problem, as long as I use '\n' through arial.ttf, there will be messy code.

kisvegabor, 2020-06-18:
I fixed it in the freetype repo. Now there is no rectangle for `\n`

kisvegabor, 2020-07-08:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2020-07-08 by kisvegabor