lvgl/lvgl #1579: Gauge angle offset issue: the label and needles don't go with angle offset

ZhaoQiang-b45475, 2020-06-15:
I create a gauge and set its angle_offset with 90，found that the label and needles are not rotated with the angle_offset, is this normal？

[image]
[image]

embeddedt, 2020-06-15:
If I understand the [documentation](https://docs.lvgl.io/v7/en/html/widgets/gauge.html#scale) correctly, the `angle_offset` property only affects the scale (the gray lines) and not the needles.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-06-15 by embeddedt

kisvegabor, 2020-06-15:
Actually I think it's a bug because can't image a case when only the ticks need to be moved.
I've fixed it `master`.

commit c20e62d4 (2020-06-15): lv_gaguge: make  `lv_gauge_get_angle_offset()` offset the labels and needles too
Fixes #1579