lvgl/lvgl #1582: Memory access violation when drawing list button with custom style.

nchurly, 2020-06-15:
I need a very simple list where focused items are highlighted with a border (monochrome display).
Here is my sample code:

	lv_obj_t * p_menu_list;
	lv_obj_t * p_list_btn;

	p_menu_list = lv_list_create(lv_scr_act(), NULL);
	lv_obj_set_size(p_menu_list, 120, 60);
	lv_obj_align(p_menu_list, NULL, LV_ALIGN_CENTER, 0, 0);
	lv_obj_reset_style_list(p_menu_list, LV_BTN_PART_MAIN);

	lv_style_t p_style;
	lv_style_init(&p_style);
	lv_style_set_border_color(&p_style, LV_STATE_FOCUSED, LV_COLOR_BLACK);
	lv_style_set_border_opa(&p_style, LV_STATE_FOCUSED, LV_OPA_100);

	p_list_btn = lv_list_add_btn(p_menu_list, NULL, "One");
	lv_obj_reset_style_list(p_list_btn, LV_BTN_PART_MAIN);
	lv_obj_add_style(p_list_btn, LV_OBJ_PART_MAIN, &p_style);

	p_list_btn = lv_list_add_btn(p_menu_list, NULL, "Two");
	lv_obj_reset_style_list(p_list_btn, LV_BTN_PART_MAIN);
	lv_obj_add_style(p_list_btn, LV_OBJ_PART_MAIN, &p_style);

It works when the lv_obj_add_style's are commented out.
Thanks.

embeddedt, 2020-06-15:
`p_style` should be `static`, so it isn't deallocated when the function exits.

nchurly, 2020-06-15:
> 
> 
> `p_style` should be `static`, so it isn't deallocated when the function exits.

Got it! It worked. I thought styles are being copied into object since they contain only one pointer to map array. I'm still learning the lib. Thank you!

closed 2020-06-15 by nchurly