lvgl/lvgl #1585: Release policy - fine tune

kisvegabor, 2020-06-16:
Hi,

The last discussion about the release policy was very useful and we set up some new rules. But in the practice some questions raised. 

The process looks like this now:
- 1st Tuesday of the month   
  - Make a major, minor, or patch release from master depending on the new features.  (also create/update release/vX branches).
  - After that merge only patches into master and add new features into the dev. (stabilize the last release)
- 3rd Tuesday of the month   
  - Make a patch release from master.  (also create/update release/vX branches).
  - After that merge the new features from the dev to master branch.   
  - In the rest of the month merge only patches into master and new features into dev branch.
- 1st Tuesday of the next month
  - Make a minor/major release and cycle starts again

So this way we will have minor/major releases every beginning of the moths and patch releases in the middle of the month. 

It could be simpler to merge dev every two weeks but it'd result in too many minor releases. 

Today I released v7.0.2 and merged dev to master in lvgl. (v7.1.0-dev in master) So people can already test the latest features from master. But what should we do with the docs? Shall it be updated to v7.1.0 to refer to be in sync with master, or keep it on v7.0.2?
What do you think?

embeddedt, 2020-06-16:
> But what should we do with the docs? Shall it be updated to v7.1.0 to refer to be in sync with master, or keep it on v7.0.2?

I have an idea for this. What if we added superscript tags to the documentation like `since 7.1` that could be placed on the ends of sentences? This would allow people to easily see whether the feature existed in previous versions. It would be great if these were automated so they disappeared when the docs were officially updated to 7.1, but even if that's not possible, I don't think they would be that messy.

Doxygen also has a [`since`](https://www.doxygen.nl/manual/commands.html#cmdsince) command which we can use for the API docs themselves.

kisvegabor, 2020-06-16:
Sounds like a good idea.
On release, a scrip can run a `sed` command to remove `(Since v*)` parts.

kisvegabor, 2020-06-17:
I've checked how it is handled in [Zephyr](https://docs.zephyrproject.org/) and [ESP](https://docs.espressif.com/).

They add links to the older versions and to the latest. 

I've restructured the project to work like this.
Now there are branches for `release/vX` and `latest`. `master` only contains the compiled html files.

Independently I think the `since` tag might be useful.

I've also added an Introduction and a Contributing section. (Feedback is very welcome)

embeddedt, 2020-06-18:
I noticed that the dropdown to choose a documentation version does not appear on any page besides `latest`. Do the other two need to be regenerated?

kisvegabor, 2020-06-19:
For the sake of simplicity, I was planning to add it when `latest` is merged into it.

It should be really added to v6. I'll do it.

I've just realized a `dev` version is also required:
- `docs/latest` -> `lvgl/master`
- `docs/dev` -> `lvgl/dev`

kisvegabor, 2020-06-25:
It's only me who feels that the policy of *" minor version at the end of the month and patch version in the middle of the month"* is messy and difficult to follow?

As fortunately, we have new features continuously the *"release a minor version every month"* way could be considered as well.
- bugfixes to `master`
- feature to `dev`
- release every 2 weeks ()minor or major, or patch if there were no features)
- after release, merge dev to master.

What do you think, isn't it too frequent? What are your good or bad experiences with other projects?

-

puzrin, 2020-06-25:
- `dev` - not necessary for features. Those can be merged to master directly.
- `next` - may be rarely useful in period when you prepare switch to new major version with breaking changes.
- bugfixes may be urgent. Leave 2-4 weeks as max lag for accumulated updates, and release patches more frequent if needed.

The only critical things to plan are major [breaking] releases. The rest can go upon demand.

kisvegabor, 2020-06-26:
Having an easy to follow schedule (e.g. every 2 weeks) for minor/major releases is a good idea IMO.
 
Is it really good practice to release even very fresh features (e.g. merged one day before the release) and make a bugfix release 3 days later?

embeddedt, 2020-06-26:
> Is it really good practice to release even very fresh features (e.g. merged one day before the release) and make a bugfix release 3 days later?

I don't think so. I think there should be some delay (at least 1-2 weeks) to allow for some amount of testing. In my opinion, it looks unprofessional if a release gets followed by a bugfix a day later.

puzrin, 2020-06-26:
> Is it really good practice to release even very fresh features (e.g. merged one day before the release) and make a bugfix release 3 days later?

But nobody forces you to merge feature branch early. Keep it unmerged until testing complete.

> Having an easy to follow schedule (e.g. every 2 weeks) for minor/major releases is a good idea IMO.

Split major/minor logic. Those are absolutely different.

For minor/patches, the only critical thing is to release bugfixes not later than after 2-4 weeks + urgent releases for rare fatal cases. Everything else is up to your personal preferences.

Major [breaking] releases are usually planned from 3 months, to not fall into "permanently broken" state.

embeddedt, 2020-06-26:
> But nobody forces you to merge feature branch early. Keep it unmerged until testing complete.

It's easier to test features all at once than to test multiple separate feature branches. That's why we made the `dev` branch to begin with.

puzrin, 2020-06-26:
> It's easier to test features all at once than to test multiple separate feature branches. That's why we made the `dev` branch to begin with.

Probably, you were influenced by previous branches struct, when `master` was recommended for download? Now `master` has the same role as old `dev`, and users are recommended to download by tags.

embeddedt, 2020-06-26:
If we release/tag `master` every 2 weeks, regardless of its contents, there is a chance that an unverified feature becomes part of a release. That should be avoided. That's why I'm suggesting that the `dev` branch needs to stay, as a place to keep features that need more testing.

I think both @kisvegabor and I would prefer not to have a large number of PRs waiting in a queue that have to be tested individually before being merged to `master`.

puzrin, 2020-06-26:
Ok, having additional staging branch is no a big deal, because does not change anything principle in project management. If it's more convenient for you at current moment - i see no problems.

kisvegabor, 2020-06-26:
Great to see that there is a consensus here. 

So turning back to the original question: is okay to make minor releases every 2 weeks?

puzrin, 2020-06-26:
> So turning back to the original question: is okay to make minor releases every 2 weeks?

In general - yes.

kisvegabor, 2020-06-28:
My only concern was the too many administration stuff (e.g. announce releases) but a straight forward easy to follow policy would compensate it. :slightly_smiling_face: 

@embeddedt do you also agree with the minor release every 2 weeks concept?

embeddedt, 2020-06-28:
It sounds good to me.

kisvegabor, 2020-06-29:
Great, I updated the docs: https://docs.lvgl.io/latest/en/html/intro/index.html#release-policy

closed 2020-06-29 by kisvegabor

embeddedt, 2020-06-29:
@kisvegabor There's still an issue with the documentation selector. No matter what branch you are actually looking at, the dropdown still shows `latest`.

kisvegabor, 2020-06-29:
Oh, yes. I forget to mention that. I require some JS "magic". 

This [JS file](https://github.com/lvgl/docs/blob/master/misc/common.js) is used for all branches.
The current branch can be extracted from the URL. Do you how to do it?

embeddedt, 2020-06-29:
That should be pretty straightforward to fix; I'll take a look at it.

embeddedt, 2020-07-20:
@kisvegabor The dropdown version selector is now fixed.

kisvegabor, 2020-07-20:
Fantastic, thank you ^^

embeddedt, 2020-08-07:
@kisvegabor Is 7.3 released? It got tagged (presumably by your script) but there was no GitHub release made for it.

kisvegabor, 2020-08-10:
@embeddedt 
I added the release yesterday. 
I've checked that GitHub actions can be used to add a release when a tag added. I'll implement it this week.

kisvegabor, 2020-09-22:
Aaah,  September was strange beacuse 1st of September was a Tuesday immediatelly, and I messed up with counting the weeks...  And just released v7.6 one week after v7.5. 

I think there is nothing to do now, next time more care must be taken...

embeddedt, 2020-09-22:
I wouldn't worry about it; these are minor releases so nothing should really break for users. Think of it as bugfixes being released sooner. :wink:

puzrin, 2020-09-22:
What are estimaes for 8.x? I'd like to skip 7.x migration.

kisvegabor, 2020-09-22:
> I wouldn't worry about it; these are minor releases so nothing should really break for users. Think of it as bugfixes being released sooner. wink

Thanks :slightly_smiling_face: 

> What are estimaes for 8.x? I'd like to skip 7.x migration.

Targeting this autumn.

embeddedt, 2020-10-31:
I think the release script is broken again... v7.7 is released but there's no GitHub release.

kisvegabor, 2020-11-02:
I do it manually because it seems simpler. 
I was interrupted during the last release and forget the GitHub release...

It seems we really need a script to do this. Maybe a GitHub action can handle if a new tag is pushed.

kisvegabor, 2020-11-02:
I've just found a GitHub action and added it: 
https://github.com/lvgl/lvgl/blob/master/.github/workflows/release.yml