lvgl/lvgl #1591: small button is hard to drag.

incity, 2020-06-18:
I created a small button size(20*20), and set draggable. but its hard  to drag.
[smallbtn_hard_to_drag.zip](https://github.com/lvgl/lvgl/files/4797447/smallbtn_hard_to_drag.zip)

kisvegabor, 2020-06-18:
Try `lv_obj_set_protect(btn, LV_PROTECT_PRESS_LOST)`

incity, 2020-06-18:
It's OK , but why need this?

embeddedt, 2020-06-18:
> It's OK , but why need this?

It's because your finger moves off the button before the button can catch up to your finger's position. Setting `LV_PROTECT_PRESS_LOST` ensures that the button continues receiving input events after your finger moves off of it.

This is the same way it works in HTML5/JavaScript. Typically the dragging is controlled by the window (and not the button itself), but the principle is the same - events are tracked even when your  finger moves outside of the button's bounding box.

incity, 2020-06-18:
thanks, by the way, for the textarea if i put the mouse on the text to start drag, it can not be dragged. but if i let the mouse leave the text and it can be dragged. how to solve this?

kisvegabor, 2020-06-18:
@incity 
We use GitHub Issues for development-related discussions.

Please ask questions in the [Forum](https://forumtest.littlevgl.com/) to make it visible for a larger community. You can simply log in with your GitHub account.

embeddedt, 2020-06-22:
I think the original question is answered so I'll close this issue.

closed 2020-06-22 by embeddedt