lvgl/lvgl #1594: Periodically set the text of Label will crashed randomly

weixiongmei, 2020-06-19:
```C
char * string_text = malloc(size_text + 1);
//removed some codes here that sets content of string_text, because it's not lvgl related
string_text[size_text] = '\0';
lv_label_set_text(label, string_text);
free(string_text);
```
When periodically set the text for a label, it will crashed at the line 123 of file lv_draw_label.c. Because the arg txt is NULL.
```C
if(txt[0] == '\0')  return;
```

The issue solved by the following code
```C
if(txt == NULL || txt[0] == '\0')  return;
```

embeddedt, 2020-06-19:
It would be better to figure out why `txt` is `NULL` in the first place; your solution is more of a workaround. My guess is that you're running out of memory and thus allocation of a text buffer fails somewhere.

weixiongmei, 2020-06-19:
> It would be better to figure out why `txt` is `NULL` in the first place; your solution is more of a workaround. My guess is that you're running out of memory and thus allocation of a text buffer fails somewhere.

I’m pretty sure it’s not because of running out of memory, because I did check the allocation, and the string_text before setting it. The above code it simplified, that’s why you didn’t see the verification before setting the text. The code is running on a ESP32, can’t do the debugging, can only trace where the exception is. Not sure I should free the string_text after lv_label_set_text or not...

embeddedt, 2020-06-19:
Freeing `string_text` should be safe because `lv_label_set_text` allocates its own buffer.

> I’m pretty sure it’s not because of running out of memory, because I did check the allocation, and the string_text before setting it.

Since a second buffer gets allocated internally, it's possible that that allocation is the one which fails. I can try running LVGL with a really small heap to see if I can make it happen.

kisvegabor, 2020-06-19:
Just to be sure, can you add (or is there) an assert after `char * string_text = malloc(size_text + 1);`.
E.g. 
```c
if(string_text == NULL) {
  printf("no mem\n"); 
  while(1);
} 
```

weixiongmei, 2020-06-19:
> Just to be sure, can you add (or is there) an assert after
> `char * string_text = malloc(size_text + 1);`?
> E.g.
> 
> ```c
> if(string_text == NULL) {
>   printf("no mem\n"); 
>   while(1);
> } 
> ```
The string_text==NULL is checked and printf(string_text); before calling the lv_label_set_text, and it's fine. Because it's running on the ESP32, and I don't have the tool to do the debugging, but the exception trace is below, hope it can helps a little more.

```
PC: 0x4011f1ed: lv_draw_label at /targets/esp32/components/module-gui/lvgl/src/lv_draw/lv_draw_label.c line 123
EXCVADDR: 0x00000000

Decoding stack results
0x4011f1ea: lv_draw_label at /targets/esp32/components/module-gui/lvgl/src/lv_draw/lv_draw_label.c line 118
0x40130423: lv_label_design at /targets/esp32/components/module-gui/lvgl/src/lv_widgets/lv_label.c line 1106
0x4011cef6: lv_refr_obj at /targets/esp32/components/module-gui/lvgl/src/lv_core/lv_refr.c line 626
0x4011cf73: lv_refr_obj at /targets/esp32/components/module-gui/lvgl/src/lv_core/lv_refr.c line 666
0x4011cfb5: lv_refr_obj_and_children at /targets/esp32/components/module-gui/lvgl/src/lv_core/lv_refr.c line 606
0x4011d041: lv_refr_area_part at /targets/esp32/components/module-gui/lvgl/src/lv_core/lv_refr.c line 504
0x4011d56a: _lv_disp_refr_task at /targets/esp32/components/module-gui/lvgl/src/lv_core/lv_refr.c line 460
0x40127ce3: lv_task_exec at /targets/esp32/components/module-gui/lvgl/src/lv_misc/lv_task.c line 398
0x40127dbb: lv_task_handler at /targets/esp32/components/module-gui/lvgl/src/lv_misc/lv_task.c line 142
```

embeddedt, 2020-06-19:
`lv_label_design` [calls `lv_draw_label`](https://github.com/lvgl/lvgl/blob/8dbd685a908995a951fff82a9e96b6527e0e542d/src/lv_widgets/lv_label.c#L1106) with the `ext->text` value, so there must be memory corruption somewhere.

weixiongmei, 2020-06-19:
> `lv_label_design` [calls `lv_draw_label`](https://github.com/lvgl/lvgl/blob/8dbd685a908995a951fff82a9e96b6527e0e542d/src/lv_widgets/lv_label.c#L1106) with the `ext->text` value, so there must be memory corruption somewhere.

Agree, but if string_text is NULL, it should failed at string_text[size_text] = '\0';

embeddedt, 2020-06-19:
I think the corruption happens later, not when you set `lv_label_set_text`.

weixiongmei, 2020-06-19:
> I think the corruption happens later, not when you call `lv_label_set_text`.

If I don't reset the text of the label, the program can runs forever..

embeddedt, 2020-06-19:
That's probably a timing or heap-related issue, since you are calling `malloc` before doing anything with LVGL. Try commenting out just the call to `lv_label_set_text` (but still allocate and check `string_text` like you have been doing) and see if the issue still happens.

kisvegabor, 2020-06-19:
Do you an OS? If so do you use mutexes as described here?
https://docs.lvgl.io/latest/en/html/porting/os.html

weixiongmei, 2020-06-19:
> Do you an OS? If so do you use mutexes as described here?
> https://docs.lvgl.io/latest/en/html/porting/os.html

Yes, ESP32 is running the FreeRTOS, but everything was called from the same task, I'm only running a single task program. But I will try the mutex see how it behaves. Thanks @embeddedt , @kisvegabor

weixiongmei, 2020-06-19:
```C
static void lv_tick_timercb(void *timer)
{
  while(xSemaphoreTake(mutex_lvgl, portMAX_DELAY) !=pdTRUE);
  lv_tick_inc(1);
  xSemaphoreGive(mutex_lvgl);
}

static void lv_task_timercb(void *timer)
{
  while(xSemaphoreTake(mutex_lvgl, portMAX_DELAY) !=pdTRUE);
  lv_task_handler();
  xSemaphoreGive(mutex_lvgl);
}
```
```C
char * string_text = malloc(size_text + 1);
//removed some codes here that sets content of string_text, because it's not lvgl related
string_text[size_text] = '\0';
while(xSemaphoreTake(mutex_lvgl, portMAX_DELAY) !=pdTRUE);
lv_label_set_text(label, string_text);
xSemaphoreGive(mutex_lvgl);
free(string_text);
```
@kisvegabor, seems like the mutex solved the issue, but wondering should I call xSemaphoreTake prior lv_tick_inc() and lv_task_handler() like the code above? Thank you so much~~~

embeddedt, 2020-06-19:
There's no need to take the mutex for `lv_tick_inc`. Other than that, this is the correct approach.

weixiongmei, 2020-06-19:
@embeddedt very appreciated~~

weixiongmei, 2020-06-19:
@embeddedt, just tested without mutex for lv_tick_inc, it'll still going to crash. So mutex must be called prior to lv_xxxxxx.

embeddedt, 2020-06-19:
>  just tested without mutex for lv_tick_inc, it'll still going to crash. So mutex must be called prior to lv_xxxxxx.

This sounds like another workaround that isn't actually fixing the issue, then. All `lv_tick_inc` does is increment a variable (and thus indirectly trigger a screen refresh when `lv_task_handler` is run next).

I think you just got lucky. :confused:

weixiongmei, 2020-06-19:
> This sounds like another workaround that isn't actually fixing the issue, then. All `lv_tick_inc` does is increment a variable (and thus indirectly trigger a screen refresh when `lv_task_handler` is [...]

That's not good, I didn't mutex for the following code, would that be the problem?
```C
static void display_render(lv_disp_drv_t * disp, const lv_area_t * area, lv_color_t * color_p)
{
    int32_t x, y;
  for (y = area->y1; y <= area->y2; y++)
  {
    for (x = area->x1; x <= area->x2; x++)
    {
      buffer_fullRendering[y * 320 + x] = *color_p;
      color_p++;
    }
  }
  lv_disp_flush_ready(disp); /* Indicate you are ready with the flushing*/
}
```

embeddedt, 2020-06-19:
You only need a mutex around `lv_task_handler` and other LVGL APIs. `display_render` is called from `lv_task_handler` so adding a mutex is pointless there.

In short, concurrency is probably not the problem; memory corruption is probably still happening.

weixiongmei, 2020-06-19:
@embeddedt Sounds bad..

embeddedt, 2020-06-19:
Without a debugger, the only way to trace this type of bug is probably to try disabling parts of your code and see if the issue still happens. You may be able to narrow it down to some specific area of memory your code touches.

With a debugger, it can be much easier as you can often set a watchpoint on `ext->text` and wait for its value to change to `NULL`.

weixiongmei, 2020-06-19:
Right, the biggest problem is the ESP32 board I have doesn't pinout the JTAG and didn't have a debugging tool for it, may have to get an official dev board and debugger to address the issue. However  Thank you so much for helping...

embeddedt, 2020-06-19:
Are you doing any other "manual" heap allocation (i.e. outside of what LVGL does)? Is `LV_MEM_CUSTOM` enabled (which would mean LVGL uses `malloc` and `free`) or is it disabled (meaning LVGL uses its own heap)?

weixiongmei, 2020-06-19:
#define LV_MEM_CUSTOM      1
What do you mean manual heap allocation? Do you mean calling malloc & free somewhere else? The testing program code is very simple, the code I posted in the thread is most likely all the code I'm using for testing.

weixiongmei, 2020-06-19:
@embeddedt curious why mutex for lv_tick_inc can avoid the issue....

embeddedt, 2020-06-19:
> What do you mean manual heap allocation? Do you mean calling malloc & free somewhere else?

Yes.

> curious why mutex for lv_tick_inc can avoid the issue....

Without a debugger I can't give a guaranteed explanation, but based on my prior experience with issues like this, timing often causes/hides the issue. Adding the mutex will change the timing of execution somewhat. Sometimes simply having the debugger connected can prevent the issue, as that can also change the timing.

weixiongmei, 2020-06-19:
Yes I do call malloc and free the buffer for the SPI communication somewhere else. Thank you for explaining what can be causing this issue. Maybe I just go get a official ESP32 dev board and a debugger to figure it out myself. Thank you~~~

pete-pjb, 2020-06-20:
Hi @weixiongmei,

I had a thought to add:

If you are using FreeRTOS, have you checked you are not overflowing the stack in any of your tasks?

You can get the stack high water mark with the [uxTaskGetStackHighWaterMark()](https://www.freertos.org/uxTaskGetStackHighWaterMark.html). This function displays the lowest free space the stack has reached for the specified task since starting your application.

Also there are many options for memory management with FreeRTOS and from experience a slight error in the configuration can cause this type of issue very easily.  If you want to share some of your FreeRTOS configuration information here also I can take a look and see if I can see anything that might be an issue if that is helpful?

Kind Regards,

Pete

weixiongmei, 2020-06-20:
@pete-pjb I think I have solved the issue. The problem is the lv_tick_inc() & lv_task_handler() were called from two different timers. So have to mutex for lv_tick_inc() to work properly. Now I'm calling these two functions from the same timer, the issue is solved now. Thank you!!
```C
static void lvgl_timercb(void *timer)
{
  lv_tick_inc(5);
  LVGL_MUTEX_TAKE
  lv_task_handler();
  LVGL_MUTEX_GIVE
}
```

embeddedt, 2020-06-20:
I still think you have not found the problem. If you look at the implementation for `lv_tick_inc`, all it does is increment a variable. I think you are just getting lucky with the timing.

Additionally, `lv_tick_inc` is supposed to run independently of `lv_task_handler`, so that LVGL's sense of time keeps moving even while `lv_task_handler` is running. Most people using LVGL have it implemented in this manner, and `lv_tick_inc`/`lv_task_handler` are designed to work concurrently with each other (they are the only functions that do so).

embeddedt, 2020-06-20:
I could be mistaken, but without a clear explanation of exactly why this solves the problem, you have no way of guaranteeing that a similar class of bug won't pop up later (possibly with worse side effects). The only way to completely fix a bug like this is to know what sets the pointer to `NULL` and fix the problem at that level, or at least know why running `lv_tick_inc` and `lv_task_handler` serially prevents the issue from occuring.

pete-pjb, 2020-06-20:
Hi @weixiongmei,

I agree with @embeddedt here you are probably getting lucky with the timing.  
If I were you I would definitely check your task stacks.  
Also are you using the FreeRTOS memory allocation functions or the standard C library malloc() and free() functions? Some implementations of the standard C library memory functions are not thread/task safe and don't work well when called from FreeRTOS tasks, as I have found out the hard way in the past.

Kind Regards, 

Pete

weixiongmei, 2020-06-20:
I use standard C library malloc() and free(), I did checked the stacks & heap usage, everything seems fine. The test code have been running for about 20 hours with no issue(set label text every 20ms), usually it crashed within an hour, most of the time around 10 minutes. The FreeRTOS has all default configurations from the ESP32 ESP-IDF, and I don't really know about the code structure where the configuration files are. But the testing code is most likely what I posted in this thread, it's just a very simple code for testing purposes, and it has 4MB of heap, and has more than 300KB of stack, and I did check the allocation & stack overflow, it seems fine. I think the issue is conflict with multi tasking/multi threading.

embeddedt, 2020-06-20:
>  I think the issue is conflict with multi tasking/multi threading.

I'm 99% sure `lv_tick_inc` and `lv_task_handler` work fine together. Many people use them in this configuration so it seems highly unlikely that that combination has a bug.

To be fair, I'll test the same thing on my STM32 board and see if a crash happens, but I very much doubt it will. :confused: I will run `lv_tick_inc` and `lv_task_handler` on separate threads too.

> I use standard C library malloc() and free()

That could be your problem; there are several articles talking about how these don't work well with FreeRTOS. An allocator which isn't thread-safe will corrupt memory very quickly. It might be worth investigating this in more detail.

weixiongmei, 2020-06-20:
@embeddedt If I wrap malloc & free with mutex into a custom function, do you think that will works? I think ESP32 allows custom allocator.

embeddedt, 2020-06-20:
I'm not sure. You'd have to investigate to see if they're thread-safe or not. Are you using [`lv_port_esp32`](https://github.com/lvgl/lv_port_esp32) or a custom ESP32 project?

weixiongmei, 2020-06-20:
@embeddedt I'm not using the lv_port_esp32, just created an empty project, and added the lvgl to the project. The following code will make the crash happens I believe.

```C
static void lv_tick_timercb(void *timer)
{
  lv_tick_inc(1);
}

static void lv_task_timercb(void *timer)
{
  lv_task_handler();
}

  esp_timer_create_args_t timer_conf = {
      .callback = lv_tick_timercb,
      .name     = "lv_tick_timer"
  };

  esp_timer_handle_t lv_tick_timer = NULL;
  esp_timer_create(&timer_conf, &lv_tick_timer);
  esp_timer_start_periodic(lv_tick_timer, 1 * 1000U);

  esp_timer_create_args_t lv_task_timer_conf = {
      .callback = lv_task_timercb,
      .name     = "lv_task_timer"
  };

  esp_timer_handle_t lv_task_timer = NULL;
  esp_timer_create(&lv_task_timer_conf, &lv_task_timer);

  esp_timer_start_periodic(lv_task_timer, 5 * 1000U);
```

embeddedt, 2020-06-20:
@weixiongmei I've been digging through past discussions on `lv_port_esp32`. I may have found your problem. ESP32 has two cores - by default tasks can run on either core. It seems that switching a task between cores can cause memory corruption, even if mutexes are used. This seems to be some ESP32-specific caching behavior since I haven't heard of it happening on other platforms.

Since I'm not familiar with ESP32, I have no idea how you would specifically solve that problem, but a workaround which has been reported to work is to pin the LVGL-related tasks to a single core. https://github.com/lvgl/lv_port_esp32/pull/84#issuecomment-602649081 (and [related diff](https://github.com/lvgl/lv_port_esp32/pull/84/files#diff-1a9cfc6173e3a434387996e46086da56R46-R48))

weixiongmei, 2020-06-20:
@embeddedt I only enabled core0, core1 is disabled at the compiled time in  the menuconfig.

embeddedt, 2020-06-20:
Hmm. I guess we are back to the memory allocator theory then. I think @pete-pjb has more experience with FreeRTOS allocators than I do so maybe he can guide you as to how you would ensure they are thread-safe.

weixiongmei, 2020-06-20:
@embeddedt OK, thank you~~~

pete-pjb, 2020-06-20:
Hi @weixiongmei,

Is there a link to the standard ESP32 build you are using which you can post on here.  I am happy to download it, take a look and get back to you, if that's helpful? 

Please bare with me though if you want me to look as I have a few commitments right now and it might take me few days to get back to you. :smiley: 

Regards, 

Pete

weixiongmei, 2020-06-20:
@pete-pjb I'm using the most recent ESP-IDF that's downloaded from the ESP official website, and added the lvgl into the hello world project, and using the codes in this thread.

BrianPugh, 2020-06-20:
Not sure if it's any help, but here's how i use lvgl in my project (I'm currently using 6.X, haven't had time to upgrade to 7.X):

https://github.com/joltwallet/jolt_wallet/blob/c43dc6f4852ff95b364b770e99e473fea758329e/jolt_os/main.c#L96

This uses a bit of meta-programming magic from this author/article/header:

https://www.chiark.greenend.org.uk/~sgtatham/mp/

https://github.com/joltwallet/jolt_wallet/blob/c43dc6f4852ff95b364b770e99e473fea758329e/jolt_os/jolt_gui/jolt_gui.h#L319

This makes the programming a bit akin to python context managers, and IMHO makes it cleaner and less likely for bugs to creep in (like forgetting to free a mutex).

pete-pjb, 2020-06-21:
Thanks for the information @weixiongmei I'll take a look and get back to as soon as I can.

pete-pjb, 2020-06-22:
Hi @weixiongmei,

cc @embeddedt, @kisvegabor 

I have taken a look at the ESP32 FreeRTOS implementation and it seems the developers have kind of done their own thing with it.

Firstly they have created their own SMP version of FreeRTOS which I am not at all familiar with and is unique to them.
I dug into the source files a bit and as far as memory management goes they are using the newlib C library and they have implemented it with support for reentrancy so on that basis it should be fine to use the malloc() and free() (also printf etc.) calls in the multi-threaded/task environment.  So I would think we can probably safely discount that as a source of potential problems in your code.

Kind Regards,

Pete

kisvegabor, 2020-06-22:
Thanks, @pete-pjb 

@weixiongmei 
To be sure that there are no issues in your code can you try NLY something like this:
```c
    static const char * txts[] = {"one", "two", "three", "four"};
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);

    while(1) {
        lv_label_set_text(label, txts[0]);
        lv_label_set_text(label, txts[1]);
        lv_label_set_text(label, txts[2]);
        lv_label_set_text(label, txts[3]);
    }
```

kisvegabor, 2020-07-08:
It might be related to https://github.com/lvgl/lvgl/issues/1639

--- 

I close this issue in lack of activity. If you have further remarks on it feel free to comment here and reopen the issue.

closed 2020-07-08 by kisvegabor