lvgl/lvgl #1597: lv_btnmatrix_set_focused_btn does not work

curly-brace, 2020-06-19:
**Describe the bug**

lv_btnmatrix_set_focused_btn does nothing

**To Reproduce**

```
    lv_obj_t* btns= lv_btnmatrix_create(lv_scr_act(), NULL);
    lv_btnmatrix_set_map(btns, buttons);
    lv_btnmatrix_set_focused_btn(btns, 0);
    lv_btnmatrix_set_btn_ctrl_all(btns, LV_BTNMATRIX_CTRL_CHECKABLE);
    lv_btnmatrix_set_one_check(btns, true);
```

**Expected behavior**

i believe it should highlight the very first button, but it does not.
removing last two lines change nothing in the example tho, but i need it to work in that scenario.

**Screenshots**
none

**Additional context**
LVGL 7.0.1

pete-pjb, 2020-06-19:
Hi @curly-brace,

We use the [forum](https://forum.lvgl.io) for questions and bug reporting, can you please post there, Github is used for development related issues.  You can login to the forum with your Github accont.  Thank you.

I have pasted your code sample above into the simulator and it appears to function correctly here is a screenshot:
[image]

closed 2020-06-19 by pete-pjb

kisvegabor, 2020-06-19:
@curly-brace 
Please update lvgl from `master`. v7.0.2 is out, maybe it was already fixed.