lvgl/lvgl #1601: LVGL widgets demo crashing on big endian

jbamaral, 2020-06-21:
**Describe the bug**

The `lv_demo_widgets` app is crashing on [one of my test terminals](https://www.verifone.com/search?keys=VX%20690) which is big endian.

**To Reproduce**
Right at start of the `lv_demo_widgets` app the loops to get property memory size never end at `_lv_style_get_mem_size` and as it allocates more and more memory the terminal crashes. If you enable assertions it will stop in a memory failure error. 

You need big endian hardware to reproduce the error. 

After investigating for a while I figured that the way `lv_style_t` map is being handled is not portable and the comparison on the loops using `style->map[i]` would never be true. I wrote a fix suggestion at #1599, but after some more investigation I figured that it didn't solve the whole problem. 

**Expected behavior**

Demo application should run as it runs in the simulator and other little endian terminals that I'm testing.

**Screenshots**
Although there is no screen shot useful for the problem (as it crashes), I think that a screen shot of the terminal after applying #1599 maybe useful. This screen shot is from a simplified version of the `lv_demo_widgets`. The full demo works fine except for these color mismatch problems in some controls. 

[image]

The code for the simplified version follows:
```c
void
simpleDemo(void)
{
    tv = lv_tabview_create(lv_scr_act(), NULL);

    if (LV_THEME_DEFAULT_INIT == lv_theme_material_init)
    {
        lv_disp_size_t disp_size = lv_disp_get_size_category(NULL);
        if (disp_size >= LV_DISP_SIZE_MEDIUM)
        {
            lv_obj_set_style_local_pad_left(
                tv, LV_TABVIEW_PART_TAB_BG, LV_STATE_DEFAULT, LV_HOR_RES / 2);
            lv_obj_t *sw = lv_switch_create(lv_scr_act(), NULL);
            if (lv_theme_get_flags() & LV_THEME_MATERIAL_FLAG_DARK)
                lv_switch_on(sw, LV_ANIM_OFF);
            lv_obj_set_event_cb(sw, color_chg_event_cb);
            lv_obj_set_pos(sw, LV_DPX(10), LV_DPX(10));
            lv_obj_set_style_local_value_str(
                sw, LV_SWITCH_PART_BG, LV_STATE_DEFAULT, "Dark");
            lv_obj_set_style_local_value_align(sw,
                                               LV_SWITCH_PART_BG,
                                               LV_STATE_DEFAULT,
                                               LV_ALIGN_OUT_RIGHT_MID);
            lv_obj_set_style_local_value_ofs_x(
                sw, LV_SWITCH_PART_BG, LV_STATE_DEFAULT, LV_DPI / 35);
        }
    }

    t1 = lv_tabview_add_tab(tv, "Controls");

    lv_style_init(&style_box);
    lv_style_set_value_align(
        &style_box, LV_STATE_DEFAULT, LV_ALIGN_OUT_TOP_LEFT);
    lv_style_set_value_ofs_y(&style_box, LV_STATE_DEFAULT, -LV_DPX(10));
    lv_style_set_margin_top(&style_box, LV_STATE_DEFAULT, LV_DPX(30));

    controls_create(t1);
}

/**********************
 *   STATIC FUNCTIONS
 **********************/

static void
controls_create(lv_obj_t *parent)
{
    lv_page_set_scrl_layout(parent, LV_LAYOUT_PRETTY_TOP);

    lv_disp_size_t disp_size = lv_disp_get_size_category(NULL);
    lv_coord_t grid_w = lv_page_get_width_grid(
        parent, disp_size <= LV_DISP_SIZE_SMALL ? 1 : 2, 1);

    lv_obj_t *h = lv_cont_create(parent, NULL);
    lv_cont_set_layout(h, LV_LAYOUT_PRETTY_MID);
    lv_obj_add_style(h, LV_CONT_PART_MAIN, &style_box);
    lv_obj_set_drag_parent(h, true);

    lv_obj_set_style_local_value_str(
        h, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, "Basics");

    lv_cont_set_fit2(h, LV_FIT_NONE, LV_FIT_TIGHT);
    lv_obj_set_width(h, grid_w);
    lv_obj_t *btn = lv_btn_create(h, NULL);
    lv_btn_set_fit2(btn, LV_FIT_NONE, LV_FIT_TIGHT);
    lv_obj_set_width(
        btn,
        lv_obj_get_width_grid(h, disp_size <= LV_DISP_SIZE_SMALL ? 1 : 2, 1));
    lv_obj_t *label = lv_label_create(btn, NULL);
    lv_label_set_text(label, "Button");

    btn = lv_btn_create(h, btn);
    lv_btn_toggle(btn);
    label = lv_label_create(btn, NULL);
    lv_label_set_text(label, "Button");

    lv_checkbox_create(h, NULL);
}

static void
color_chg_event_cb(lv_obj_t *sw, lv_event_t e)
{
    if (LV_THEME_DEFAULT_INIT != lv_theme_material_init)
        return;
    if (e == LV_EVENT_VALUE_CHANGED)
    {
        uint32_t flag = LV_THEME_MATERIAL_FLAG_LIGHT;
        if (lv_switch_get_state(sw))
            flag = LV_THEME_MATERIAL_FLAG_DARK;

        LV_THEME_DEFAULT_INIT(lv_theme_get_color_primary(),
                              lv_theme_get_color_secondary(),
                              flag,
                              lv_theme_get_font_small(),
                              lv_theme_get_font_normal(),
                              lv_theme_get_font_subtitle(),
                              lv_theme_get_font_title());
    }
}
```

**Additional context**

I've opened  #1599 with a partial fix but I decided to close it while investigating further. Although this fix avoids the crash and all controls that I tested looks like the same behavior of the app running on little endian terminals and the simulator, I decide to close it for now while a fix for the `lv_style_attr_t` behavior is not available.

jbamaral, 2020-06-21:
After thinking about a low impact way to fix the attribute usage I pushed jbamaral@cdf38a9fdd which I think it's acceptable (at least for a hotfix).

Behavior improved a lot but I still am getting some errors that I didn't figure out yet :( Behavior on little endian is ok with these changes. 

I'm attaching some photos of components with problems in order to make it easier for who knows the code base to figure out where to start from (I'm having hard time here as I just started studying this code to try and solve this problem...)

- Keyboard with some artifacts
[image]

- Checkbox selected doesn't work
[image]

- Checkbox not selected (OK) for reference
[image]

embeddedt, 2020-06-21:
Looks like it's having troubles finding the right font symbol to draw. Normal characters seem to display without issues.

jbamaral, 2020-06-22:
Ok. With your hint I've found the problem and fixed here jbamaral@8a0a5d47f4229b50973cae501b7135222902cb0f

As I don't see any problem running the lv_demo_widgets.c I'll reopen the PR and let you decide if those fixes are ok for a hotfix. 

Thanks for the help!

jbamaral, 2020-06-23:
solved by #1599

closed 2020-06-23 by jbamaral