lvgl/lvgl #1602: Roller text not being refreshed

embeddedt, 2020-06-21:
**Describe the bug**

The roller text is not being redrawn on the extreme left and right of the letters when you scroll it.

**To Reproduce**

Run `lv_demo_widgets` on a 480x272 screen size. Choose the "Selectors" tab and try scrolling the roller.

**Expected behavior**

Text should be redrawn.

kisvegabor, 2020-06-22:
It happens because the default font is too large and can't with into the `width - padding` size.

We can lower the default font size to ensure it fits better on different resolutions and DPIs.

embeddedt, 2020-06-22:
Shouldn't it get refreshed even if it doesn't fit?

kisvegabor, 2020-06-22:
The edges of the text are out of the scrollable part of the roller. By default, the width of the roller is set to the width of the text but if the width is set manually (as in the demo) it can be too short. 
Anyway, probably it's better to not respect the padding then cropping the text. 
If you agree I can update it accordingly.

embeddedt, 2020-06-22:
> The edges of the text are out of the scrollable part of the roller.

I see.

> Anyway, probably it's better to not respect the padding then cropping the text. If you agree I can update it accordingly.

I agree. I can't see any specific use-case for the roller where someone would want the text cropped.

kisvegabor, 2020-06-22:
I started to add it but realized it's not that simple as I thought at first look because there is no way to draw out of the parent. 

Now the roller's scrollable part is not exposed to the user (there are only `BG` and `SELECTED` parts) and the scrollable is always transparent with paddings = 0.

kisvegabor, 2020-07-08:
For now, I fixed it by setting the 14 px font as default. 

It seems it will be manageable with the new scrolling mechanism.

closed 2020-07-08 by kisvegabor